espressif/arduino-esp32 #6445: Curl library

guoquan1963, 2022-03-18:
### Related area

Http Request

### Hardware specification

ESp32

### Is your feature request related to a problem?

Curl libarary

### Describe the solution you'd like

I want to communicate with https server.
Htts server with security token.
If I use command "curl -i -k -H "X_Auth_Token:secureSnoreToken" https://192.168.100.1:9882/snore/termmgmt/install/gps", it works fine.
Can anybody tell me if I need to implement curl on ESP32 and how? Or I can use http client to do so and how?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2022-03-19:
The Arduino API uses [WiFiClientSecure](https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFiClientSecure) for TLS connections.  There are examples in that library.

VojtechBartoska, 2022-03-31:
Hello @guoquan1963, did you try approach shared by @lbernstone ?

closed 2022-04-20 by VojtechBartoska