espressif/arduino-esp32 #6449: ESP32C3 the CDC is not disconnected after OTA leaving the USB port unfonctional

0x0fe, 2022-03-19:
### Board

ESP32-C3 dev module

### Device Description

ESP32C3 Dev module

### Hardware Configuration

dna

### Version

v2.0.0

### IDE Name

arduino IDE

### Operating System

windows 10

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

CDC

### Description

with CDC enabled an OTA update is done, the system resets correctly to the new firmware however the CDC port is not disconnected and reconnected, as a result the arduino console, which is still open, does not show the new port, and even closing/reopening the console does not work, in fact the CDC port seems non functional and only a physical power cycle, with USB unplug, can restore the CDC.
I am not sure of what is not done but for having dealt with similar issues pf CDC / bootlaoder on STM32 it looks like the USB disconnection is not enforced after the OTA switches to the new firmware. I searched and found a reference to a single function which seems to provide such function, however i did not managed to use it yet.
https://github.com/espressif/arduino-esp32/blob/00214d5c2a1c2b1904f2caf6f0d5ddfe952331ff/tools/sdk/esp32/include/hal/include/hal/usb_phy_hal.h#L77

### Sketch

```cpp
default OTA exaxmple with CDC enabled
```

### Debug Message

```plain
none
```

### Other Steps to Reproduce

to reproduce, just use the default OTA example and make sure CDC is enable, you may print some message with a soft timer in order to see that the port is active, after the OTA you will see that the message is not printed anymore on the serial console unless USB cable is disconnected and reconnected.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-11:
Hello, are you able to test this on Arduino core esp32 v2.0.3-rc1? Thanks!

VojtechBartoska, 2022-08-04:
Closing this issue as expired due to no answer. If needed, you can reopen it. Thanks for understanding.

closed 2022-08-04 by VojtechBartoska