espressif/arduino-esp32 #6452: UDP destination port is always 0

AcuarioCat, 2022-03-20:
### Board

ESP32 dev module, ESP32-C3

### Device Description

Plain module

### Hardware Configuration

No

### Version

v2.0.2

### IDE Name

Visual Micro

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Sending UDP packets always sets destination to 0, ignores destination port from beginPacket parameter.

Tested on ESP32, ESP32-C3
Same code tested on ESP8266 and works correctly, destination port as expected.

Monitoring with WireShark UDP source port: 4440, destination port: 0

### Sketch

```cpp
#include <WiFiUDP.h>
#include <WiFi.h>

const char* ssid = "MYSSID";
const char* wifipass = "MYPASS";
WiFiUDP udp;

void setup() {
    Serial.begin(115200);
    WiFi.begin(ssid, wifipass);	
    udp.begin(4440);

}

void loop() {

    Serial.println("Send UDP");
    udp.beginPacket("0.0.0.0", 4321);
    udp.println("hello world udp test!");
    udp.endPacket();
    delay(5000);
}
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

AcuarioCat, 2022-03-29:
It seems there is a difference between ESP8266 and ESP32 when it comes to the destination IP address.
Whereas ESP8266 is quite happy to use 0.0.0.0 it seems the ESP32 only sends data when it is sent to a broadcast destination ip so 255.255.255.255 works, as does (in my case) 192.168.0.255 but a non-broadcast address (such as 192.168.0.111) does not work and no data is sent.

VojtechBartoska, 2022-04-11:
Hello @AcuarioCat, seems you have figure out the answer on your own. Can I consider this as answered?

closed 2022-04-23 by AcuarioCat