espressif/arduino-esp32 #6461: Crash loop when using esp_coex_preference_set function

FernandoGarcia, 2022-03-21:
### Board

ESP32-Wrover IE 8MB

### Device Description

DevKit

### Hardware Configuration

No.

### Version

other

### IDE Name

Platformio

### Operating System

Windows 11

### Flash frequency

40 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hi!

I'm trying to use the function `esp_coex_preference_set(ESP_COEX_PREFER_BT)` but after connect to ESP32 access point I'm getting a crash loop according to debug message below. 

I have attached the code and the ESP8266React lib (framework) is [here](https://github.com/FernandoGarcia/esp8266-react/tree/make_library).

Webserver and bluetooth can work fine separately without `esp_coex_preference_set(ESP_COEX_PREFER_BT)` function but together just bluetooth works fine.

I have an ESP32-Wroover IE 8MB and I'm using PIO 5.2.5 with espressif32 3.5.0.

Can someone give some help?

Best regards.

### Sketch

```cpp
#define BLUETOOTH_NAME "Parking Sensor"
#define BLUETOOTH_PINCODE "0000"

#include <ESP8266React.h>
#include "BluetoothSerial.h"
#include <ArduinoJson.h>
#include <esp_coexist.h>

BluetoothSerial SerialBT;

AsyncWebServer server(80);
ESP8266React esp8266React(&server);

unsigned start_millis;

void bluetoothHandle(String msg);

void setup()
{
  esp_coex_preference_set(ESP_COEX_PREFER_BT); // Priority for bluetooth

  Serial.begin(115200);
  
  esp8266React.begin();
  server.begin();

  SerialBT.setPin(BLUETOOTH_PINCODE);
  SerialBT.begin(BLUETOOTH_NAME);  
}

void loop()
{
  String msg;

  esp8266React.loop();

  if (SerialBT.available())
  {
    esp8266React.loop();
    start_millis = millis();
    msg = SerialBT.readString();

    bluetoothHandle(msg);
  }
}

void bluetoothHandle(String msg)
{
  char charBuf[100];
  msg.toCharArray(charBuf, 100);

  DynamicJsonDocument Json(100);

  Serial.print("t1 = ");
  Serial.print(millis() - start_millis);
  Serial.print(", ");

  String mode[2] = {"reverse", "break"};
  String sensor[8] = {"a", "b", "c", "d", "e", "f", "g", "h"};

  Json["mode"] = mode[random(0, 2)];
  Json["sensor"]   = sensor[random(0, 8)];
  Json["distance"] = random(0, 200);

  serializeJson(Json, SerialBT);

  Serial.print("cmd = ");
  Serial.println(charBuf);
  Serial.print("res = ");
  serializeJson(Json, Serial);
  Serial.print(", t2 = ");
  Serial.println(millis() - start_millis);
}
```

### Debug Message

```plain
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[E][WiFiGeneric.cpp:137] wifiLowLevelInit(): esp_wifi_init 4353
[E][BluetoothSerial.cpp:865] isReady(): BT is not initialized. Call begin() first
Starting software access point
[E][WiFiAP.cpp:109] softAP(): passphrase too short!
Starting captive portal on 192.168.4.1
abort() was called at PC 0x401932d3 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f694:0x3ffd31e0 0x4008f911:0x3ffd3200 0x401932d3:0x3ffd3220 0x4019331a:0x3ffd3240 0x401929ef:0x3ffd3260 0x40192d26:0x3ffd3280 0x40192a8d:0x3ffd32a0 0x401af79a:0x3ffd32c0 0x401b0019:0x3ffd3300 0x400e78d7:0x3ffd3330 0x400e8f48:0x3ffd3350 0x400dfa2e:0x3ffd3370 0x400d3a9e:0x3ffd3390 0x400ecf38:0x3ffd33e0 0x40090986:0x3ffd3400
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-03-21:
espressif32 3.5.0. is old core 1.0.6. It is not supported anymore since 2.0.x is out.

FernandoGarcia, 2022-03-21:
Sorry but I don't understand your answer.

For what I can see [here](https://github.com/platformio/platform-espressif32/releases/tag/v3.5.0) espressif32 3.5.0 for PIO is based in ESP-IDF 4.3.2 and [ESP32 for Arduino IDE](https://github.com/espressif/arduino-esp32/releases/tag/2.0.2) is version 2.0.2 based in ESP-IDF 4.4-beta1.

Jason2866, 2022-03-22:
Yes, and since you wrote you are using Platformio you use old Arduino core 1.0.6
(espressif32 3.5.0 == Arduino 1.0.6)

FernandoGarcia, 2022-03-22:
I'm trying to use the solution you gave [here](https://github.com/platformio/platform-espressif32/issues/619#issuecomment-1061971705).

`platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3/platform-espressif32-2.0.3.zip`

 But after fix compiling errors like changing `SYSTEM_EVENT_STA_STOP` to `ARDUINO_EVENT_WIFI_STA_STOP` the station mode doesn't works well.

I can connect to AP and set WiFi credentials but it's getting an invalid IP.

[image]

Looking the router settings I can see the IP 192.168.1.132 but the controller doesn't responds to this IP.

Sometimes the controller disconnects from network with reason 201.

Any suggestion to fix it?

VojtechBartoska, 2022-03-31:
Hello, Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

FernandoGarcia, 2022-03-31:
Hi!

Unfortunately I can't test with Arduino IDE because the framework I'm using is not compatible with it.

Using latest [Tasmota framework-arduino-espressif32](https://github.com/tasmota/arduino-esp32/releases/tag/2.0.3rc1) based on 2.0.3-RC1 I have the invalid IP issue yet.

Best regards.

FernandoGarcia, 2022-04-02:
Hi @me-no-dev!

Could you point me the way to find if this problem with invalid IP is coming from your library `WiFi.h`?

[Here](https://github.com/FernandoGarcia/esp8266-react/tree/make_library_2.0.x) you can find the framework I'm using.

The changes I did on framework to make it compatible with `Arduino-Espressif32 2.0.x` are [here](https://github.com/FernandoGarcia/esp8266-react/commit/810adcb305000fc6665689120416ea119867e5fd). 

The framework works fine with `Arduino-Espressif32 1.0.6`.

Thanks in advance.

me-no-dev, 2022-04-02:
@FernandoGarcia the information given is not enough. How do you start WiFi and how do you get that IP to the interface? You need to try minimal example using this repository and regular Arduino first, the see what is wrong. IP when we start WiFi here is fine.

me-no-dev, 2022-04-02:
You also have an exception up there and backtrace, but you forgot to decode it ;)

FernandoGarcia, 2022-04-02:
Hi!

Thanks for you answer.

I found [the line](https://github.com/FernandoGarcia/esp8266-react/blob/make_library_2.0.x/src/WiFiSettingsService.cpp#L74) that is causing the IP problem.

You can try reproduce the error with [HelloServer](https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/examples/HelloServer/HelloServer.ino) example and adding `WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);` before `WiFi.begin(ssid, password);`

Do you know why this line works in 1.06 but doesn't works in 2.0.x?

Best regards.

Jason2866, 2022-04-02:
Probably because to be align with arduino https://www.arduino.cc/en/Reference/wiFiConfig

closed 2022-04-05 by FernandoGarcia