espressif/arduino-esp32 #6462: Problem scrolling text on LCD using millis in v2.0.0 or later

mooballs, 2022-03-21:
### Board

ESP32 Dev Module

### Device Description

Standard 1602 LCD display with standard I2C convertor.

### Hardware Configuration

I2C  SDA=GPIO21, SCL=GPIO22

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

The text in the attached code will scroll across the screen if I use delay() but if I try millis() it just flickers as if its flying past very quickly (regardless of the value of milliseconds I use).

### Sketch

```cpp
#include "LiquidCrystal_I2C.h"

unsigned long times = millis();

int lcdColumns = 16;
int lcdRows = 2;

LiquidCrystal_I2C lcd(0x27, lcdColumns, lcdRows);

void setup() {

  lcd.init();
  Serial.begin(115200);
  lcd.backlight();
}

void loop()

{{

    {
      lcd.setCursor(0, 0);
      lcd.print("Hello");

      lcd.setCursor(0, 1);
      lcd.print("Mate");
    }
  }

{ for (int positionCounter = 0; positionCounter < 16; positionCounter++) {
      //      scroll one position left:

      if ( (millis() - times) >= 350) //Has one second passed?

      {

        lcd.scrollDisplayLeft();
        times = millis();           //and reset time.
      }
    }
  }

}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)configsip: 0, SPIWP:0xeeclk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00mode:DIO, clock div:1load:0x3fff0030,len:1324ho 0 tail 12 room 4load:0x40078000,len:13508load:0x40080400,len:3604entry 0x400805f0[��mum����2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
```

### Other Steps to Reproduce

 It compiles fine, but will not display properly on the LCD (I've tried I2c and connecting the LCD directly to the esp32 with the same results). millis() works fine in v1.0.6 but I need to use 2.0.2 as some of the features in the project I'm working on are dependant on the latest version. I've also tried several different Liquid Crystal libraries and another esp32 board with no luck. I can't find any mention of the issue online at all.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

LioRei, 2022-03-23:
/!\ there is a big difference between delay and millis. Delay is a blocking function where millis is not so your code has a problem. You need to block your execution until the good amount of time has passed.

To just block, you can do a while loop like this:

mooballs, 2022-03-23:
Thanks for having a look. I should have said, but it needs to be unblocking as there's other stuff going on in my main sketch. I tried the while () method and it still didn't work. 
Either way, the point is the code I've posted should work and it doesn't.
It works in v1.0.6, but not 2.x.x, which is why I've raised it as an issue. It works if I use delay() but not millis() (or micros, or esp_timer_get_time).
If somebody had five minutes to hook up an LCD1602 and try it I'd be most grateful. Or point me in the direction of what could possibly be messing it up (I'll try yet another board, but I keep coming back to fact that it used to work and now I've updated the arduino-esp32 version it doesn't).

me-no-dev, 2022-03-23:
I am closing this because it's a coding issue. Feel free to continue the discussion though :)
```cpp
void loop(){
  static int positionCounter = -1;

  if(positionCounter < 0){
    lcd.setCursor(0, 0);
    lcd.print("Hello");

    lcd.setCursor(0, 1);
    lcd.print("Mate");

    positionCounter = 0;
    times = millis();

  } else if((positionCounter >= 0) && ((millis() - times) >= 350)){
    lcd.scrollDisplayLeft();
    times = millis();
    positionCounter++;
    if(positionCounter == 16){
      positionCounter = -1;
    }
  }
}
```

closed 2022-03-23 by me-no-dev

mooballs, 2022-03-23:
Have you actually tried it with an LCD? Because your code does exactly the same as mine, which is to say, doesn't work.
As I say, it works in 1.0.6, but not in 2.x.x, so if its a coding issue, how could this be? (I'm not being rude, I genuinely want to know).

me-no-dev, 2022-03-23:
your code does the following thing:
- prints Hello Mate

- loops 16 times to check if time limit has been reached, which would never be, because they happen fast

- repeat

mooballs, 2022-03-23:
Thanks for that mate, that makes sense and I had wondered about that bit of code. 
BUT ; why would it work in an earlier version of arduino-esp32 (making me think the code was sound)?
plus I've tried your code and I'm getting the same result as mine.

me-no-dev, 2022-03-23:
the issue is probably i the library you are using. We require that Wire API is properly used. If you link me to the library, I can probably tell you what is wrong and how to fix it

mooballs, 2022-03-23:
Yeah man, its all set up correctly. Its so bizarre that it works in the earlier version but not in the later ones (both your code and, weirdly, mine). Thanks for helping anyway, I don't know what I'm going to do now but I'll keep digging.

me-no-dev, 2022-03-24:
it must be contrast issue. Library and code (my code) worked out of the box here. First I connected the LCD to 3.3V which was too low (later adjusting contrast made it show the text) and it worked out of the box on 5V. No changes to the library or anything.

me-no-dev, 2022-03-24:
Wish you luck!