espressif/arduino-esp32 #6465: lib-builder: review and extract non-chip-specific sdkconfig options into a common sdkconfig file

igrr, 2022-03-22:


VojtechBartoska, 2022-03-23:
Deleting Milestone, it will be done after releasing v2.0.3.

VojtechBartoska, 2022-03-31:
Adding it back to 2.0.3 milestone to not forget on it.

me-no-dev, 2022-04-11:
```
## esp32

CONFIG_BTDM_CTRL_MODE_BTDM=y
CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE=20
CONFIG_BT_BTC_TASK_STACK_SIZE=8192
CONFIG_BT_CLASSIC_ENABLED=y
CONFIG_BT_A2DP_ENABLE=y
CONFIG_BT_SPP_ENABLED=y
CONFIG_BT_HFP_ENABLE=y
CONFIG_BT_STACK_NO_LOG=y
CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST=y
CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=y
CONFIG_ESP32_SPIRAM_SUPPORT=y
CONFIG_ESP32_ULP_COPROC_ENABLED=y
CONFIG_ESP32_XTAL_FREQ_AUTO=y
CONFIG_ESP_INT_WDT_TIMEOUT_MS=300
# CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1 is not set
CONFIG_FREERTOS_FPU_IN_ISR=y
# CONFIG_USE_WAKENET is not set
# CONFIG_USE_MULTINET is not set

## esp32s2

CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP=y
CONFIG_ARDUINO_RUN_CORE0=y
CONFIG_ARDUINO_EVENT_RUN_CORE0=y
CONFIG_TINYUSB_MSC_BUFSIZE=4096
CONFIG_TINYUSB_DESC_DFU_RT_STRING="Espressif DFU Device"
# CONFIG_SPI_MASTER_ISR_IN_IRAM is not set
# CONFIG_SPI_SLAVE_ISR_IN_IRAM is not set
CONFIG_ESP32S2_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32S2_SPIRAM_SUPPORT=y
CONFIG_ESP32S2_KEEP_USB_ALIVE=y
CONFIG_ESP_HTTP_CLIENT_ENABLE_BASIC_AUTH=y
CONFIG_ESP_HTTPS_SERVER_ENABLE=y
CONFIG_ESP_INT_WDT_TIMEOUT_MS=1000
# CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0 is not set
CONFIG_ESP32_WIFI_CSI_ENABLED=y
# CONFIG_ESP32_WIFI_IRAM_OPT is not set
# CONFIG_ESP32_WIFI_RX_IRAM_OPT is not set
CONFIG_ESP_WIFI_FTM_ENABLE=y
# CONFIG_USE_WAKENET is not set
# CONFIG_USE_MULTINET is not set

## esp32s3

CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP=y
CONFIG_TINYUSB_MSC_BUFSIZE=4096
CONFIG_BT_BLE_42_FEATURES_SUPPORTED=y
CONFIG_BLE_MESH=y
# CONFIG_SPI_MASTER_ISR_IN_IRAM is not set
# CONFIG_SPI_SLAVE_ISR_IN_IRAM is not set
CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32S3_SPIRAM_SUPPORT=y
CONFIG_ESP32S3_RTC_CLK_CAL_CYCLES=576
CONFIG_ESP_HTTP_CLIENT_ENABLE_BASIC_AUTH=y
CONFIG_ESP_HTTPS_SERVER_ENABLE=y
CONFIG_ESP32S3_UNIVERSAL_MAC_ADDRESSES_TWO=y
# CONFIG_ESP_SLEEP_GPIO_RESET_WORKAROUND is not set
CONFIG_ESP_INT_WDT_TIMEOUT_MS=1000
# CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1 is not set
CONFIG_ESP32_WIFI_CSI_ENABLED=y
# CONFIG_ESP32_WIFI_IRAM_OPT is not set
# CONFIG_ESP32_WIFI_RX_IRAM_OPT is not set
CONFIG_ESP_WIFI_FTM_ENABLE=y
CONFIG_SR_WN_MODEL_WN8_QUANT=y
CONFIG_SR_WN_WN8_HIESP=y
CONFIG_SR_MN_ENGLISH=y
CONFIG_SR_MN_EN_MULTINET5_SINGLE_RECOGNITION_QUANT8=y

## esp32c3

CONFIG_ARDUINO_RUN_CORE0=y
CONFIG_ARDUINO_EVENT_RUN_CORE0=y
CONFIG_BT_BLE_BLUFI_ENABLE=y
CONFIG_BT_BLE_42_FEATURES_SUPPORTED=y
CONFIG_ESP32C3_RTC_CLK_CAL_CYCLES=576
CONFIG_ESP_HTTPS_SERVER_ENABLE=y
# CONFIG_ESP32_WIFI_IRAM_OPT is not set
# CONFIG_ESP32_WIFI_RX_IRAM_OPT is not set
CONFIG_ESP_WIFI_FTM_ENABLE=y
CONFIG_FATFS_LFN_HEAP=y
CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=2304
CONFIG_MBEDTLS_CAMELLIA_C=y

## common

CONFIG_BOOTLOADER_LOG_LEVEL_NONE=y
CONFIG_COMPILER_OPTIMIZATION_SIZE=y
CONFIG_COMPILER_CXX_EXCEPTIONS=y
CONFIG_COMPILER_STACK_CHECK_MODE_NORM=y
CONFIG_COMPILER_WARN_WRITE_STRINGS=y
CONFIG_HTTPD_MAX_REQ_HDR_LEN=1024
CONFIG_HTTPD_WS_SUPPORT=y
CONFIG_ESP_IPC_TASK_STACK_SIZE=1024
CONFIG_FMB_TIMER_PORT_ENABLED=y
CONFIG_FREERTOS_HZ=1000
CONFIG_LOG_DEFAULT_LEVEL_ERROR=y
# CONFIG_LOG_COLORS is not set
CONFIG_LWIP_MAX_SOCKETS=16
CONFIG_LWIP_DHCP_RESTORE_LAST_IP=y
CONFIG_LWIP_DHCP_OPTIONS_LEN=128
CONFIG_LWIP_SNTP_MAX_SERVERS=3
CONFIG_LWIP_DHCP_GET_NTP_SRV=y
CONFIG_MBEDTLS_PSK_MODES=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK=y

## xtensa

CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ARDUHAL_ESP_LOG=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=4096
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=0
CONFIG_ETH_SPI_ETHERNET_DM9051=y
CONFIG_ETH_SPI_ETHERNET_W5500=y
CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=2048
CONFIG_ESP_MAIN_TASK_STACK_SIZE=4096
CONFIG_ESP_TASK_WDT_PANIC=y
CONFIG_ESP_TIMER_TASK_STACK_SIZE=4096
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=8
CONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM=8
CONFIG_ESP32_WIFI_CACHE_TX_BUFFER_NUM=16
# CONFIG_ESP32_WIFI_ENABLE_WPA3_SAE is not set
CONFIG_FATFS_CODEPAGE_850=y
CONFIG_FATFS_LFN_STACK=y
# CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT is not set
# CONFIG_FREERTOS_ASSERT_ON_UNTESTED_FUNCTION is not set
CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y
CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=1024
CONFIG_HEAP_POISONING_LIGHT=y
CONFIG_LWIP_ETHARP_TRUST_IP_MAC=y
# CONFIG_LWIP_DHCP_DOES_ARP_CHECK is not set
CONFIG_LWIP_TCP_SYNMAXRTX=6
CONFIG_LWIP_TCP_MSS=1436
CONFIG_LWIP_TCP_RTO_TIME=3000
CONFIG_LWIP_TCPIP_TASK_STACK_SIZE=2560
CONFIG_LWIP_TCPIP_TASK_AFFINITY_CPU0=y
# CONFIG_MBEDTLS_ASYMMETRIC_CONTENT_LEN is not set
CONFIG_MBEDTLS_SSL_PROTO_DTLS=y
CONFIG_OPENSSL_ASSERT_DO_NOTHING=y
CONFIG_PTHREAD_TASK_STACK_SIZE_DEFAULT=2048
CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS=10
CONFIG_SPI_FLASH_ERASE_YIELD_TICKS=2
CONFIG_SPI_FLASH_WRITE_CHUNK_SIZE=4096

## ble

CONFIG_BT_ENABLED=y
```

igrr, 2022-05-04:
- [ ] Interrupt watchdog: CONFIG_ESP_INT_WDT_TIMEOUT_MS=300 on ESP32, =1000 on S2 and S3, and default (probably 300) on C3. I think it makes sense to set it same for all targets.
- [ ] Task watchdog: on ESP32 and ESP32-S3 it is disabled for core 1, for ESP32-S2 it is disabled for core 0. But for C3 it is not disabled. Probably the expected behavior is "disabled for the core where setup & loop run". In that case, we can make it dependent on whether the chip is single/dual core.
- [ ] CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP=y is set for S2 and S3, but not for ESP32 or C3. I think it's not a chip-dependent option, should be same for all chips.
- [ ] CONFIG_ESP_HTTP_CLIENT_ENABLE_BASIC_AUTH=y is set for S3 and S3, but not for ESP32 or C3. I think it's not a chip-dependent option, should be same for all chips.
- [ ] CONFIG_ESP_HTTPS_SERVER_ENABLE=y is set for all chips except C3. Also seems like a non-chip-specific option.
- [ ] CONFIG_FATFS_LFN_HEAP=y for C3 and CONFIG_FATFS_LFN_STACK=y for Xtensa. I think this should be same for all chips.
- [ ] CONFIG_MBEDTLS_CAMELLIA_C=y only for C3. Doesn't look chip dependent either. Either we need it everywhere or not.
- [ ] These options are xtensa-only but in fact they don't seem to be architecture dependent. Recommend making them common:
    ```
    CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
    CONFIG_ARDUHAL_ESP_LOG=y
    CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=4096
    CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
    CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=0
    CONFIG_ETH_SPI_ETHERNET_DM9051=y
    CONFIG_ETH_SPI_ETHERNET_W5500=y
    CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=2048
    CONFIG_ESP_MAIN_TASK_STACK_SIZE=4096
    CONFIG_ESP_TASK_WDT_PANIC=y
    CONFIG_ESP_TIMER_TASK_STACK_SIZE=4096
    CONFIG_FATFS_CODEPAGE_850=y
    CONFIG_FATFS_LFN_STACK=y
    # CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT is not set
    # CONFIG_FREERTOS_ASSERT_ON_UNTESTED_FUNCTION is not set
    CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y
    CONFIG_HEAP_POISONING_LIGHT=y
    CONFIG_LWIP_ETHARP_TRUST_IP_MAC=y
    # CONFIG_LWIP_DHCP_DOES_ARP_CHECK is not set
    CONFIG_LWIP_TCP_SYNMAXRTX=6
    CONFIG_LWIP_TCP_MSS=1436
    CONFIG_LWIP_TCP_RTO_TIME=3000
    CONFIG_LWIP_TCPIP_TASK_STACK_SIZE=2560
    CONFIG_LWIP_TCPIP_TASK_AFFINITY_CPU0=y
    # CONFIG_MBEDTLS_ASYMMETRIC_CONTENT_LEN is not set
    CONFIG_MBEDTLS_SSL_PROTO_DTLS=y
    CONFIG_OPENSSL_ASSERT_DO_NOTHING=y
    CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS=10
    CONFIG_SPI_FLASH_ERASE_YIELD_TICKS=2
    CONFIG_SPI_FLASH_WRITE_CHUNK_SIZE=4096
    ```

me-no-dev, 2022-05-04:
Note to me to also investigate https://github.com/espressif/esp32-arduino-lib-builder/pull/55

me-no-dev, 2022-05-04:
@igrr `CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP` actually caused issues on ESP32 and was recently disabled for it.

me-no-dev, 2022-05-09:
> Task watchdog: on ESP32 and ESP32-S3 it is disabled for core 1, for ESP32-S2 it is disabled for core 0. But for C3 it is not disabled. Probably the expected behavior is "disabled for the core where setup & loop run". In that case, we can make it dependent on whether the chip is single/dual core.

Unfortunately that requires to be able to "deselect" menuconfig option, which is not possible (unless we change how that is originally defined in IDF's Kconfig).

me-no-dev, 2022-05-09:
@igrr here is the final version. Please verify:
```
## esp32

CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=4096
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=0
CONFIG_BTDM_CTRL_MODE_BTDM=y
CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE=20
CONFIG_BT_ENABLED=y
CONFIG_BT_BTC_TASK_STACK_SIZE=8192
CONFIG_BT_CLASSIC_ENABLED=y
CONFIG_BT_A2DP_ENABLE=y
CONFIG_BT_SPP_ENABLED=y
CONFIG_BT_HFP_ENABLE=y
CONFIG_BT_STACK_NO_LOG=y
CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST=y
CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=y
CONFIG_ESP32_SPIRAM_SUPPORT=y
CONFIG_ESP32_ULP_COPROC_ENABLED=y
CONFIG_ESP32_XTAL_FREQ_AUTO=y
# CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1 is not set
CONFIG_FREERTOS_FPU_IN_ISR=y
# CONFIG_USE_WAKENET is not set
# CONFIG_USE_MULTINET is not set

## esp32s2

CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=4096
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=0
CONFIG_TINYUSB_MSC_BUFSIZE=4096
CONFIG_TINYUSB_DESC_DFU_RT_STRING="Espressif DFU Device"
CONFIG_ESP32S2_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32S2_SPIRAM_SUPPORT=y
CONFIG_ESP32S2_KEEP_USB_ALIVE=y
# CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0 is not set
CONFIG_ESP32_WIFI_CSI_ENABLED=y
CONFIG_ESP_WIFI_FTM_ENABLE=y
# CONFIG_USE_WAKENET is not set
# CONFIG_USE_MULTINET is not set

## esp32s3

CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=4096
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=0
CONFIG_TINYUSB_MSC_BUFSIZE=4096
CONFIG_BT_ENABLED=y
CONFIG_BT_BLE_42_FEATURES_SUPPORTED=y
CONFIG_BLE_MESH=y
CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32S3_SPIRAM_SUPPORT=y
CONFIG_ESP32S3_RTC_CLK_CAL_CYCLES=576
CONFIG_ESP32S3_UNIVERSAL_MAC_ADDRESSES_TWO=y
# CONFIG_ESP_SLEEP_GPIO_RESET_WORKAROUND is not set
# CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1 is not set
CONFIG_ESP32_WIFI_CSI_ENABLED=y
CONFIG_ESP_WIFI_FTM_ENABLE=y
CONFIG_SR_WN_MODEL_WN8_QUANT=y
CONFIG_SR_WN_WN8_HIESP=y
CONFIG_SR_MN_ENGLISH=y
CONFIG_SR_MN_EN_MULTINET5_SINGLE_RECOGNITION_QUANT8=y

## esp32c3

CONFIG_BT_ENABLED=y
CONFIG_BT_BLE_BLUFI_ENABLE=y
CONFIG_BT_BLE_42_FEATURES_SUPPORTED=y
CONFIG_ESP32C3_RTC_CLK_CAL_CYCLES=576
# CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0 is not set
CONFIG_ESP_WIFI_FTM_ENABLE=y
CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=2304

## common

CONFIG_ARDUHAL_ESP_LOG=y
CONFIG_BOOTLOADER_LOG_LEVEL_NONE=y
CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP=y
CONFIG_COMPILER_OPTIMIZATION_SIZE=y
CONFIG_COMPILER_CXX_EXCEPTIONS=y
CONFIG_COMPILER_STACK_CHECK_MODE_NORM=y
CONFIG_COMPILER_WARN_WRITE_STRINGS=y
CONFIG_ESP_HTTPS_SERVER_ENABLE=y
CONFIG_ESP_HTTP_CLIENT_ENABLE_BASIC_AUTH=y
CONFIG_ESP_INT_WDT_TIMEOUT_MS=300
CONFIG_ESP_IPC_TASK_STACK_SIZE=1024
CONFIG_ESP_MAIN_TASK_STACK_SIZE=4096
CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=2048
CONFIG_ESP_TASK_WDT_PANIC=y
CONFIG_ESP_TIMER_TASK_STACK_SIZE=4096
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=8
CONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM=8
CONFIG_ESP32_WIFI_CACHE_TX_BUFFER_NUM=16
# CONFIG_ESP32_WIFI_ENABLE_WPA3_SAE is not set
# CONFIG_ESP32_WIFI_IRAM_OPT is not set
# CONFIG_ESP32_WIFI_RX_IRAM_OPT is not set
CONFIG_ETH_SPI_ETHERNET_DM9051=y
CONFIG_ETH_SPI_ETHERNET_W5500=y
CONFIG_FATFS_CODEPAGE_850=y
CONFIG_FATFS_LFN_STACK=y
# CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT is not set
CONFIG_FMB_TIMER_PORT_ENABLED=y
CONFIG_FREERTOS_HZ=1000
# CONFIG_FREERTOS_ASSERT_ON_UNTESTED_FUNCTION is not set
CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y
CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=1024
CONFIG_HEAP_POISONING_LIGHT=y
CONFIG_HTTPD_MAX_REQ_HDR_LEN=1024
CONFIG_HTTPD_WS_SUPPORT=y
CONFIG_LOG_DEFAULT_LEVEL_ERROR=y
# CONFIG_LOG_COLORS is not set
CONFIG_LWIP_ETHARP_TRUST_IP_MAC=y
# CONFIG_LWIP_DHCP_DOES_ARP_CHECK is not set
CONFIG_LWIP_TCP_SYNMAXRTX=6
CONFIG_LWIP_TCP_MSS=1436
CONFIG_LWIP_TCP_RTO_TIME=3000
CONFIG_LWIP_TCPIP_TASK_STACK_SIZE=2560
CONFIG_LWIP_TCPIP_TASK_AFFINITY_CPU0=y
CONFIG_LWIP_MAX_SOCKETS=16
CONFIG_LWIP_DHCP_RESTORE_LAST_IP=y
CONFIG_LWIP_DHCP_OPTIONS_LEN=128
CONFIG_LWIP_SNTP_MAX_SERVERS=3
CONFIG_LWIP_DHCP_GET_NTP_SRV=y
CONFIG_MBEDTLS_PSK_MODES=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK=y
CONFIG_MBEDTLS_CAMELLIA_C=y
# CONFIG_MBEDTLS_ASYMMETRIC_CONTENT_LEN is not set
CONFIG_MBEDTLS_SSL_PROTO_DTLS=y
CONFIG_OPENSSL_ASSERT_DO_NOTHING=y
CONFIG_PTHREAD_TASK_STACK_SIZE_DEFAULT=2048
CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS=10
CONFIG_SPI_FLASH_ERASE_YIELD_TICKS=2
CONFIG_SPI_FLASH_WRITE_CHUNK_SIZE=4096
# CONFIG_SPI_MASTER_ISR_IN_IRAM is not set
# CONFIG_SPI_SLAVE_ISR_IN_IRAM is not set

```

me-no-dev, 2022-05-09:
[this](https://github.com/espressif/arduino-esp32/pull/6718) clears the necessity to select `CONFIG_ARDUINO_RUN_CORE0` and friends for single core chips

igrr, 2022-05-18:
@me-no-dev Sorry I didn't notice your comment. The list looks good to me, I didn't spot any issue.

If you'd like to simplify things even further, you can move some chip-specific settings to common ones, like for example `CONFIG_ESP_WIFI_FTM_ENABLE=y` or ``CONFIG_TINYUSB_MSC_BUFSIZE=4096`. On chips where these options don't exist there will be just a warning printed by kconfiglib, and the setting will be ignored.