espressif/arduino-esp32 #6466: Compile error when use HTTPClient.h with SKD builded by esp32 arduino lib builder

lexfee, 2022-03-23:
### Board

ESP32 WROOM32-E

### Device Description

esp32dev

### Hardware Configuration

N/A

### Version

v2.0.2

### IDE Name

Plaatformio

### Operating System

Windows 10

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I was trying to use [esp32-arduino-lib-builder](https://github.com/espressif/esp32-arduino-lib-builder) to build a customer Arduino SDK. However, the project can not build if I use the SDK built by the lib-builder along with HTTPClient.h.  I use the BasicHttpClient example below and get a bunch of errors about `undefined reference to `mbedtls_ssl_******'`. Regarding the esp32-arduino-lib-builder, I changed nothing, just use the default sdkconfig setting, and use `./build.sh` to build, and copy the files from "out" to ".platformio\packages\framework-arduinoespressif32". I can build other examples, e.g. [WebUpdate.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/examples/WebUpdate/WebUpdate.ino). So far, it seems the problem only happens when using HTTPClient.h along with customer SDK from lib builder. the project can be built if I switch back to the original arduino-esp32 SDK. 

I also tried to copy the original sdkconfig from arduino-esp32/tools/sdk/esp32/ and replace the sdkconfig.esp32, it won't work. Still got the same error.  Any settings I need to turn on? 

### Sketch

```cpp
/**
 * BasicHTTPClient.ino
 *
 *  Created on: 24.05.2015
 *
 */

#include <Arduino.h>

#include <WiFi.h>
#include <WiFiMulti.h>

#include <HTTPClient.h>

#define USE_SERIAL Serial

WiFiMulti wifiMulti;

/*
const char* ca = \
"-----BEGIN CERTIFICATE-----\n" \
"MIIEkjCCA3qgAwIBAgIQCgFBQgAAAVOFc2oLheynCDANBgkqhkiG9w0BAQsFADA/\n" \
"MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n" \
"DkRTVCBSb290IENBIFgzMB4XDTE2MDMxNzE2NDA0NloXDTIxMDMxNzE2NDA0Nlow\n" \
"SjELMAkGA1UEBhMCVVMxFjAUBgNVBAoTDUxldCdzIEVuY3J5cHQxIzAhBgNVBAMT\n" \
[... 165 lines cut ...]
c:/users/deary/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/deary/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:246: undefined reference to `mbedtls_ssl_get_verify_result'
c:/users/deary/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib8d8\libWiFiClientSecure.a(ssl_client.cpp.o): in function `stop_ssl_socket(sslclient_context*, char const*, char const*, char const*)':
C:/Users/deary/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:292: undefined reference to `mbedtls_ssl_free'
c:/users/deary/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/deary/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:293: undefined reference to `mbedtls_ssl_config_free'
c:/users/deary/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib8d8\libWiFiClientSecure.a(ssl_client.cpp.o): in function `data_to_read(sslclient_context*)':
C:/Users/deary/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:294: undefined reference to `mbedtls_ssl_read'
c:/users/deary/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/deary/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:300: undefined reference to `mbedtls_ssl_get_bytes_avail'
c:/users/deary/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib8d8\libWiFiClientSecure.a(ssl_client.cpp.o): in function `send_ssl_data(sslclient_context*, unsigned char const*, unsigned int)':
C:/Users/deary/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:367: undefined reference to `mbedtls_ssl_write'
c:/users/deary/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib8d8\libWiFiClientSecure.a(ssl_client.cpp.o): in function `get_ssl_receive(sslclient_context*, unsigned char*, int)':
C:/Users/deary/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:384: undefined reference to `mbedtls_ssl_read'
c:/users/deary/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib8d8\libWiFiClientSecure.a(esp_crt_bundle.c.o):(.literal.esp_crt_bundle_attach+0x10): undefined reference to `mbedtls_ssl_conf_ca_chain'
c:/users/deary/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib8d8\libWiFiClientSecure.a(esp_crt_bundle.c.o):(.literal.esp_crt_bundle_attach+0x14): undefined reference to `mbedtls_ssl_conf_verify'
c:/users/deary/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib8d8\libWiFiClientSecure.a(esp_crt_bundle.c.o): in function `esp_crt_bundle_attach':
C:/Users/deary/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/esp_crt_bundle.c:196: undefined reference to `mbedtls_ssl_conf_ca_chain'
c:/users/deary/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/deary/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/esp_crt_bundle.c:197: undefined reference to `mbedtls_ssl_conf_verify'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
========================================================= [FAILED] Took 27.03 seconds =========================================================
The terminal process "C:\Users\deary\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-03-23:
the errors point that the linker was not able to find the functions. It's not a preprocessor error, so you can not fix it by include :)

The issue points that you either had mbedtls disabled in the lib-builder or that the lib did not copy over (was not found).

Jason2866, 2022-03-23:
You should use the latest available for platformio (....patch2)
And with eesptool v3.1 you will have issues too (linking).
Espressif just released esptool.py v.3.3 today

lexfee, 2022-03-24:
I don't know which setting controls the mbedtls on and off, there are lots of setting related to mbedtls. But the thing is I copied the original sdkconfig, I assume all important settings should be on by default. I check the files inside the tools/sdk/esp32, the total file number is the same as the original SDK. 

I tried using the latest esptool 3.3 but the result is the same. The toolchains used is "8.4.0+2021r1".

Anything else I should try?

Jason2866, 2022-03-24:
@lexfee The Platformio platform is outdated. Try
```
https://github.com/Jason2866/platform-espressif32.git
```

lexfee, 2022-03-25:
I think I identified the broken file.  It is the `libmbedtls.a` causing the problem.  The file created by lib-builder is only 104kb, while the original file is 1203kb. If I replace this file with the original one,  all of the problems are gone. The project can be built, and my code is working. So the lib-builder did create the `libmbedtls.a`  file but somehow in a much smaller size.  There is no error when I use `./build.sh`. 

The esp-idf was installed using `./tools/install-esp-idf.sh`. I checked the files inside esp-idf/components/mbedtls, to me, it seems all the files are there, including a submodule ([mbedtls](https://github.com/espressif/mbedtls)).

[image]

Jason2866, 2022-03-25:
@lexfee So "something" with your Lib builder setup is wrong. The Github Actions lib builder is generating the file `libmbedtls.a` correctly.

lexfee, 2022-03-26:
The only setting would be `menuconfig` which I did not touch yet.  The installation of modules and submodules are all done automatically. I am using WSL 2,  so I also tried to reset the Ubuntu-18.04 and start over., which did not help.  Any setting for mbedtls I should check? It is looks like all files are good except the `libmbedtls.a`. Any log I should check?

Jason2866, 2022-03-27:
Which branch do you use from the Arduino lib builder?
The builds i make are from branch `/esp32-s3-support`. Not saying master branch is not working (i just do not use to make builds from, so idk).
Tbh, no idea what to check. I use github actions for building the framework.
Maybe this is easier for you too. You can just clone my fork. After a successfull run a complete framework is released. [https://github.com/Jason2866/esp32-arduino-lib-builder](https://github.com/Jason2866/esp32-arduino-lib-builder)

closed 2022-04-20 by VojtechBartoska

my-dudhwala, 2023-08-28:
same problem occurred again here...

Zontex, 2024-06-13:
Same problem here