espressif/arduino-esp32 #6467: How to control LED of (PHY) RTL8201

savagerex, 2022-03-23:
### Related area

Peripheral

### Hardware specification

ESP32

### Is your feature request related to a problem?

please help to esp32 control   LED of (PHY) RTL8201

in the core 2.0.2. we found the file (eth_phy_regs_struct.h).

it's seem to control LED. right ???

### Describe the solution you'd like

we want to control  LED of (PHY) RTL8201

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-03-25:
Hello @savagerex,

thanks for your contribution. The request is now being discussed internally. I will let you now when we have some resolution.

david-cermak, 2022-04-01:
Hi @savagerex 

Yes, you're correct, the header file [eth_phy_regs_struct.h](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/include/eth_phy_regs_struct.h) contains register definitions, but only the generic ones, those that are mostly used for all supported phy's. You can check specific driver's sources for other definitions, specific to each device. As for the RTL802, it's located in the `src`, but doesn't include many:

https://github.com/espressif/esp-idf/blob/36b83fcb9811437102a5b6b156221eeffe1e8fd7/components/esp_eth/src/esp_eth_phy_rtl8201.c#L24-L48

Please, refer to the datasheet of the RTL802 and check for the available functionality to setup the control LEDs. (seems like the regs 18, 19, pg 7 would do).
Unfortunately, we don't support writing to these registers from user code, so I'd suggest starting with some simple IDF example to play with the registers directly.

Then, It is possible to create a custom driver, something similar to this [esp_idf_eth_w6100](https://github.com/david-cermak/MadsHHLunds_esp_idf_eth_w6100) and work with it directly from user code without any kind of modification in SDKs.

VojtechBartoska, 2022-04-06:
Hi @savagerex, was the last comment from David helpful?

savagerex, 2022-04-07:
OK!! THX

VojtechBartoska, 2022-04-21:
Closing as solved.

closed 2022-04-21 by VojtechBartoska