espressif/arduino-esp32 #6468: assertion "heap != NULL && "free() target pointer is outside heap areas" in workd Bluetooth BLE

denisalexee, 2022-03-23:
### Board

Chip is ESP32-D0WD (revision 1) Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None

### Device Description

other

### Hardware Configuration

GPIO 18&& 19 in I2C

### Version

other

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

in connect BLE assertion "heap != NULL && "free() target pointer is outside heap areas"

### Sketch

```cpp
#include <Wire.h>
#include "SparkFunHTU21D.h"
#include "MAX30105.h"           // Библиотека пульсометра

#include <GyverOLED.h>          // Библиотека экрана
#include <Rtc_Pcf8563.h>        // Библиотека часов реального времени
#include "MPU6050.h"            // Библиотека гироскоп
#include "heartRate.h"
#include "spo2_algorithm.h"  
#include <FS.h>                // Доступ к файловой системе
#include <SPIFFS.h>

// bluetooth
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>
//----------------bitmap--------------------------
#include "Still.h"
#include "batt.h"
#include "Bluetooth.h"
#include "daw_x.h"
#include "heart.h"
#include "mist_two.h"
#include "vibr.h"

HTU21D myHumidity;
[... 918 lines cut ...]
```

### Debug Message

```plain
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c", line 267, function: heap_caps_free
abort() was called at PC 0x400dec0b on core 1
ELF file SHA256: 0000000000000000
Backtrace: 0x4008f5bc:0x3ffc7bb0 0x4008f839:0x3ffc7bd0 0x400dec0b:0x3ffc7bf0 0x400821af:0x3ffc7c20 0x40086da5:0x3ffc7c40 0x4000bec7:0x3ffc7c60 0x400dbee7:0x3ffc7c80 0x400dbef5:0x3ffc7ca0 0x400d2dc2:0x3ffc7cc0 0x400dc51c:0x3ffc7d60 0x400908ae:0x3ffc7d80

0x4008f5bc: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 715
0x4008f839: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 715
0x400decab: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)
0x400821af: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 215
0x40086da5: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 42
0x400dbf87: String::invalidate() at C:\Users\alekseev_da.NNIIRT\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/WString.cpp line 859
0x400dbf95: String::~String() at C:\Users\alekseev_da.NNIIRT\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/WString.cpp line 859
0x400d2e62: loop() at D:\work\ESP\clock\clock/clock.ino line 848
0x400dc5bc: loopTask(void*) at C:\Users\alekseev_da.NNIIRT\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/main.cpp line 23
0x400908ae: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 355 (discriminator 1)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-03-23:
There are 2 places in your loop function where you write outside of the address limits.
```c
char x[1];
dtostrf(temperature, 5/*Полная_длина_строки*/, 1/*Количество_символов_после_запятой*/,x);

char ax[1];
sprintf(ax, "%d", minTempPar);
```

In both cases you write way more than 1 character (and terminating zero) to those arrays. Fix that first :)

denisalexee, 2022-03-25:
char x[8];                                                   //
	  dtostrf(temperature, 5, 1,x);
    pTxTimeCharacteristic ->setValue((std::string)x);
    pTxTimeCharacteristic ->notify();
   
    char ax[8];                                                   //

   sprintf(ax, "%d", minTempPar);
   pTxminTempParCharacteristic ->setValue((std::string)ax);
   pTxminTempParCharacteristic ->notify();

denisalexee, 2022-03-25:
thanks for helping

closed 2022-03-25 by denisalexee