espressif/arduino-esp32 #6475: SD card read/write extremely slow with version 2.0.2

boeckhoff, 2022-03-23:
### Board

Adafruit ESP-32 Feather

### Device Description

Feather is connected to a SD card module through jumper wires 

### Hardware Configuration

nothing

### Version

v2.0.2

### IDE Name

Arduino IDE (up to date)

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

When using the new version of the core (2.0.2) I get extremely slow SD card read/write speeds (1048576 bytes read for 829442 ms !!). With the same hardware and the 1.0.6 core I get 2640 ms for the same data read.

With the same hardware but using version 1.0.6 I get

```
SD Card Type: SDHC
SD Card Size: 15193MB
Listing directory: /
  DIR : System Volume Information
  FILE: test.txt  SIZE: 1048576
  FILE: foo.txt  SIZE: 13
  FILE: wavfile.wav  SIZE: 94182
Creating Dir: /mydir
Dir created
Listing directory: /
  DIR : System Volume Information
  FILE: test.txt  SIZE: 1048576
  FILE: foo.txt  SIZE: 13
  DIR : mydir
  FILE: wavfile.wav  SIZE: 94182
Removing Dir: /mydir
Dir removed
Listing directory: /
  DIR : System Volume Information
Listing directory: System Volume Information
[  6089][E][vfs_api.cpp:27] open(): System Volume Information does not start with /
Failed to open directory
  FILE: test.txt  SIZE: 1048576
  FILE: foo.txt  SIZE: 13
  FILE: wavfile.wav  SIZE: 94182
Writing file: /hello.txt
File written
Appending to file: /hello.txt
Message appended
Reading file: /hello.txt
Read from file: Hello World!
Deleting file: /foo.txt
File deleted
Renaming file /hello.txt to /foo.txt
File renamed
Reading file: /foo.txt
Read from file: Hello World!
1048576 bytes read for 2640 ms
1048576 bytes written for 8532 ms
```

### Sketch

```cpp
[SD-Test sketch for ESP32](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/examples/SD_Test/SD_Test.ino)
```

### Debug Message

```plain
With Core@2.0.2

[..other info..]
1048576 bytes read for 829442 ms
1048576 bytes written for 216050 ms
```

With Core@1.0.6

```
[..other info..]
1048576 bytes read for 2640 ms
1048576 bytes written for 9652 ms
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-03-23:
Duplicate of https://github.com/espressif/arduino-esp32/issues/6295. Should be fixed in arduino-esp32 v2.0.3. See also workarounds in https://github.com/espressif/arduino-esp32/issues/6295.  v2.0.3 should be released soon.

Duplicate https://github.com/espressif/arduino-esp32/issues/6338

VojtechBartoska, 2022-03-24:
closing as the solution, thanks for linking a solution @mrengineer7777!

closed 2022-03-24 by VojtechBartoska