espressif/arduino-esp32 #6476: UART send data from buffer twice when read is called before print.

MiyukiDark, 2022-03-23:
### Board

ESP32-DEVKITC-32UE

### Device Description

ESP32-DEVKITC-32UE module

### Hardware Configuration

UART/Serial 0 and 2
WiFi

### Version

v2.0.2

### IDE Name

Arduino IDE 2.0 + PlatformIO

### Operating System

Windows 10 

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I already started an issue in ESP-IDF https://github.com/espressif/esp-idf/issues/8635

But it might be Arduino specific issue.
Data from UART2 are sent twice when I call Serial2.readString();  before.

It behaves the same on old 1.0.3 and new 2.0.2

With code as shown and with a received message in the buffer is this "#APP:cw:data?\n" sent twice with about 1ms pause

### Sketch

```cpp
#include <WiFi.h>
#include <WiFiUdp.h>

#include <Preferences.h>

static Preferences configuration_storage;
static String ssid_used;
static String pswd_used;

static WiFiUDP udp_to_SW;
static WiFiUDP udp_from_Sensor;
static WiFiUDP udp_from_SensorPing;

void MainSenderTask(void * parameter)
{
  TickType_t  xLastWakeTime = xTaskGetTickCount();
  Serial.println("Enter Main sender Task");
  for( ;; )
  {
    //run this task every x ms 
    vTaskDelayUntil(&xLastWakeTime, pdMS_TO_TICKS(1000));

    //Send a packet
    udp_to_SW.beginPacket(udpBCast, udpForSW);
    if (Serial2.available() > 0)
    {
      String str = Serial2.readString(); 
      int str_len = str.length() + 1;
      char char_array[str_len];
      str.toCharArray(char_array, str_len);
      udp_to_SW.printf(char_array, str.length()); 
    }
    else
    {
      udp_to_SW.printf("empty");
    }
    udp_to_SW.endPacket();

    //ping sensors
    udp_from_SensorPing.beginPacket(udpBCast, udpForSensorPing);
    udp_from_SensorPing.printf("ping");
    udp_from_SensorPing.endPacket();

    //request data from main app
    Serial2.print("#APP:cw:data?\n");

    // If packet received...
    char packet[255];
    int packetSize = udp_to_SW.parsePacket();
    if (packetSize) 
    {
      int len = udp_to_SW.read(packet, 255);
      if (len > 0)
      {
        packet[len] = '\0';
      }
      //send received packet to main app
      Serial2.println(packet);
      //copy received packet to debug output
      Serial.print("Packet from APP received: ");
      Serial.println(packet);
    }
  }
}

void mode_WithPC_setup()
{
  configuration_storage.begin(NVM_Name, false);
  ssid_used = configuration_storage.getString("ssid", default_ssid);
  pswd_used = configuration_storage.getString("pswd", default_pswd);
  configuration_storage.end();

  Serial.printf("Starting WiFi: %s@%s \n", ssid_used.c_str(), pswd_used.c_str());

  WiFi.mode(WIFI_AP);
  WiFi.softAP(ssid_used.c_str());
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));

  Serial.printf("TX power: %i \n", WiFi.getTxPower());

  udp_to_SW.begin(udpForSW);
  udp_from_Sensor.begin(udpForSensors);
  udp_from_SensorPing.begin(udpForSensorPing);

  xTaskCreate(
    MainSenderTask,    // Function that should be called
    "PC communication Task",   // Name of the task (for debugging)
    2000,            // Stack size (bytes)
    NULL,            // Parameter to pass
    1,               // Task priority
    NULL             // Task handle
  );
}

void setup() {
  //debug meessage uart
  Serial.begin(115200);
  //communicate with main APP
  Serial2.begin(115200);

  mode_WithPC_setup();
}

void loop() {

}
```

### Debug Message

```plain
Nothing related in debug
```

### Other Steps to Reproduce

Must have a message in receive buffer to behave this way.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

MiyukiDark, 2022-03-24:
It is way stranger than I expected. 
It might be some other issue because the code actually **executes twice** the whole task 
This is way too strange
So where the hell is the problem. It is a simple piece of code. How can it execute twice? 

```cpp
void MainSenderTask(void * parameter)
{
  TickType_t  xLastWakeTime = xTaskGetTickCount();
  Serial.println("Enter Main sender Task");
  for( ;; )
  {
    //run this task every x ms 
    vTaskDelayUntil(&xLastWakeTime, pdMS_TO_TICKS(1000));

    uint8_t run_countert;

    //Send a packet
    udp_to_SW.beginPacket(udpBCast, udpForSW);
    if (Serial2.available() > 0)
    {
      String str = Serial2.readString(); 
      int str_len = str.length() + 1;
      char char_array[str_len];
      str.toCharArray(char_array, str_len);
      udp_to_SW.printf(char_array, str.length()); 
      Serial.println(str.c_str());
    }
    else
    {
      udp_to_SW.printf("empty");
      Serial.printf("empty\n");
    }
    udp_to_SW.endPacket();

    //ping sensors
    udp_from_SensorPing.beginPacket(udpBCast, udpForSensorPing);
    udp_from_SensorPing.printf("ping");
    udp_from_SensorPing.endPacket();

    run_countert++;
    Serial.println(run_countert);

    //request data from main app
    Serial2.print("#APP:cw:data?\n");

    // If packet received...
    char packet[255];
    int packetSize = udp_to_SW.parsePacket();
    if (packetSize) 
    {
      int len = udp_to_SW.read(packet, 255);
      if (len > 0)
      {
        packet[len] = '\0';
      }
      //send received packet to main app
      Serial2.println(packet);
      //copy received packet to debug output
      Serial.print("Packet from APP received: ");
      Serial.println(packet);
    }
  }
}
```

And this is the result every second

```
197
2:0:1:0:1:0:1:0:1:0:0:0:0:1
198
empty
```

VojtechBartoska, 2022-03-24:
Hi @MiyukiDark, thanks for opening an issue.

@SuGlider can you please take a look? thanks!

SuGlider, 2022-03-25:
@MiyukiDark 
The sketch you posted can't be compiled. It seems to be part of a bigger sketch.
There are missing parts. Please provide some simpler sketch that reproduces the issue.

One possible reason for this issue would be if there are two '\n' or a '\r\n' at the end of the input from Serial2.
It may lead to a ```String``` correct read and then in the next ```run_countert``` it outputs nothing because it is just a '\n' or '\r'.

As alternative, please try this piece of code to replace your original code and report back with results.
Thanks.

### Note:

Are you sure that ```udp_to_SW.printf(char_array, length);``` is correct?
```printf()``` takes ```char_array``` as format and ```length``` as parameter to be printed.

```printf(const char *format, ...)```
```2:0:1:0:1:0:1:0:1:0:0:0:0:1``` from your output example doesn't seem to be a valid ```printf() formatting```, if it is indeed desired to send ```length```.

```cpp
    //Send a packet
    udp_to_SW.beginPacket(udpBCast, udpForSW);
    if (Serial2.available() > 0)
    {
      char char_array[256];  // assumption: it will never exceed 255 characters, based on the example output
      uint8_t length = 0;
      while (Serial2.avaiable() && length < 255) {
        char ch = Serial2.read();
        Serial.printf("%x ", ch); // checking byte code read...
        char_array[length++] = ch;
      }
      char_array[length] = '\0';
      Serial.println();
      
      udp_to_SW.printf(char_array, length); 
      Serial.println(char_array);
    }
    else
    {
      udp_to_SW.printf("empty");
      Serial.printf("empty\n");
    }
    udp_to_SW.endPacket();
```

MiyukiDark, 2022-03-25:
Here is a compilable sketch

UDP printf is not a standard a printf it has it own format 

```cpp
#include <WiFi.h>
#include <WiFiUdp.h>

#define apIP IPAddress(8, 8, 4, 4) // The default android DNS
#define udpBCast  "8.8.4.255"

#define udpForSW  11611
#define udpForSensors  11612
#define udpForSensorPing  11613
#define udpForPairing 11614

//The udp library class
static WiFiUDP udp_to_SW;
static WiFiUDP udp_from_Sensor;
static WiFiUDP udp_from_SensorPing;

void MainSenderTask(void * parameter)
{
  TickType_t  xLastWakeTime = xTaskGetTickCount();
  Serial.println("Enter Main sender Task");
  for( ;; )
  {
    //run this task every x ms 
    vTaskDelayUntil(&xLastWakeTime, pdMS_TO_TICKS(1000));

    static uint8_t run_counter = 0;

    //Send a packet
    udp_to_SW.beginPacket(udpBCast, udpForSW);
    if (Serial2.available() > 0)
    {
      String str = Serial2.readString(); 
      int str_len = str.length() + 1;
      char char_array[str_len];
      str.toCharArray(char_array, str_len);
      udp_to_SW.printf(char_array, str.length()); 
      Serial.println(char_array);
    }
    else
    {
      udp_to_SW.printf("empty");
      Serial.println("empty");
    }
    udp_to_SW.endPacket();

    //ping sensors
    udp_from_SensorPing.beginPacket(udpBCast, udpForSensorPing);
    udp_from_SensorPing.printf("ping");
    udp_from_SensorPing.endPacket();

    //request data from main app
    Serial2.print("#APP:cw:data?\n");

    run_counter++;
    Serial.println(run_counter);

    // If packet received...
    char packet[255];
    int packetSize = udp_to_SW.parsePacket();
    if (packetSize) 
    {
      int len = udp_to_SW.read(packet, 255);
      if (len > 0)
      {
        packet[len] = '\0';
      }
      //send received packet to main app
      Serial2.println(packet);
      //copy received packet to debug output
      Serial.print("Packet from APP received: ");
      Serial.println(packet);
    }
  }
}

void setup()
{
   //debug meessage uart
  Serial.begin(115200);
  //communicate with main APP
  Serial2.begin(115200);

  WiFi.mode(WIFI_AP);
  WiFi.softAP("test_wifi");
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));

  Serial.println("ESP-IDF version is: " + String(esp_get_idf_version()));

  udp_to_SW.begin(udpForSW);
  udp_from_Sensor.begin(udpForSensors);
  udp_from_SensorPing.begin(udpForSensorPing);

  xTaskCreate(
    MainSenderTask,    // Function that should be called
    "PC communication Task",   // Name of the task (for debugging)
    2000,            // Stack size (bytes)
    NULL,            // Parameter to pass
    1,               // Task priority
    NULL             // Task handle
  );
}

void loop() {

}

```
Here is the what happens on port 
[image]

and on Serial output is this every second 
```
2:0:1:0:1:0:1:0:1:0:0:0:0:1
6
empty
7
```

MiyukiDark, 2022-03-25:
When the other side did not reply (nothing is in the receive buffer to process) it work just right 
[image]

SuGlider, 2022-03-25:
@MiyukiDark 
I tested a simplified (without UDP) code and it seems to run OK... 
I have no idea why you may have 2 tasks running. 
I think you may need to debug your code further.

SuGlider, 2022-03-25:
@MiyukiDark 

The issue is in ```vTaskDelayUntil(&xLastWakeTime, pdMS_TO_TICKS(1000));```
Given that ```xLastWakeTime``` is never updated, it has no effect and doesn't block for 1 second.

https://www.freertos.org/vtaskdelayuntil.html

MiyukiDark, 2022-03-25:
> @MiyukiDark
> 
> The issue is in `vTaskDelayUntil(&xLastWakeTime, pdMS_TO_TICKS(1000));` Given that `xLastWakeTime` is never updated, it has no effect and doesn't block for 1 second.
> 
> https://www.freertos.org/vtaskdelayuntil.html

@[SuGlider](https://github.com/SuGlider)

No, it isn't
It is **exactly what is in the example** 
And it is called once per second 
It is just a pointer, vTaskDelayUntil updates it

MiyukiDark, 2022-03-25:
Sorry I did close instead of just comment

SuGlider, 2022-03-25:
@MiyukiDark 

I need a code example that demonstrates that the issue you reported is related to the ESP32 Arduino Core.

Please send us the smaller/simpler code that can be used to validate the issue.

You application has not related components such Wireless communication, message exchange and a state machine that may be also related to the issue.

I run some testing cases and I can't reproduce the issue within a simplified sketch that uses UART and Strings.

Thanks.

MiyukiDark, 2022-03-25:
@SuGlider [SuGlider](https://github.com/SuGlider)

Here is just the core of it with just Serials and it behaves the same 
When something is received it executes twice 
Without incoming message behaves as expected

```cpp
void MainSenderTask(void * parameter)
{
  TickType_t  xLastWakeTime = xTaskGetTickCount();
  Serial.println("Enter Main sender Task");
  for( ;; )
  {
    //run this task every x ms 
    vTaskDelayUntil(&xLastWakeTime, pdMS_TO_TICKS(1000));

    static uint8_t run_counter = 0;

    //Send a packet
    if (Serial2.available() > 0)
    {
      String str = Serial2.readString(); 
      int str_len = str.length() + 1;
      char char_array[str_len];
      str.toCharArray(char_array, str_len); 
      Serial.println(char_array);
    }
    else
    {
      
      Serial.println("empty");
    }

    //request data from main app
    Serial2.print("#APP:cw:data?\n");

    run_counter++;
    Serial.println(run_counter);
  }
}

void setup()
{
   //debug meessage uart
  Serial.begin(115200);
  //communicate with main APP
  Serial2.begin(115200);

  Serial.println("ESP-IDF version is: " + String(esp_get_idf_version()));

  xTaskCreate(
    MainSenderTask,    // Function that should be called
    "PC communication Task",   // Name of the task (for debugging)
    2000,            // Stack size (bytes)
    NULL,            // Parameter to pass
    1,               // Task priority
    NULL             // Task handle
  );
}

void loop() {

}
```

ESP-IDF version is: v4.4-beta1-189-ga79dc75f0a
ESP32-arduino 2.0.2

SuGlider, 2022-03-25:
@MiyukiDark 

This is what I can see with this latest sketch:

#### Using:
ESP-IDF version is: v4.4-beta1-189-ga79dc75f0a
ESP32-arduino 2.0.2

1) When no "2:0:1:0:1:0:1:0:1:0:0:0:0:1" is sent through Serial2, everything goes well. Every 1 second a number and "empty" goes to Serial and "#APP:cw:data?" goes to Serial2.
2) When I send "2:0:1:0:1:0:1:0:1:0:0:0:0:1" to Serial2, it delays more than 1 second and it seems like there is a duplicated output as you describe, which seems like both UARTs output are frozen for 2 seconds and then everything goes to output all at the same time.

## Solution:

This is not an issue, but actually a feature of Arduino Stream Class.

Arduino Stream Class (inherited by UART) has a timeout when reading streams of data, such as Strings.
Default timeout is 1 second, which explains the issue you reported.
Clear this timeout and everything will work as desired.
```Serial2.setTimeout(0);```

Reference:
https://www.arduino.cc/reference/en/language/functions/communication/stream/streamsettimeout/

``` cpp
void setup()
{
   //debug meessage uart
  Serial.begin(115200);
  //communicate with main APP
  Serial2.begin(115200);

  // Set 0 to Strem Class timeout used in UART when reading Strings 
  Serial2.setTimeout(0);

  Serial.println("ESP-IDF version is: " + String(esp_get_idf_version()));

  xTaskCreate(
    MainSenderTask,    // Function that should be called
    "PC communication Task",   // Name of the task (for debugging)
    2000,            // Stack size (bytes)
    NULL,            // Parameter to pass
    1,               // Task priority
    NULL             // Task handle
  );
}
```

closed 2022-03-25 by SuGlider

MiyukiDark, 2022-03-25:
[SuGlider](https://github.com/SuGlider)

Thank you, with  Serial2.setTimeout(0); it works fine 

Just how can 1 second wait cause duplicate output 
It is not single duplicity, but duplicity on both ports for all the time 
I can understand it will cause it for the first time to shift this message 
But when it gets a reply every second it should behave fine just delayed 
Or something other happens?

SuGlider, 2022-03-25:
> Just how can 1 second wait cause duplicate output It is not single duplicity, but duplicity on both ports for all the time I can understand it will cause it for the first time to shift this message But when it gets a reply every second it should behave fine just delayed Or something other happens?

Actually there is no duplicate output. This is the sequence:

1)  waits 1 second and sete xLastWakeTime to NOW.
    //run this task every x ms 
    vTaskDelayUntil(&xLastWakeTime, pdMS_TO_TICKS(1000));

2) Receives the "packet" on Serial2.  It holds 1 second because of ```Stream::setTimeout(1000)``` as default.
3) After this 1 second, xLastWakeTime is already late...
4) Outputs on Serial the "packet"  ---> "2:0:1:0:1:0:1:0:1:0:0:0:0:1"
5) Send the request -- Ouputs  on Serial2 "#APP:cw:data?\n"
6) Prints on Serial the Couter
7) Restart the LOOP (for())
8) runs vTaskDelayUntil(&xLastWakeTime, pdMS_TO_TICKS(1000)); -----> BUT it doesn't wait because as (3) xLastWakeTime it late more than 1 second  ----> NO EFFECT HERE, NO TIME TO WAIT
9) repeats (2), (3), ....

MiyukiDark, 2022-03-26:
[SuGlider](https://github.com/SuGlider)
Oh I see
It results in 2 second interval.
I should have seen it in the first place.
It will be obvious 🤦 
Thanks for your help 

Anyway, it brings up a question. Shouldn't there be a warning in references or somewhere it has this feature default on? 
I think it can cause some confusion.

SuGlider, 2022-03-26:
>Anyway, it brings up a question. Shouldn't there be a warning in references or somewhere it has this feature enabled by default?
>I think it can cause some confusion.

Actually Arduino Stream is as old as Arduino itself.

This is the original Arduino API. 
It is documented in Arduino Site.

ESP32 Arduino just follows Arduino API.

Reference:

https://www.arduino.cc/reference/en/language/functions/communication/stream/

https://www.arduino.cc/reference/en/language/functions/communication/stream/streamsettimeout/

MiyukiDark, 2022-03-26:
Sorry I do not know where I was looking 
It is clearly there