espressif/arduino-esp32 #6477: esp32 default program

MatheusCaltran, 2022-03-23:
### Related area

Board Support

### Hardware specification

Support ESP32

### Is your feature request related to a problem?

I verified that some ESP32 doesn't work in tja1050, I wanted the default program to compare between those that work and those that don't

### Describe the solution you'd like

I would like a program that runs:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5564
load:0x40078000,len:0
load:0x40078000,len:13756
entry 0x40078fb4
[0;32mI (29) boot: ESP-IDF v3.0.3 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 08:53:32[0m
[0;32mI (29) boot: Enabling RNG early entropy source...[0m
[0;32mI (34) boot: SPI Speed      : 40MHz[0m
[0;32mI (38) boot: SPI Mode       : DIO[0m
[0;32mI (42) boot: SPI Flash Size : 4MB[0m
[0;32mI (46) boot: Partition Table:[0m
[0;32mI (49) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (57) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (64) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (72) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (79) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (87) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (94) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (102) boot: End of partition table[0m
[0;32mI (106) boot: No factory image, trying OTA 0[0m
[0;32mI (111) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x200f4 (131316) map[0m
[0;32mI (166) esp_image: segment 1: paddr=0x0012011c vaddr=0x3ffc0000 size=0x02d6c ( 11628) load[0m
[0;32mI (171) esp_image: segment 2: paddr=0x00122e90 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (173) esp_image: segment 3: paddr=0x00123298 vaddr=0x40080400 size=0x0cd78 ( 52600) load[0m
[0;32mI (203) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0xdf390 (914320) map[0m
[0;32mI (524) esp_image: segment 5: paddr=0x0020f3b0 vaddr=0x4008d178 size=0x01f14 (  7956) load[0m
[0;32mI (527) esp_image: segment 6: paddr=0x002112cc vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (539) boot: Loaded app from partition at offset 0x100000[0m
[0;32mI (539) boot: Disabling RNG early entropy source...[0m
Bin version(Wroom32):1.1.2

I (660) wifi: wifi firmware version: de47fad
I (660) wifi: config NVS flash: enabled
I (660) wifi: config nano formating: disabled
I (670) wifi: Init dynamic tx buffer num: 32
I (670) wifi: Init data frame dynamic rx buffer num: 32
I (670) wifi: Init management frame dynamic rx buffer num: 32
I (676) wifi: wifi driver task: 3ffdecc0, prio:23, stack:3584
I (681) wifi: Init static rx buffer num: 10
I (684) wifi: Init dynamic rx buffer num: 32
I (689) wifi: wifi power manager task: 0x3ffdfd8c prio: 21 stack: 2560
I (721) wifi: mode : softAP (24:d7:eb:59:5e:55)
I (729) wifi: mode : sta (24:d7:eb:59:5e:54) + softAP (24:d7:eb:59:5e:55)
I (733) wifi: mode : softAP (24:d7:eb:59:5e:55)

### Describe alternatives you've considered

I know that from the ESP32 IDF you can use esptool.py read_mac but I've tried and nothing worked, I wanted a program for the arduino IDE

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2022-03-24:
I am not going to write a program for you, but the esptool included with arduino (or available at https://github.com/espressif/esptool) can give you the base mac address from a connected computer.
```
lbernstone@IdeaPad:~$ ~/.arduino15/packages/esp32/tools/esptool_py/3.1.0/esptool.py chip_id
esptool.py v3.1
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:af:77:70
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-C3 has no Chip ID. Reading MAC instead.
MAC: 7c:df:a1:af:77:70
Hard resetting via RTS pin...
```

RussNelson, 2023-11-10:
I have the same question -- where might I find the source for "Bin version(Wroom32):1.1.2" ?

Ev3rdown, 2024-02-15:
Posting here for posterity.

Found the most likely source to be [this specific version of ESP-AT](https://github.com/espressif/esp-at/tree/v1.1.2.0).
You can find [binaries here](https://www.espressif.com/en/products/sdks/esp-at/resource).

Abdull, 2024-08-09:
> I have the same question -- where might I find the source for "Bin version(Wroom32):1.1.2" ?

* https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Binary_Lists/esp_at_binaries.html
  * in particular: https://dl.espressif.com/esp-at/firmwares/esp32/ESP32-WROOM-32/ESP32-WROOM-32-AT-V1.1.2.0.zip