espressif/arduino-esp32 #6478: Issue with HTTPS Post

StarOfSlytherin, 2022-03-24:
### Board

NodeMCU 32S

### Device Description

Just the Dev Module

### Hardware Configuration

No

### Version

latest master

### IDE Name

Platform IO (Not exactly sure about Version. I'm using the latest version available on Platform IO)

### Operating System

PopOS! 21.10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

460800

### Description

I have a HTTPS Server running on the Cloud based on Flask. It has to respond by sending a JSON. I have observed that while testing on httpbin.org I don't get this error. So I'm trying to figure out the meaning of this error to fix it.

This is the error I'm getting in `while (https.available() == 0)`
`[E][ssl_client.cpp:36] _handle_error(): [data_to_read():287]: (-80) UNKNOWN ERROR CODE (0050)`

I know the code is working because I get a JSON sometimes (check the codeblocks below)

Error:
```
HTTPS Test Program
Connecting to WiFi | SSID: *********...
Connected to WiFi | Signal: -54dBm
Connecting to HTTPS Server...
Sending Data via HTTPS POST...
......................................................................................................................................................................................................................................................................................................................................................................[E][ssl_client.cpp:36] _handle_error(): [data_to_read():287]: (-80) UNKNOWN ERROR CODE (0050)
```

Expected Output:
```
HTTPS Test Program
Connecting to WiFi | SSID: *********...
Connected to WiFi | Signal: -54dBm
Connecting to HTTPS Server...
Sending Data via HTTPS POST...
..........................................................................................................................................................................................................................................................................................................................................................
HTTP/1.0 200 OK
Content-Type: text/html; charset=utf-8
Content-Length: 207
Server: Werkzeug/2.0.3 Python/3.6.8
Date: Thu, 24 Mar 2022 07:57:14 GMT

{"MAC": "FF:FF:FF:FF:FF:FF", "Hardware": "ESP32_Prototype", "status": 76890}
```
I don't get any errors on the server side.

### Sketch

```cpp
#include <Arduino.h>
#include <WiFiClientSecure.h>
#include <WiFi.h>
#include <sslFile.h>
#include <HTTPClient.h>

const char *SSID = "";
const char *PASS = "";

void connect2WiFi()
{
    WiFi.begin(SSID, PASS);
    Serial.printf("Connecting to WiFi | SSID: %s", SSID);
    while (WiFi.isConnected() == false)
    {
        Serial.print(".");
        delay(1000);
    }
    Serial.println();
    if (WiFi.isConnected() == true)
    {
        Serial.printf("Connected to WiFi | Signal: %d\n", WiFi.RSSI());
    }
}

void HTTPS_Post()
{
    const char *DATA = "{\"MAC\":\"FF:FF:FF:FF:FF:FF\",\"Hardware\":\"ESP32_Prototype\"}";

    WiFiClientSecure https;
    https.setCACert(CA_Cert);
    int connHTTPS = https.connect(serverName, serverPort);
    Serial.println("Connecting to HTTPS Server...");
    if (connHTTPS == 1)
    {
        Serial.println("Sending Data via HTTPS POST...");
        https.println("POST / HTTP/1.1");
        https.print("Host: ");
        https.println(url);
        https.println("Content-Type: application/json");
        https.print("Content-Length: ");
        https.println(strlen(DATA));
        https.println("Connection: Close");
        https.println();
        // Body
        https.println(DATA);
        https.println();

        while (https.available() == 0)
        {
            Serial.print(".");
        }
        Serial.println();

        while (https.available())
        {
            char c = https.read();
            Serial.print(c);
        }
        https.stop();
    }
    else
    {
        https.stop();
        Serial.println("Connection Failed");
        ESP.restart();
    }
}

void setup()
{
    Serial.begin(115200);
    Serial.println("HTTPS Test Program");
    connect2WiFi();
    delay(100);
    HTTPS_Post();
}

void loop()
{
    delay(1000);
}
```

### Debug Message

```plain
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():287]: (-80) UNKNOWN ERROR CODE (0050)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-03-24:
How do you use latest Arduino master with platformio?

StarOfSlytherin, 2022-03-24:
As I mentioned in the IDE Name section, I don't know the exact version. I'm using the latest version that's available on Platform IO

Jason2866, 2022-03-24:
The last version available from Platformio is 1.0.6.
This version is not supported anymore

StarOfSlytherin, 2022-03-24:
So manual update to the latest version would fix the issue?

Jason2866, 2022-03-24:
You can NOT manually update easily the Arduino version when used with Platformio.
I have done a custom setup for Platformio which works fine from the feedbacks i got.
```
platform = https://github.com/Jason2866/platform-espressif32.git
```

StarOfSlytherin, 2022-03-24:
Thank you, I'll check out

StarOfSlytherin, 2022-03-25:
> You can NOT manually update easily the Arduino version when used with Platformio. I have done a custom setup for Platformio which works fine from the feedbacks i got.
> 
> ```
> platform = https://github.com/Jason2866/platform-espressif32.git
> ```

This fix worked. But on other projects I'm getting compile errors.
```
/home/pop/.platformio/packages/framework-arduinoespressif32@src-650cd46feb0152d88070dcc7c482fb21/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/ip4_addr.h:63:37: error: expected ')' before numeric constant
 #define IPADDR_NONE         ((u32_t)0xffffffffUL)
                             ~       ^~~~~~~~~~~~
/home/pop/.platformio/packages/framework-arduinoespressif32@src-650cd46feb0152d88070dcc7c482fb21/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/inet.h:71:29: note: in expansion of macro 'IPADDR_NONE'
 #define INADDR_NONE         IPADDR_NONE
                             ^~~~~~~~~~~
/home/pop/.platformio/packages/framework-arduinoespressif32@src-650cd46feb0152d88070dcc7c482fb21/cores/esp32/IPAddress.h:94:17: note: in expansion of macro 'INADDR_NONE'
 const IPAddress INADDR_NONE(0, 0, 0, 0);
                 ^~~~~~~~~~~
```

Jason2866, 2022-03-25:
On 'other projects' is a unspecific problem description. Providing a example sketch would help.

mobizt, 2022-03-26:
Since the content length header is **strlen(DATA)**, the payload to send should match the header.

Then replace these lines

```cpp
// Body
https.println(DATA);
https.println();
```

with

```cpp
// Body
https.print(DATA);
```

StarOfSlytherin, 2022-03-28:
> Since the content length header is **strlen(DATA)**, the payload to send should match the header.
> 
> Then replace these lines
> 
> ```c++
> // Body
> https.println(DATA);
> https.println();
> ```
> 
> with
> 
> ```c++
> // Body
> https.print(DATA);
> ```

Thanks, this seems to work fine. What exactly was the issue? Extra line is causing the issue?

mobizt, 2022-03-28:
Yes, that can cause the issue, depending on server because of invalid payload.

closed 2022-03-28 by StarOfSlytherin