espressif/arduino-esp32 #6480: ESP32-S2 Writing Only 512B of Good Data to XTSD08G Using SD Library

Legohead259, 2022-03-25:
### Board

ESP32-S2 WROOM

### Device Description

This device is a custom PCB with the ESP32-S2 WROOM as the main MCU and various peripheral devices. The schematic can be found [here](https://github.com/Legohead259/Project-Thetis-CAD/blob/main/Documentation/Thetis_RevF3_SCH.pdf). 

### Hardware Configuration

The device in question is the [XTX XTSD08G](https://www.lcsc.com/product-detail/NAND-FLASH_XTX-XTSD08GLGEAG_C558840.html) flash storage module connected to the ESP's SPI bus. All hardware is confirmed verified to work independently and the XTSD shows up as an SD card, as expected, and does not show this issue with the SAME51. I have replaced the XTSD on the board with a brand new, fresh from factory chip. The XTSD modules have also been formatted in FAT16 from the factory.

### Version

v2.0.2

### IDE Name

VS Code with Arduino Plugin

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

The sketch below should continually append "World!" to a file called "hello.txt" located in the root directory of the XTSD (which acts like a standard SD card). The code compiles, uploads, and runs flawlessly, but when the files on the XTSD are checked (using [Chegewara's ESPTinyUSB SD MSC example](https://github.com/chegewara/EspTinyUSB/blob/master/examples/device/msc/sd_msc/sd_msc.ino)) the file only has 512 bytes of good data as shown below:
```
HelloWorld!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
World!
[... 259 lines cut ...]
    // readFile(SD, "/hello.txt");
    // deleteFile(SD, "/foo.txt");
    // renameFile(SD, "/hello.txt", "/foo.txt");
    // readFile(SD, "/foo.txt");
    // testFileIO(SD, "/test.txt");
    Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
    Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));
}

void loop(){
    appendFile(SD, "/hello.txt", "World!\n");
    Serial.printf("Used space: %lluMB\n\n", SD.usedBytes() / (1024 * 1024));
    delay(500);
}
```

### Debug Message

```plain
N/A Code compiles and uploads without issue
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-03-25:
You mention USB MSC, but is the log above from Arduino's SD lib reading the file or from USB?

P-R-O-C-H-Y, 2022-03-25:
Hi @Legohead259,

The version 2.0.2 has bugs in SD library. Can you try to clone master branch of Arduino-esp32 and test with that? There are instructions [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html). Look for manual installation.

Please set "Core debug level" to "VERBOSE" and post the debug message :)

Legohead259, 2022-03-27:
> You mention USB MSC, but is the log above from Arduino's SD lib reading the file or from USB?

That is the contents of the file, as copied over USB.

> Hi @Legohead259,
>
> The version 2.0.2 has bugs in SD library. Can you try to clone master branch of Arduino-esp32 and test with that? There are instructions [...]

Hi there! I think I installed the master branch successfully, however the same issue occurs during testing. I set the the core level debug to verbose, but did not see any additional outputs during compiling. Is the debug message posted over Serial when the chip boots?

P-R-O-C-H-Y, 2022-03-28:
@Legohead259 Yes, exactly. Post the debug messages printed over Serial. There must be some errors about sd, why its failing to write to file.

Legohead259, 2022-03-28:
@P-R-O-C-H-Y I set the debug level to verbose and did not see anything additionally outputted. Are the debug messages sent over UART or USB?

Legohead259, 2022-03-30:
@P-R-O-C-H-Y 

I have it printing statements through USB. I can see the usual print messages ("Appending to File", "File Appended", "Used space..."). However, there does not seem to be any additional debug messages like you have described. Another bug? Or do the debug messages specifically use the UART GPIO pins? If it is the latter, that will be a problem...

P-R-O-C-H-Y, 2022-03-30:
@Legohead259 
If you set debug level to verbose and you don't see any debug messages, you can try call `Serial.setDebugOutput(true);` in `setup()` after `Serial.Begin()`. But I think this should be enabled by default. If your code is different from sketch provided, use instead of Serial the "serial you use".

closed 2022-04-20 by VojtechBartoska