espressif/arduino-esp32 #6481: Export compiled binary - merged

owenduffy, 2022-03-25:
### Related area

Export compiled binary

### Hardware specification

ESP32

### Is your feature request related to a problem?

It would be useful if the Menu/Sketch/ExportCompileBinary exported a merged binary rather than just a part of the merged image, leaving it to the user to find the other bits which are in temporary directories.

### Describe the solution you'd like

That export binary exports the merged binary file, ready for uploading at address 0x0.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Jason2866, 2022-03-28:
If you change to Platformio as IDE, this is easily possible. Just add [this](https://github.com/arendst/Tasmota/blob/development/pio-tools/post_esp32.py) script and you will get a merged firmware.

igrr, 2022-04-11:
esptool.py recently got support of `merge_bin` command which can combine multiple binaries into one. So this can be implemented.

I'm just a bit worried that this will be a breaking change in the behavior of "export compiled binary" command. Maybe we should postpone this change until the next major release.

owenduffy, 2022-04-13:
To clarify what I am asking for... here is the line of the build when the compiled module is uploaded to the chip:

`C:\Users\owen\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.1.0/esptool.exe --chip esp32 --port COM19 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0xe000 C:\Users\owen\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/partitions/boot_app0.bin 0x1000 C:\Users\owen\AppData\Local\Temp\arduino_build_400366/EspRelay.ino.bootloader.bin 0x10000 C:\Users\owen\AppData\Local\Temp\arduino_build_400366/EspRelay.ino.bin 0x8000 C:\Users\owen\AppData\Local\Temp\arduino_build_400366/EspRelay.ino.partitions.bin 
`
You will see that it uploads 4 files at various addresses, some of which are in the temp build directory.

I would like that when I export a compiled binary from the IDE, that it merges these four files to create a 'combined' hex load file.

me-no-dev, 2022-04-13:
Adding link to [export hooks support in platform.txt](https://arduino.github.io/arduino-cli/0.21/platform-specification/#recipes-to-export-compiled-binary). We need to investigate if we can use those to combine and export a merged binary as well. To note, this needs to be additional binary, because it can not be used for OTA.

owenduffy, 2022-04-13:
OK, I understand you need to keep the binary for OTA, perhaps another menu option to "export combined binary"?

me-no-dev, 2022-04-13:
we can not alter the menu there. We can try to export both at the same time.

owenduffy, 2022-04-13:
What a good idea!

crbyxwpzfl, 2024-08-26:
hi I just stumbled across this and thought I leave this here.
with the arduino cli you can specify a `--build-path ./firmware` where all the various binaries are put. including `.merged.bin`. With this binary eg. allows to programm a esp via [browser esptool](https://espressif.github.io/esptool-js/)
this example compiles for an esp32 board to find `fqbn` of your board use `arduino-cli board search ... `
``` 
arduino-cli compile --fqbn esp32:esp32:adafruit_feather_esp32s3 --build-path ./firmwarefloder
```
hope this is relevant even though its not done via ui.

lbernstone, 2024-08-26:
This issue is completed. It should be closed.

closed 2024-08-27 by me-no-dev

torntrousers, 2024-08-27:
what is the solution - how do you get the combined binary?

Jason2866, 2024-08-27:
@torntrousers 3 posts before

torntrousers, 2024-08-27:
> @torntrousers 3 posts before

https://github.com/espressif/arduino-esp32/issues/6481#issuecomment-2310911317

So only with the Arduino CLI not the IDE?