espressif/arduino-esp32 #6482: HTTPUpdateServer does not update SPIFFS or LittleFS

mrWheel, 2022-03-25:
### Board

ESP32 Wrover Module

### Device Description

Nothing, just the bare board

### Hardware Configuration

Nothing

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

macOS 12.2.1

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Created a SPIFS or LittleFS bin.

Tried to upload using the `WebUpdater` example sketch

Got these error's while trying to upload LittleFS.bin:
Monitor:
<pre>
Update: WebUpdater.littlefs.bin
Bad Size Given
.Bad Size Given
</pre>
Browser:
<pre>
Update error: Bad Size Given
</pre>

Same errors while trying to upload SPIFFS.bin

### Sketch

```cpp
/*
  To upload through terminal you can use: curl -F "image=@firmware.bin" esp32-webupdate.local/update
*/

#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include <HTTPUpdateServer.h>

#ifndef STASSID
#define STASSID "AandeWiFi"
#define STAPSK  "3741TS12tl"
#endif

const char* host = "esp32-webupdate";
const char* ssid = STASSID;
const char* password = STAPSK;

WebServer httpServer(80);
HTTPUpdateServer httpUpdater(true);

void setup(void) {

  Serial.begin(115200);
  Serial.println();
  Serial.println("Booting Sketch...");
  WiFi.mode(WIFI_AP_STA);
  WiFi.begin(ssid, password);

  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    WiFi.begin(ssid, password);
    Serial.println("WiFi failed, retrying.");
  }
  
  MDNS.begin(host);
  if (MDNS.begin("esp32")) {
    Serial.println("mDNS responder started");
  }

  httpUpdater.setup(&httpServer);
  httpServer.begin();

  MDNS.addService("http", "tcp", 80);

  Serial.print("IP address [");
  Serial.print(WiFi.localIP());
  Serial.println("]");
  Serial.printf("HTTPUpdateServer ready! Open http://%s.local/update in your browser\n", host);
}

void loop(void) {
  httpServer.handleClient();
}
```

### Debug Message

```plain
no other debug info
```

### Other Steps to Reproduce

Updating the `WebUpdater.ino.bin` or other `.ino.bin` files without a problem.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-03-31:
Hello, are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

mrWheel, 2022-04-03:
@VojtechBartoska 
Sorry for the delay.

Tested with 2.0.3-RC1, same result.

Used [Tools]->[ESP32 LittleFS Data Upload] to create a `WebUpdater.littlefs.bin` file
<pre>
16:16:16.174 -> IP address [192.168.12.24]
16:16:16.174 -> HTTPUpdateServer ready! Open http://esp32-webupdate.local/update in your browser
16:17:08.883 -> Update: WebUpdater.littlefs.bin
16:17:08.883 -> Bad Size Given
16:17:08.883 -> .Bad Size Given
</pre>

Then I user [Tools]->[ESP32 Sketch Data Upload] to create a `WebUpdater.spiffs.bin` file
<pre>
16:19:30.751 -> Booting Sketch...
16:19:33.591 -> mDNS responder started
16:19:33.591 -> IP address [192.168.12.24]
16:19:33.591 -> HTTPUpdateServer ready! Open http://esp32-webupdate.local/update in your browser
16:22:58.507 -> Update: WebUpdater.spiffs.bin
16:22:58.507 -> Bad Size Given
16:22:58.507 -> .Bad Size Given
</pre>
Update firmware still works:
<pre>
16:17:08.883 -> .Bad Size Given
16:19:16.387 -> Update: WebUpdater.ino.bin
16:19:16.387 -> ............... lots of dots ...................Update Success: 751504
16:19:29.959 -> Rebooting...
16:19:30.097 -> ets Jun  8 2016 00:22:57
</pre>

VojtechBartoska, 2022-04-04:
Thanks for testing @mrWheel, I'm adding this to our Roadmap and we will take a look. :)

nospam2k, 2022-05-13:
I'm not sure where things are on this but a simple fix is to just change HTTPUpdateServer.h as follows:

`
if (_serial_output)
    Serial.printf("Update: %s\n", upload.filename.c_str());
    if (upload.name == "filesystem") {
        if (!Update.begin(SPIFFS.totalBytes(), U_SPIFFS)) {//start with max available size << HERE
            if (_serial_output) Update.printError(Serial);
        }
    }
...
`
To:
`
if (!Update.begin(UPDATE_SIZE_UNKNOWN, U_SPIFFS)) {//start with max available size << HERE
`

IoTManagerProject, 2022-09-18:
Good Day. 
I also try to find way esp32 littlefs http update. Are you have any update regarding this issue? I transfer my project to esp32 little fs and now become so sad that no http OTA available...

zekageri, 2022-10-27:
I'm using Async server for the update. Firmware is updating but LittleFs don't.
Here is the relevant part of my sketch:

```cpp

Update.onProgress([this](unsigned int progress, unsigned int total){
    int percent = (progress * 100) / total;
    if( percent == latestUpdatePercent ){ return; }
    latestUpdatePercent = percent;
    debug.print(
        DEBUG_INFO,
        "Update progress: %d byte from total of %d bytes. Which is %d%%\n",
        progress, total, percent
    );
});

 /*
*   Handling the firmware or the file system upload.
*/
server->on("/firmware", HTTP_POST, [this](AsyncWebServerRequest *request) {},
    [this](AsyncWebServerRequest *request, const String &filename, size_t index, uint8_t *data, size_t len, bool final) {
        
    // Starting the update since the index is 0.
    if (!index) {
        final = !firmwareStart(request);
    }

    // Writing bytes to the partition.
    if (Update.write(data, len) != len) {
        final = true;
    }

    // Check if the final bytes are written.
    // End the update and respond accordingly.
    if (final) {
        if (!Update.end(true)) {
           firmwareError(request);
        }else{
           firmwareSuccess(request);
        }
     }
});

/*
*   Handling the start of the firmware update process.
*   Checks if the file is a LittleFs file or a firmware.
*   Client will send a flag indicating if it is a flash update or not
*   Client also sends the fileSize to be able to track the progress.
*/
boolean AdminSystem::firmwareStart(AsyncWebServerRequest *request){
    int command     = U_FLASH;
    int fileSize    = UPDATE_SIZE_UNKNOWN;
    if (request->hasArg("fileSize")) {
        fileSize = request->arg("fileSize").toInt();
    }
    if (request->hasArg("isFlash")) {
        boolean isFlash = request->arg("isFlash").toInt();
        if (isFlash) {
            command = U_SPIFFS;
        }
    }
    
    if ( !Update.begin( fileSize, command ) ) { return false; }
    sys.startFirmware();
    return true;
}

/*
*   Handling the success event.
*/
void AdminSystem::firmwareSuccess(AsyncWebServerRequest *request){
    request->onDisconnect([]{ sys.endFirmware(); sys.restart(); });
    request->send(200, "text/plain", "Success");
}

/*
*   Handling the error event.
*/
void AdminSystem::firmwareError(AsyncWebServerRequest *request){
    sys.endFirmware();
    request->send(500, "text/plain", "Error");
}
```

LittleFs update can't even start.
It says that Firmware upload ended.

HugoML, 2022-11-02:
httpUpdate supports SPIFFS, but not LittleFS. 
Need to add this to the feature request.

mrWheel, 2022-12-03:
> I'm not sure where things are on this but a simple fix is to just change HTTPUpdateServer.h as follows:
> 
> ```
> if (_serial_output)
>     Serial.printf("Update: %s\n", upload.filename.c_str());
>     if (upload.name == "filesystem") {
>         >>>> if (!Update.begin(SPIFFS.totalBytes(), U_SPIFFS)) {//start with max available size <<<<
>             if (_serial_output) Update.printError(Serial);
>         }
>     }
> ...
> ```
> 
> To:
> 
> ```
> if (!Update.begin(UPDATE_SIZE_UNKNOWN, U_SPIFFS)) {//start with max available size
> ```

Hm.. tried this but strange things happen after flashing LitteFS.

```
| Available LittleFS space [     0]kB
|            LittleFS Size [  1472]kB
|            LittleFS Used [1507328]bytes
```
.. which is not correct!

mrWheel, 2022-12-03:
Is there any update for this problem?

First I had to convert all my ESP8266 code to LittleFS and now I'm switching to ESP32 and than it seems LittleFS is not supported ...

kel30a, 2023-02-02:
> 
 it looks like it is working for me with the following line: 
`if(!Update.begin(LittleFS.totalBytes(), command)){
    Update.printError(Serial);
}`

Using Update @ 2.0.0

**IDE Name**
VS Code and PlatformIO

**Operating System**
Windows 11

HamzaHajeir, 2023-05-14:
Is there an update in this issue so far?
It seems there's no introduced API for LittleFS, current is `updateSpiffs(...)`,

VojtechBartoska, 2023-05-15:
@HamzaHajeir There is no update yet, we'll triage this for upcoming major release 3.0.0, no date yet

tobiasfaust, 2024-01-31:
Hi, any news?
i get an `Wrong Magic Byte` by using `if(!Update.begin(LittleFS.totalBytes(), U_SPIFFS))`
I´m afraid, no U_LITTLEFS is available

I´m using those:
<pre>
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.2.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5)
 - tool-esptoolpy @ 1.40201.0 (4.2.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 60 compatible libraries
Scanning dependencies...
Dependency Graph
|-- PubSubClient @ 2.8.0
|-- ArduinoJson @ 7.0.2
|-- ArduinoQueue @ 1.2.5
|-- StreamUtils @ 1.8.0+sha.03fafbb
|-- ESPAsyncWiFiManager @ 0.31.0+sha.3b9a58a
|-- Uptime Library @ 1.0.0+sha.fa29a94
|-- ESP Async WebServer @ 1.2.3
|-- AsyncTCP @ 1.1.1
|-- DNSServer @ 2.0.0
|-- FS @ 2.0.0
|-- LittleFS @ 2.0.0
|-- Update @ 2.0.0
|-- WiFi @ 2.0.0
Building in release mode
</pre>

VojtechBartoska, 2024-02-20:
Same here @lucasssvaz, can you please investigate this? Thanks

lucasssvaz, 2024-03-06:
The update of the SPIFFS partition failed because it was not initialized. Make sure to add something like this to your code:

```
if(!SPIFFS.begin(true)){
    Serial.println("SPIFFS Mount Failed");
    return;
  }
```

closed 2024-03-06 by lucasssvaz