espressif/arduino-esp32 #6485: Cant' connect to network.  OK in release 1.0.6

fanfanlatulipe26, 2022-03-27:
### Board

ESP32 Dev Kit

### Device Description

nothing

### Hardware Configuration

nothing

### Version

v2.0.2

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

Can't connect to a wifi network.  STA mode.
Standard example HelloServer in for the library WebServer fails.

**Exactly** the same code compiled with 1.0.6  works perfectly.

But it seems that the results depend on the network itself. I have a demo working perfectly on an other network at home and this demo fails on the network I use this day.(of course after updating ssid and password  ;-=)
Network uses WPA (TKIP + AES) 
.

### Sketch

```cpp
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>

const char* ssid = "Alice_Maman";
const char* password = "xxxxxxxxxxxxxxxx";

WebServer server(80);

const int led = 2;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp32!");
  digitalWrite(led, 0);
}

void handleNotFound() {
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void) {
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp32")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
  delay(2);//allow the cpu to switch to other tasks
}
```

### Debug Message

```plain
Print debug when example is compiles with 2.0.2:
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

[   165][V][WiFiGeneric.cpp:272] _arduino_event_cb(): STA Started
[   167][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   167][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START

....[  2236][V][WiFiGeneric.cpp:289] _arduino_event_cb(): STA Disconnected: SSID: Alice_Maman, BSSID: 00:00:00:00:00:00, Reason: 201
[  2236][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  2244][W][WiFiGeneric.cpp:852] _eventCallback(): Reason: 201 - NO_AP_FOUND
[  2252][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
....[  4317][V][WiFiGeneric.cpp:289] _arduino_event_cb(): STA Disconnected: SSID: Alice_Maman, BSSID: 00:00:00:00:00:00, Reason: 201
[  4317][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  4325][W][WiFiGeneric.cpp:852] _eventCallback(): Reason: 201 - NO_AP_FOUND
[  4333][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0

======================
Print debug when example is compiles with 1.0.6

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START

......[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.0.16, MASK: 255.255.255.0, GW: 192.168.0.254
.
Connected to Alice_Maman
IP address: 192.168.0.16
MDNS responder started
HTTP server started
```

### Other Steps to Reproduce

In the IDE I select the board  "ESP32 Dev Module"

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-03-27:
WPA really? Not WPA2. WPA is highly insecure. Do not use anymore.

fanfanlatulipe26, 2022-03-27:
WPA is at least more secure than WEP.
The box from my internet provider does not offer WPA2.

Jason2866, 2022-03-27:
Open a ticket to get the box changed.

pipi61, 2022-03-27:
Try change to wpa2 on your router

Jason2866, 2022-03-27:
> WPA is at least more secure than WEP. The box from my internet provider does not offer WPA2.

@pipi61

fanfanlatulipe26, 2022-03-28:
Back home, I ran some tests with my usual internet box where I have more choice with the authentication scheme and I confirm that It appears that WPA is no longer supported in 2.0.2.
In my opinion this choice is more than questionable:

- First of all it introduces incompatibility / regression with 1. 0.6
- It is up to the user to choose the level of security. It is his own responsibility.
- It introduces more discrepencies with the latest documentations

Even the latest [doc from Espressif]( https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html) states "Various security modes for the above (WPA, WPA2, WEP, etc.)" and from [arduino-esp32 support documentation](https://github.com/espressif/arduino-esp32/blob/master/docs/source/api/wifi.rst) "Security modes (WPA, WPA2, WEP, etc.)"

me-no-dev, 2022-03-28:
"fix" is coming. More of an option to allow WPA actually :)

commit 5f6d093d (2022-04-28): Add more controls over WiFi.begin and WiFi.scan (#6651)

Added options to WiFi STA for connect:
- setMinSecurity: default WIFI_AUTH_WPA2_PSK
- setScanMethod: default WIFI_FAST_SCAN
- setSortMethod: default WIFI_CONNECT_AP_BY_SIGNAL (required all channels scan method)

Added parameters for SSID and BSSID to WiFi.scanNetworks()

Fixes: #6485
Fixes: Any issue about WiFi connecting slower now than in 1.0.x
closed 2022-04-28 by me-no-dev