espressif/arduino-esp32 #6486: ESP32  How to set high resistance power supply for GPIO

hzhh110, 2022-03-27:
### Related area

ESP32

### Hardware specification

ESP32

### Is your feature request related to a problem?

ESP32  How to set high resistance power supply for GPIO

### Describe the solution you'd like

ESP32  How to set high resistance power supply for GPIO

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-03-30:
@P-R-O-C-H-Y Can you help with answering this please?

VojtechBartoska, 2023-02-16:
Hello, closing this issue as expired. We don't know the answer and there is also not any answer from the Community. Thanks for understanding.

closed 2023-02-16 by VojtechBartoska