espressif/arduino-esp32 #6488: SDWebServer example has issues with paths

menachers, 2022-03-27:
### Board

DOIT ESP32 DEVKIT v1

### Device Description

Plain module with SD reader

### Hardware Configuration

Nothing special besides SD reader.

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

When using the basic example for SDWebServer on edit mode the path are generated wrong.
A relative path is generated with a double slash // at the beginning resulting that no files can be uploaded or downloaded.
If the path in the request is manually corrected in the browser everything works fine.
So I think the problem is located in the htm file provided here:
[https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples/SDWebServer/SdRoot/edit](url
)

This is happening:
[Screenshot 2022-03-27 154213](https://user-images.githubusercontent.com/11912920/160284622-4b230a49-b0af-414c-bd04-28ed7e3b0f92.jpg)

I tried to modify the javascript but I have no clue where to start since I have no idea where the path variable "comes from". 

### Sketch

```cpp
-
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2022-03-31:
Hi @menachers,
I have tested the issue and I made a change in the example to make it work. 
I will post a PR where you can see the changes later today.

P-R-O-C-H-Y, 2022-03-31:
@menachers You can test the changes from PR #6505 till it gets merged. But don't update to 2.0.3-RC1 yet otherwise the SDWebServer won't work. SD reading needs to be fixed. Wait for 2.0.3 release.

closed 2022-03-31 by me-no-dev

menachers, 2022-03-31:
Thank you for the quick fix. It works fine now.

PR #6505 (2022-03-31): SDWebServer example fix for 2.x.x versions
## Summary
Fixing path to the files in edit mode.
Needs to be changed according to SD changes from 1.0.6 to 2.0.0.

## Impact
None.

## Related links
Closes #6488