espressif/arduino-esp32 #6490: esp_wpa2.h Not Found when Building ardunio-esp32

esambell, 2022-03-28:
### Board

M5 StickC

### Device Description

M5StickC no additional peripherals.

### Hardware Configuration

USB Connection

### Version

v2.0.2

### IDE Name

VS Code

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

1500000

### Description

Include of esp_wpa2.h in WiFiST.cpp fails. Needed to manually hardcode the path using: 

#include "C:\esp\esp-idf\components\wpa_supplicant\esp_supplicant\include\esp_wpa2.h"

Arduino-ESP: 2.0.2-102-g8fe0efe8

### Sketch

```cpp
N/A
```

### Debug Message

```plain
[1/1] cmd.exe /C "cd /D C:\VS_Code\FlyOver_Next_Show_Timer_Watch\build\bootloader\esp-idf\esptool_py && C:\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/VS_Code/FlyOver_Next_Show_Timer_Watch/build/bootloader/bootloader.bin"
Bootloader binary size 0x62b0 bytes. 0xd50 bytes (12%) free.
[3/12] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/WiFi/src/WiFiSTA.cpp.obj
FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/WiFi/src/WiFiSTA.cpp.obj
ccache C:\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../components/arduino/variants/esp32 -I../components/arduino/cores/esp32 -I../components/arduino/libraries/ArduinoOTA/src -I../components/arduino/libraries/AsyncUDP/src -I../components/arduino/libraries/BLE/src -I../components/arduino/libraries/BluetoothSerial/src -I../components/arduino/libraries/DNSServer/src -I../components/arduino/libraries/EEPROM/src -I../components/arduino/libraries/ESP32/src -I../components/arduino/libraries/ESPmDNS/src -I../components/arduino/libraries/Ethernet/src -I../components/arduino/libraries/FFat/src -I../components/arduino/libraries/FS/src -I../components/arduino/libraries/HTTPClient/src -I../components/arduino/libraries/HTTPUpdate/src -I../components/arduino/libraries/LittleFS/src -I../components/arduino/libraries/NetBIOS/src -I../components/arduino/libraries/Preferences/src -I../components/arduino/libraries/RainMaker/src -I../components/arduino/libraries/SD_MMC/src -I../components/arduino/libraries/SD/src -I../components/arduino/libraries/SimpleBLE/src -I../components/arduino/libraries/SPIFFS/src -I../components/arduino/libraries/SPI/src -I../components/arduino/libraries/Ticker/src -I../components/arduino/libraries/Update/src -I../components/arduino/libraries/USB/src -I../components/arduino/libraries/WebServer/src -I../components/arduino/libraries/WiFiClientSecure/src -I../components/arduino/libraries/WiFi/src -I../components/arduino/libraries/WiFiProv/src -I../components/arduino/libraries/Wire/src -I../components/arduino/cores/esp32/libb64 -IC:/esp/esp-idf/components/newlib/platform_include -IC:/esp/esp-idf/components/freertos/include -IC:/esp/esp-idf/components/freertos/include/esp_additions/freertos -IC:/esp/esp-idf/components/freertos/port/xtensa/include -IC:/esp/esp-idf/components/freertos/include/esp_additions -IC:/esp/esp-idf/components/esp_hw_support/include -IC:/esp/esp-idf/components/esp_hw_support/include/soc -IC:/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -IC:/esp/esp-idf/components/esp_hw_support/port/esp32/. -IC:/esp/esp-idf/components/heap/include -IC:/esp/esp-idf/components/log/include -IC:/esp/esp-idf/components/lwip/include/apps -IC:/esp/esp-idf/components/lwip/include/apps/sntp -IC:/esp/esp-idf/components/lwip/lwip/src/include -IC:/esp/esp-idf/components/lwip/port/esp32/include -IC:/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/esp/esp-idf/components/soc/include -IC:/esp/esp-idf/components/soc/esp32/. -IC:/esp/esp-idf/components/soc/esp32/include -IC:/esp/esp-idf/components/hal/esp32/include -IC:/esp/esp-idf/components/hal/include -IC:/esp/esp-idf/components/hal/platform_port/include -IC:/esp/esp-idf/components/esp_rom/include -IC:/esp/esp-idf/components/esp_rom/include/esp32 -IC:/esp/esp-idf/components/esp_rom/esp32 -IC:/esp/esp-idf/components/esp_common/include -IC:/esp/esp-idf/components/esp_system/include -IC:/esp/esp-idf/components/esp_system/port/soc -IC:/esp/esp-idf/components/esp_system/port/public_compat -IC:/esp/esp-idf/components/esp32/include -IC:/esp/esp-idf/components/xtensa/include -IC:/esp/esp-idf/components/xtensa/esp32/include -IC:/esp/esp-idf/components/driver/include -IC:/esp/esp-idf/components/driver/esp32/include -IC:/esp/esp-idf/components/esp_pm/include -IC:/esp/esp-idf/components/esp_ringbuf/include -IC:/esp/esp-idf/components/efuse/include -IC:/esp/esp-idf/components/efuse/esp32/include -IC:/esp/esp-idf/components/vfs/include -IC:/esp/esp-idf/components/esp_wifi/include -IC:/esp/esp-idf/components/esp_event/include -IC:/esp/esp-idf/components/esp_netif/include -IC:/esp/esp-idf/components/esp_eth/include -IC:/esp/esp-idf/components/tcpip_adapter/include -IC:/esp/esp-idf/components/esp_phy/include -IC:/esp/esp-idf/components/esp_phy/esp32/include -IC:/esp/esp-idf/components/esp_ipc/include -IC:/esp/esp-idf/components/app_trace/include -IC:/esp/esp-idf/components/esp_timer/include -IC:/esp/esp-idf/components/spi_flash/include -IC:/esp/esp-idf/components/mbedtls/port/include -IC:/esp/esp-idf/components/mbedtls/mbedtls/include -IC:/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -IC:/esp/esp-idf/components/mdns/include -IC:/esp/esp-idf/components/console -IC:/esp/esp-idf/components/esp_adc_cal/include -IC:/esp/esp-idf/components/wifi_provisioning/include -IC:/esp/esp-idf/components/protocomm/include/common -IC:/esp/esp-idf/components/protocomm/include/security -IC:/esp/esp-idf/components/protocomm/include/transports -IC:/esp/esp-idf/components/nvs_flash/include -IC:/esp/esp-idf/components/nghttp/port/include -IC:/esp/esp-idf/components/nghttp/nghttp2/lib/includes -IC:/esp/esp-idf/components/fatfs/diskio -IC:/esp/esp-idf/components/fatfs/vfs -IC:/esp/esp-idf/components/fatfs/src -IC:/esp/esp-idf/components/wear_levelling/include -IC:/esp/esp-idf/components/sdmmc/include -IC:/esp/esp-idf/components/app_update/include -IC:/esp/esp-idf/components/bootloader_support/include -IC:/esp/esp-idf/components/spiffs/include -IC:/esp/esp-idf/components/openssl/include -IC:/esp/esp-idf/components/esp_hid/include -IC:/esp/esp-idf/components/esp_https_ota/include -IC:/esp/esp-idf/components/esp_http_client/include -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=C:/VS_Code/FlyOver_Next_Show_Timer_Watch=. -fmacro-prefix-map=C:/esp/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu++11 -fno-exceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\"v4.4\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -DARDUINO=10812 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"ESP32_DEV\" -DARDUINO_VARIANT=\"esp32\" -DESP32 -MD -MT esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/WiFi/src/WiFiSTA.cpp.obj -MF esp-idf\arduino\CMakeFiles\__idf_arduino.dir\libraries\WiFi\src\WiFiSTA.cpp.obj.d -o esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/WiFi/src/WiFiSTA.cpp.obj -c ../components/arduino/libraries/WiFi/src/WiFiSTA.cpp
../components/arduino/libraries/WiFi/src/WiFiSTA.cpp:45:10: fatal error: esp_wpa2.h: No such file or directory
 #include "esp_wpa2.h"
```

### Other Steps to Reproduce

None

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-03-28:
Needs commit https://github.com/espressif/arduino-esp32/commit/a2077711311ab77e509b79e45dfa3e0a07294c7a which is not merged in main branch

Jason2866, 2022-03-28:
It is fixed. Commit is merged

VojtechBartoska, 2022-03-31:
Hello, up to the merged commit mention above. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

esambell, 2022-04-04:
Hello,

I'm happy to test this, but I'm actually using VS Code, and my Git skills are pretty terrible, but I think I'm on the correct version since when I ran Git Tag the last tag was 2.0.3-RC1. I am no longer seeing the issue.

Thanks,

Eric

Jason2866, 2022-04-04:
Yo can test with VS Code and Platformio by using this
```
[env:esp32]
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
board = esp32dev
framework = arduino
```

closed 2022-04-20 by VojtechBartoska