espressif/arduino-esp32 #6495: Partition Table/Option for 16MB flash (and 32MB flash)

hamza765, 2022-03-28:
### Related area

Board Support

### Hardware specification

ESP32-WROVER-B

### Is your feature request related to a problem?

I'm trying to flash my wrover-b with a 16MB flash partition, but i can't find any support at all for that. There is a option for ` (2mb app/12.5mb fat)` but i'm using LittleFS, so that won't work. I'm currently stuck using No OTA (1mb app/ 3mb spiffs) which works, but only allows for 25% of my total capacity. Additionally, we have a order of chips coming in soon that will have 32MB total capacity (256Mb) of NOR flash mem. so it'll be even smaller of a ratio

### Describe the solution you'd like

A custom board template, partition table, or just a start point for someone completely new to the world of embedded systems. I've googled so much for an answer before coming here as this is a last  resort of sorts. I'd make the custom table, but don't really understand the mem address or size on the custom tables i've  seen.

Thanks in advance!

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Jason2866, 2022-03-28:
For 16MB a partition example. You can change like you want the OTA partitions and
 SPIFFS (used for LittleFS)
Just do the hex math
esp32_partition_app2944k_spiffs10M.csv
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x2F0000,
app1,     app,  ota_1,   0x300000, 0x2F0000,
spiffs,   data, spiffs,  0x5F0000,0xA10000,
```

VojtechBartoska, 2022-04-11:
@hamza765 Was the comment above helpful?

lbernstone, 2022-04-26:
You must have the first partition offset to 0x9000.  You can just use size past there if you prefer.  The total space on a 16M device is 0x1000000.  So, 0x1000000 - 0x9000 = 0xFF7000 left.  Subtract 0x5000 for the nvs partition leaves 0xFF2000.  If you continue for the partitions in Jason2866's example above, you will see that it uses all the space.  https://www.calculator.net/hex-calculator.html, but I'm sure you have one on your computer.

VojtechBartoska, 2022-04-27:
I'm adding this to our Documentation Roadmap as this is something which needs to be explained.

hamza765, 2022-04-27:
Does this look right for the 32M?
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,           ,  0x2000,
app0,     app,  ota_0,         ,  0x500000,
app1,     app,  ota_1,         ,  0x500000,
spiffs,   data, spiffs,        ,  0x15F0000,
```

hamza765, 2022-04-27:
I spent an hour and made a simple partition generator:
https://docs.google.com/spreadsheets/u/0/d/16Hgrb1dloG2keiTx_JBniaQgJ_Ou8zg6xa3p3GGYM2k/copy

The link will ask you if you want to make a copy of the sheets as i don't want anyone messing up the original

closed 2022-05-30 by me-no-dev

trullock, 2023-08-15:
@hamza765 is the hero we need but don't deserve!

PepeTux, 2024-04-23:
@hamza765 great job and thanks for the Partition Generator !

Also you can use the tool:

[The ESP32 Partition Tool is a utility designed to facilitate creating custom partition schemes in Arduino IDE 1.8.x & PlatformIO.](https://github.com/serifpersia/esp32partitiontool)
Install this tool on the old IDE and use it to generate the partition file.

[image]

Remember to generate the partition file with the name "partitions.csv" and paste on your sketch folder to override the default/selected in the menu: Tools >> Partition Schema>> default/other.

regards
Jose Luis

PR #6758 (2022-05-30): [DOCS] Added partition table tutorial
## Description of Change

This PR adds the partition table details and the steps for creating a custom one.

## Tests scenarios

N/A

## Related links

Closes #4551
Closes #6495