espressif/arduino-esp32 #6496: Cannot simultaneously read/write from Serial

msun0948, 2022-03-29:
### Board

WEMOS Lolin32 Lite

### Device Description

WEMOS Lolin32 Lite

### Hardware Configuration

Pins 0 and 4 are jumped together to test simultaneous serial communication
[image]
.

### Version

v2.0.2

### IDE Name

Arduino IDE 1.8.15

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I want to communicate with another sensor later via one serial port (Serial1) and made a quick sketch to test sending and receiving on one serial port.
This sketch on 2.0.2 provides the output:
> Started..

Before hanging indefinitely.

Tested on 1.0.6 the same code/sketch provides the output:

> Started..
Hello World
Hello World
Hello World
Hello World
Hello World

### Sketch

```cpp
// Defines
#define TX1_PIN 0
#define RX1_PIN 4

void setup() {
  // put your setup code here, to run once:
  // Initialising delay
  delay(1000); // Debug output serial
  Serial.begin(9600); // Hardware serial 1, jumpered TX1 -> RX1
  Serial1.begin(9600, SERIAL_8N1, RX1_PIN, TX1_PIN);
  Serial.println("Started..");
}

char txdata[] = "Hello World";
char buffer[16] = { 0 };

void loop() {
  
  Serial1.println(txdata);
  delay(1);
  
  char c;
  if(Serial1.available()){
    delay(10);
    while(Serial1.available()){
      c = Serial1.read();
      Serial.print(c);
    }
  }
delay(1000);
}
```

### Debug Message

```plain
n/a, code compiles
Absence of Serial out is the issue.

Version ESP32-Arduio Core 2.0.2 Serial Output:
Started..

Version ESP32-Arduio Core 1.0.6 Serial Output:
Started..
Hello World
Hello World
Hello World
Hello World
Hello World
...
```

### Other Steps to Reproduce

I also tested using 2 channels to communicate and it works, but I would like to use just 1 Serial channel like how I could in version 1.0.6.

2Channel Output:

> TX2: Hello World
RX[12]: Hello World
TX2: Hello World
RX[12]: Hello World
...

2Channel Serial code:
```
/*
 Name:    EPS32RX.ino
 Created: 8/16/2021 3:03:58 PM
 Author:  MarkYun-SikSung

 Ref: http://www.iotsharing.com/2017/05/how-to-use-serial-arduino-esp32-print-debug.html
*/

// Library includes

// Defines
#define TX1_PIN 4
#define RX1_PIN 0
#define TX2_PIN 16
#define RX2_PIN 17

// the setup function runs once when you press reset or power the board
void setup() {
  // Initialising delay
  delay(1000);
  
  // Debug output serial
  Serial.begin(9600);

  // Setup Hardware Serial
  //HardwareSerial(1);  // Setup 2 serial port, Not required?
  Serial1.begin(9600, SERIAL_8N1, RX1_PIN, TX1_PIN);  // unsigned long baud, uint32_t config=SERIAL_8N1, int8_t rxPin=-1, int8_t txPin=-1, bool invert=false, unsigned long timeout_ms = 20000UL
                            // Baud = 9600, default protocol 8bit (8 data bits, no parity, one stop bit), RXpin = 16, TXpin = 17 
  Serial2.begin(9600, SERIAL_8N1, RX2_PIN, TX2_PIN);
}

// the loop function runs over and over again until power down or reset
constexpr char txdata[] = "Hello World";
char buffer[16] = { 0 };
void loop() {
  if (Serial1.available()) {
    uint8_t read_I = Serial1.read(buffer, sizeof(txdata));
    Serial.printf("RX[%d]: ", read_I); Serial.println(buffer);
    memset(buffer, 0x00, sizeof(buffer));
  }
  else {
    Serial.print("TX2: "); Serial.println(txdata);
    Serial2.write(txdata, sizeof(txdata));
  }
  delay(100);
}
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-03-29:
Hello @msun0948, Thanks for your contribution. We will soon release version 2.0.3-RC1 (this week). Can you please wait and retest your code on it? We did some changes on Serial from version 2.0.2 which can solve your problem.

msun0948, 2022-03-29:
@VojtechBartoska 
Thanks for getting back to me.
I'll try again later then.

msun0948, 2022-03-30:
@VojtechBartoska 
The jumped serial test code works with version 2.0.3-RC1.
Thanks for your help!
I'm unsure of the correct etiquette... Do I need to "Close with comment"?

VojtechBartoska, 2022-03-30:
@msun0948 Thanks for your fast testing. I will close a issue. Good luck with your Arduino projects and if needed, don't hesitate to contribute.

closed 2022-03-30 by VojtechBartoska