espressif/arduino-esp32 #6499: cannot use variable times

codemee, 2022-03-30:
### Board

LOLIN D32, DOIT ESP32 DEVKit

### Device Description

Plain LOLIN D32, DOIT ESP32 DEVKit module. 

### Hardware Configuration

All pins not connected.

### Version

v2.0.2

### IDE Name

Arduino IDE 1.8.13

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

The value of variables times was overritten with 620773686.

### Sketch

```cpp
int times = 32;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.print("\ntimes = ");
  Serial.println(times);
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
[2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

times = 620773686
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-03-30:
you need to make it `static int times = 32;` if you want to have it initialized with particular value.

closed 2022-03-30 by me-no-dev

codemee, 2022-03-30:
Thanks. Can you explain why does it need to be static ? If it didn't named 'times',  everything works. And the same code works fine with  ESP32 Arduino Core 1.0.6.

me-no-dev, 2022-03-30:
I would guess that the variable is used (also not as static) somewhere in the SDK source. That is why it is good to keep local variables static :)

codemee, 2022-03-30:
I guess this file does the trick https://github.com/espressif/arduino-esp32/blob/371f382db7dd36c470bb2669b222adf0a497600d/tools/sdk/esp32/ld/esp32.rom.syscalls.ld