espressif/arduino-esp32 #6500: LAN8720 PHY can fail to initialize or crash. Methods to monitor or query this are inaccessible for arduino core >=2.0.0

smuellener, 2022-03-30:
### Board

ESP32 PICO D4 with LAN7820

### Device Description

Custom

### Hardware Configuration

LAN7820 connected to ESP32 via RMII in a standard way. ETH.begin() can be called without arguments.

### Version

v2.0.2

### IDE Name

Arduino IDE with VisualMicro

### Operating System

Windows 11

### Flash frequency

40Hz

### PSRAM enabled

no

### Upload speed

115200

### Description

Before arduino-esp32 v. 2.0.0 we could call the following function:

- esp_eth_smi_read()

Those functions helped us to monitor the PHY more closely as the PHY can crash during operation (e.g. due to electrostatic discarges). In case the phy crashed, we reinitialized it using

- phy_lan8720_init()

Those functions are not available anymore in arduino core version >= 2.0.0 therefore this monitoring is not possible anymore.

Please reintroduce a possibility to read the PHY registers so that we can monitor the correct operation of the PHY again. We are particularly interested in:

- Is the phy answering to register read operations generally
- Was the auto negotiation successful

### Sketch

```cpp
-
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-01:
Hello @smuellener, can you please test this on 2.0.3-RC1 too?

smuellener, 2022-04-01:
@VojtechBartoska yes this applies to 2.0.3-RC1 as well.
I will close this issue though because in the meantime we tested those scenarios and we will always get the Disconnect() event on any PHY crash due to electrostatic discharges. That is enough for us to reset the PHY in that case. So from our point of view, we do not need to access the low-level PHY registers anymore.

closed 2022-04-01 by smuellener

TD-er, 2022-04-01:
@smuellener How do you reset the Phy?

smuellener, 2022-04-01:
@TD-er  using a dedicated gpio

TD-er, 2022-04-01:
I was affraid you would say that, as my boards don't have the LAN8720 RST pin connected to a GPIO.

Does power cycle (of the LAN chip) help here on these issues?

TD-er, 2022-04-01:
Yep, I can measure the voltage on the (separate) LAN voltage circuit.
As a matter of fact, I have done this a while ago while debugging some issues with LAN.
[image]