espressif/arduino-esp32 #6502: 2.0.3 RC1 issues with  ESPAsyncWebServer Processor

dsilletti, 2022-03-30:
### Board

ESP32 Dev Module

### Device Description

DevKitC

### Hardware Configuration

SPI connected Display

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

Mac OS 12.3

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

after the upgrade to 2.0.3 RC1 the ESPAsyncWebServer Processor is not working as expected, is cutting the output in more points without any placeholders involved. Downgrading to 2.0.2 fix the issue.

I was already using the github master repo till approximately one month ago with no issues, I upgraded today to the latest master and the same code is not working anymore. Downgrading to 2.0.2 fix the issue.

### Sketch

```cpp
server.on("/status", HTTP_GET, [](AsyncWebServerRequest *request){
    if(!request->authenticate(http_username, http_password.c_str())) {
      return request->requestAuthentication();
    }
    request->send(LittleFS, "/status.html", String(), false, processor);
  }).setFilter(ON_STA_FILTER);
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

try to serve a web page (4-5Kb) with processor, it should cut parts of the output in one or more point, with no placeholders involved in that points.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

dsilletti, 2022-03-30:
I tried to revert to previous commits to find the issue.

I found the commit that is breaking the code:
Edited VFSFileImpl::read to use both read/fread https://github.com/espressif/arduino-esp32/commit/7b89b39e1087742ada6e8992159390f885c07919

please fix it or remove it.

reverting to the previous commit  905f8f2 Warns about SSP only available for ESP32 (#6455) fixed  finally the issue and the web page is served full.

me-no-dev, 2022-03-30:
@P-R-O-C-H-Y take a look please :)

@dsilletti thanks for digging in! Will get fixed!

Jason2866, 2022-03-31:
We run in issues to. See https://github.com/espressif/arduino-esp32/discussions/6507

dsilletti, 2022-04-05:
Hi @P-R-O-C-H-Y,
basically my sketch is using EspAsyncWebserver to serve html files from the LITTLEFS, using also the processor to replace some placeholders with some strings.
I think the issue is with the file read only, the output most of the times is missing big chunks, also in more parts of the html file, the missing chunks are always the same (probably according to the buffer size you configured in the PR).
Only 10-20% of the requests are successful with full html served.
I will try to provide a short sketch that does just this, with an example html page to upload to the SPIFFS/LITTLEFS partition.

P-R-O-C-H-Y, 2022-04-05:
> Hi @P-R-O-C-H-Y,
> basically my sketch is using EspAsyncWebserver to serve html files from the LITTLEFS, using also the processor to replace some placeholders with some strings.
> I think the problem is with reading the file, the output most of the time is missing large chunks, even in multiple parts of the html file, the missing chunks are always the same (probably based on the buffer size you configured in the PR).
> Only 10-20% of the requests are successful with full html served.
> I will try to provide a short sketch that does just this, with an example html page to upload to the SPIFFS/LITTLEFS partition.

@dsilletti 
Can you test your sketch with this fixes #6536 ? If you will still have some issues, you can provide some sketch for testing :)

dsilletti, 2022-04-05:
Hi @P-R-O-C-H-Y , 
I just tested it and unfortunately is not fixing the issue.

The behaviour now is different, before the chunks were just missing, now they are added,  repeated in a part of the file where they should not be.

On a 243 lines html page, before I had just 194 lines, now I have 294 lines with a repeated chunk pasted in the middle of some other lines.

P-R-O-C-H-Y, 2022-04-05:
Hi @dsilletti,
Thanks for testing. There is still work in progress on the PR. We are discussing changes here #6507 so you can read about that more. Some users are creating test cases which helps me a lot to make everything work perfectly :)

If you can provide the sketch it will be awesome :)

P-R-O-C-H-Y, 2022-04-05:
@dsilletti Updated PR, can you test if its still the same? Thank you

dsilletti, 2022-04-05:
Still the same, please try with this test sketch and test html file.

When you will see the source of the webpage served (using your last PR), from line 189  to line 230 you will find a chunk of the previous part of the html file.

I added also the OTA to the sketch so you will be also able to create the LittleFS.bin file with mklittlefs and upload it from the /ota web page.

```
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
#include <LittleFS.h>
#include <Update.h>

AsyncWebServer server(80);

// Replace with your network credentials
const char* ssid = "YOUR_SSID";
const char* password = "YOUR_PASSWORD";

// Replaces placeholder with value
String processor(const String& var){
  Serial.println(var);
  if (var == "IP_ADDRESS") return WiFi.localIP().toString();
  else if (var == "SUBNET_MASK") return WiFi.subnetMask().toString();
  else if (var == "GATEWAY") return WiFi.gatewayIP().toString();
  else if (var == "DNS") return WiFi.dnsIP().toString();
  else return String();
}

void setup() {

  Serial.begin(115200);

  // Initialize LittleFS
  if(!LittleFS.begin(true)){
    Serial.println("An Error has occurred while mounting LittleFS");
    return;
  }

  // Connect to Wi-Fi
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi..");
  }

  Serial.println(WiFi.localIP().toString());

  // Route for root / web page
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(LittleFS, "/index.html", String(), false, processor);
  });

  
  // OTA Firmware Update
  server.on("/ota", HTTP_GET, [](AsyncWebServerRequest *request) {
    Serial.println("OTA Page");
    String page = F("<form method='POST' enctype='multipart/form-data'>"
      "<input type='file' accept='.bin' name='update' >"
      "<button type='submit'>Upload</button>"
      "</form>");
    request->send(200, "text/html", page);
  });
  
  server.on("/ota", HTTP_POST, [](AsyncWebServerRequest *request) {},[](AsyncWebServerRequest *request, const String& filename, size_t index, uint8_t *data, size_t len, bool final) {
   String body;
   size_t content_len;
   if (!index){
      Serial.println("Update");
      content_len = request->contentLength();
      // if filename includes LittleFS, update the LittleFS partition
[... 227 lines cut ...]
         }
         formattedTime +=secs;
         formattedTime +=" sec";

         return formattedTime;
      }

      setInterval(function(){
         var xhttp=new XMLHttpRequest();
         xhttp.onreadystatechange=function(){
            if(this.readyState==4 && this.status==200){            
               var data=JSON.parse(this.responseText);
               for(var key of Object.keys(data)){
                  document.getElementById(key).innerHTML=data[key];
               }    
            }
         };
         xhttp.open('GET','/status.json',true);
         xhttp.send();
      },1000);
   </script>

</body>
</html>
```

P-R-O-C-H-Y, 2022-04-05:
@dsilletti  I will test that, thank you for great sketch.
I have added another commit to PR, can you test with latest changes please? TY

dsilletti, 2022-04-05:
You are welcome! Tested now, still the same issue.

P-R-O-C-H-Y, 2022-04-12:
Hi @dsilletti, i have linked PR to this issue which will fix the problems. It will revert the commit which was breaking the FS.

PR #6569 (2022-04-21): BUGFIX: FS read + speed improvements for SD
## Summary
1. Reverted changes #6456 that are breaking the FS reads.
2. Files internal buffer is now set to 4Kb by default where the recommended size cannot be read from _stat.

## Impact
Fix FS issues in 2.0.3-RC1 release.
Improve SD reading speed (approx 40-50%)

## Related links
Closes #6507 
Closes #6502 
Closes #6516
closed 2022-04-21 by me-no-dev