espressif/arduino-esp32 #6514: USB as Host examples

versamodule, 2022-04-01:
### Board

esp32-S3

### Device Description

ESP32-S3

### Hardware Configuration

NA

### Version

v2.0.2

### IDE Name

ESP-IDF

### Operating System

Win10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

1152000

### Description

The docs say "The USB host mode, you can connect devices on the ESP32, like external modems, mouse and keyboards." 
"https://docs.espressif.com/projects/arduino-esp32/en/latest/api/usb.html#usb-as-host

So where are the examples for say connecting a keyboard to the ESP32 ?

### Sketch

```cpp
na
```

### Debug Message

```plain
na
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-04-01:
https://github.com/espressif/arduino-esp32/tree/master/libraries/USB/examples

VojtechBartoska, 2022-04-01:
@pedrominatel Maybe something to be consider at docs?

versamodule, 2022-04-11:
Sorry for not closing it.

closed 2022-04-11 by versamodule