espressif/arduino-esp32 #6517: fs.open is slower as filename increments

adamb0403, 2022-04-01:
### Board

ESP32 dev module

### Device Description

Adafruit sd card module connected to ESP using default spi pins. Interfacing with SD using spi and default SD library

### Hardware Configuration

32x32 RGB matrix also connected.

### Version

latest master

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

I have files stored on SD card and they are named as follows:

1, 2, 3, ... 253, 254.

All files are 3kb in size and contain ascii characters, and alternate files are exact copies.

I noticed that as the code continued to open and close files for reading (Starting from 1, incrementing by 1, finishing at 254), the output to my display was becoming slower and slower. I pinpointed the exact cause by measuring the time taken to execute some lines for file 1 and file 254. This line was causing the issue:

image = fs.open(fname, FILE_READ);

Where fname is a number ranging from 1 - 254. It seems as the number increases it takes longer to open the file:

Time for file 254 = 74ms
Time for file 1 = 1ms

The time decreases in a uniform maner as file goes to 1. It is not caused by the filename length (ie. 254 is 3 digit, 1 is 1 digit), file 101 had a proportionate increase in open time than file 200, they were not the same.

Anyone have any idea why this is happening? It is a crucial issue in my code as the output needs to be at a constant speed and not decrease as the file increments.

### Sketch

```cpp
for (byte x=1; x<=254; x++) { // Iterate for all images on sd card
      fname = "/" + String(x); // Form image file name
      unsigned long t1 = millis();
      image = fs.open(fname, FILE_READ); //open image file for reading
      unsigned long t2 = millis();
      unsigned long time = (t2-t1);
      Serial.println("Time = " + (String) time);
```

### Debug Message

```plain
.
```

### Other Steps to Reproduce

I've tried on multple computers.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

adamb0403, 2022-04-02:
Anyone able to help?

me-no-dev, 2022-04-02:
I think that this is to be expected. The ESP is not a computer and has limited amount of memory, so it does not cache the FAT partition tables (and folders). It needs to go and find the file you want to open, which means that it needs to read quite a bit of file tables, before it get's to it. Splitting files in folders might make things faster ;)

adamb0403, 2022-04-02:
Hi @me-no-dev thanks for the reply. That makes sense. Splitting each file into its own folder hasn't helped unfortunatly. Would getting an ESP32 Wrover module with external flash to store the data on fix the issue? Is there any alternative with the hardware I currently have

me-no-dev, 2022-04-04:
not each file in it's own folder. More like 16 folders with 16 files each. Wrover will not help, you need change in strategy. I am not quite sure what your target is though

adamb0403, 2022-04-04:
I'm using the data to program GIFs on an LEd matrix display. The data for
one frame is stored in 1 file. If there's more frames or struggles to keep
a consistent frame rate for the reasons above.

Forgive me my knowledge on storage devices is not extensive, but why would
the external flash on Wroover not help?

I will try the 16x16 method asap, thanks

VojtechBartoska, 2022-04-11:
Any updates on this?

adamb0403, 2022-04-11:
Splitting files into folders as above did help make the open times consistent, albeit at a slower speed. Eventually I just ended up using the internal flash through SPIFFS as at the moment, my files are small enough to be able to use the 1.5MB available with the wroom 32e.

VojtechBartoska, 2022-05-03:
@adamb0403 so can I consider this as solved? thanks

closed 2022-05-03 by VojtechBartoska

ws2bhhh, 2023-07-15:
Having a similar problem. I have some files named 1, 2, 3....355. each file is about 1KB.
Using the internal flash.
After using millis() to get the duration of the open() operation, I have noticed a severe performance degradation when opening files named 1, 2, 3, 4, 10, 100. This is reproduced consistently on both LittleFS and SPIFFS.
Files with other names usually take 5 to 7ms to open. All of the above files take about 98ms to load on SPIFFS, while on LittleFS this number increases from 20ms to over 70ms.
I eventually solved this by using the FFat filesystem.
As these are all opened sequentially (i.e. open 1, close, then open 2 and so on), I still suspect that it has nothing to do with the number of files.