espressif/arduino-esp32 #6519: Flashing any Sketch results in SHA-256 comparison fail and sketch doesn't run

hedgemybets, 2022-04-01:
### Board

ESP32-C3-DevKitC-02U

### Device Description

Stock board with USB cable to PC running Arduino IDE.

### Hardware Configuration

No

### Version

v2.0.2

### IDE Name

Arduino IDE v1.8.19

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600 and tried 512000 and 115200

### Description

Using new Espressif board received last week. In Arduino IDE, selected ESP32C3 Dev Module under Tools/Board and compile any sketch, SHA-256 comparison fail occurs on boot and no other output.
Tried 40 MHz, various upload speeds, both QIO and DIO flash modes, same results with anything I tried.

One unusual thing I see during flashing is: "Chip is unknown ESP32-C3 (revision 3)"

### Sketch

```cpp
#define LED 2

void setup() {
    pinMode(LED, OUTPUT);
}

void loop() {
    digitalWrite(LED, HIGH);
    delay(100);
    digitalWrite(LED, LOW);
    delay(100);
    Serial.println("In Loop");
}
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x420
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x236c
SHA-256 comparison failed:
Calculated: ccb0d00bac7e84e1d90a12e4f75f4ab6c5f7e71bb209afd5819c4c9557a6db71
Expected: c9cf160580940ec7801c73b16423824e72ad12055c732e83ce66332240af42a7
Attempting to boot anyway...
entry 0x403ce000
```

### Other Steps to Reproduce

I also have ESP-IDF v4.4 installed and setting the target for esp32c3 I can build and flash the Wifi Scan example and the app runs fine.

Flash size is set for 4MBs, which was the default and what is listed in specification.

BTW, I didn't see PSRAM listed as config option in Arduino IDE, so I answered no above.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-04-01:
Please change your title to ESP32-C3 bootloop.  The SHA comparison failing here is cosmetic, and due to the way arduino compiles.
This seems to indicate a problem with the 2nd stage bootloader, as this is the behavior I get when I try to upload a program to the unsupported beta C3 dev board that I have. I forget that it doesn't work every couple weeks, and it does not give any indication why it fails :)
Make sure you have debug level set to verbose and provide any additional logging you might get from that. 
Try to [erase the flash](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#erase-flash-erase-flash-erase-region) and then reprogram- perhaps you have an error in the 2nd stage bootloader, and starting clean will fix it.

hedgemybets, 2022-04-01:
Thanks for the quick reply. I didn't call it bootloop as the ESP doesn't continuously reboot (i. e., the message doesn't continuously repeat) as in issue #6013. The boot process hangs and no further output is visible. Of course there could be a loop hanging things up in the background so I did change the title per your request.

I did try erasing the flash and it didn't change the results. Also, turning on the debug level didn't result in any more output to share. I'm just happy that I have a workaround even though I can't use the Arduino IDE integration.

Jason2866, 2022-04-02:
Please use latest Arduino core 2.0.3-rc1. GPIO handling has been fefactored.
C3 has never PSRAM so there no menu entry.
Your test sketch does work fine with my to C3 boards.
I do NOT use Arduino IDE. I have made a PlatformIO port since a official version is not available from Platformio crew. You can use in Platformio with
```
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
```

Jason2866, 2022-04-02:
I could imagine that your board is not identified as rev3. IDF libs are configured and compiled to be only compatible with boards  >= rev3. In this case boot process will stop.
In this case you can try a other Platformio Arduino Core build from me which has a modified sdkconfig (we need for our Project Tasmota) which supports older C3 boards too.
```
 platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3.zip
```

ggalisky, 2022-04-03:
I had this same issue, used the change that Jason provided for the ini file and now I get nothing on the serial monitor after uploading code. 

My ini file:
[env:arduino-esp32c3]
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
board = esp32-c3-devkitm-1
framework = arduino
platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
board_build.f_flash = 80000000L
board_build.flash_mode = dio

monitor_speed = 115200

My main.cpp:

`#include "Arduino.h"

void setup() {
    pinMode(10,INPUT);
    Serial.begin(115200);

}

void loop(){
    int HED_OUTPUT = digitalRead(10);
    if (HED_OUTPUT == false){
      Serial.print("BUTTON ON: ");
    }
    else {
      Serial.print("BUTTON_OFF: ");
    }
    Serial.println(HED_OUTPUT);
    delay(100);

}`

Jason2866, 2022-04-03:
Just place this in the ini file. 
```
[env:arduino-esp32c3]
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
board = esp32-c3-devkitm-1
framework = arduino
monitor_speed = 115200
```

ggalisky, 2022-04-03:
> 

Even after changing my ini to this I still get a blank serial monitor. The same sketch on the Arduino IDE works with this board so I know it is a platform.io issue. Thank you for getting back to me quickly

Jason2866, 2022-04-03:
Just deleted folder `.platformio` and used this `platformio.ini`
```
[env:esp32c3]
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
board = esp32-c3-devkitm-1
framework = arduino

```
and this sketch (my board has the Led on GPIO 3)
```
#include <Arduino.h>

#define LED 3

void setup() {
    pinMode(LED, OUTPUT);
}

void loop() {
    digitalWrite(LED, HIGH);
    delay(100);
    digitalWrite(LED, LOW);
    delay(100);
    Serial.println("In Loop");
}
```
LED is blinking, and no serial message

So sketch is working in general. I have to look what is different regarding Serial ArduinoIDE setup and current PlatformIO setup

Jason2866, 2022-04-03:
I highly doubt it was working in Arduino IDE since there is ` Serial.begin(115200);` missing before `Serial.println("In Loop");`
Adding this and i get
```
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: a9753a4fc647c6545c1b919ef08db429130a48592727edca270f1e5a3da0d0a9
Expected: 3bf6ef2cf3b9eefcd4b3c70cc5d1ce5138292d101a5cb1d5db6fbebf081b0a19
Attempting to boot anyway...
entry 0x403ce000
In Loop
In Loop
In Loop
In Loop
In Loop
In Loop
.
.
.
```

Jason2866, 2022-04-03:
So there is NO issue at all.

hedgemybets, 2022-04-03:
I've retested and all is working fine now with the Arduino IDE, so I'm closing the issue. I believe my test code was missing the Serial.begin(115200) line as @Jason2866 noted above for @ggalisky's issue. So I wasn't seeing any output. Also, I do know that my larger project code did have Serial.begin(9600), but my monitor was probably in 115200. Usually when I do this I see some garbage characters which would remind me to change the baud rate, but in this case I didn't see any.

Thanks for the help!

closed 2022-04-03 by hedgemybets

ggalisky, 2022-04-03:
> I've retested and all is working fine now with the Arduino IDE, so I'm closing the issue. I believe my test code was missing the Serial.begin(115200) line as @Jason2866 noted above for @ggalisky's [...]

I had the Serial.begin(XXXXX) in my first post though... and still have no output on the Serial monitor for Platform.io. It does work on the Arduino IDE which makes me think it has something to do with "USB CDC on boot" option that is enabled in the Arduino IDE. When I turn this option off in the Arduino IDE the serial monitor stops working.

Jason2866, 2022-04-03:
This is related https://github.com/espressif/arduino-esp32/issues/6089#issuecomment-1077607496

Sys64736, 2022-04-11:
I have a C3 with no UART and USB connected directly to 18 and 19
I'll get no Serial output unless I have this in the .ini

  -D ARDUINO_USB_MODE=1
  -D ARDUINO_USB_CDC_ON_BOOT=1

Then Serial output works great.

Also using @Jason2866 git link for the platform and using this latest repo for package

QGB, 2022-05-15:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 4ab8657f6f5aa80b81c7dd649db3bd64a98999ed7c65acc5a2ab77ce80caf525
Expected: 6babbc7b9ea63e04e0fc89ec224f6268aa3f84392018f73ab33721f768ce2eb6
Attempting to boot anyway...
entry 0x403ce000
Hello world!
This is esp32c3 chip with 1 CPU core(s), WiFi/BLE, silicon revision 3, 4MB external flash
Minimum free heap size: 320128 bytes
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
Restarting in 3 seconds...
Restarting in 2 seconds...

```