espressif/arduino-esp32 #6524: esp-idf compiler error

nimaltd, 2022-04-03:
### Board

Nodemcu-32s

### Device Description

on httpsotaupdate.h file. please add '#ifndef .... #endif'

### Hardware Configuration

any

### Version

v2.0.2

### IDE Name

eclipse

### Operating System

win10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

115200

### Description

can not compile

### Sketch

```cpp
#include "httpsotaupdate.h"
```

### Debug Message

```plain
error
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-04-03:
Arduino 2.0.3-rc1 as IDF component does compile without any error with IDF4.4
Please use actual version. Old core versions will be not updated if issues occour.

VojtechBartoska, 2022-04-04:
Thanks for clear explanation @Jason2866! I'm closing this one, @nimaltd if you need you can reopen.

closed 2022-04-04 by VojtechBartoska