espressif/arduino-esp32 #6525: Continues reboot when enabling PS-RAM

rezaneam, 2022-04-03:
### Board

ESP32-PICO-MINI-02

### Device Description

ESP32-PICO-DevKitM-2

### Hardware Configuration

[env:pico32]
platform = espressif32
board = pico32
framework = arduino
monitor_speed = 115200
upload_port = COM3
build_flags = 
	 -DBOARD_HAS_PSRAM
	 -mfix-esp32-psram-cache-issue

### Version

v1.0.6

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

default

### PSRAM enabled

yes

### Upload speed

default

### Description

I am trying to enable the internal PS-RAM in ESP32-PICO-V3-02, but the chip keeps rebooting with the following error code.

### Sketch

```cpp
ESP.getFreePsram()
```

### Debug Message

```plain
E (133) psram: Not a valid or known package id: 6
abort() was called at PC 0x40085653 on core 0

ELF file SHA256: 0000000000000000
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-04-03:
Pico does not need the PSRAM fix since it is a fairly new chip and does not have the PSRAM bug.
Arduino Core 1.0.6 does NOT support this variant. You have to use actual core 2.0.3-rc1
There is no official version (yet) for Platformio for this core version.
You can use my inofficial Platformio build for using the latest core.
```
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
```

VojtechBartoska, 2022-04-07:
Hello @rezaneam, 1.0.6 version is quite old and not supported anymore, as mentioned above, please test your code on 2.0.3-rc1.

rezaneam, 2022-04-07:
> Pico does not need the PSRAM fix since it is a fairly new chip and does not have the PSRAM bug. Arduino Core 1.0.6 does NOT support this variant. You have to use actual core 2.0.3-rc1 There is no [...]

Thanks that works perfectly. Probably need to be on this branch for a while :)

VojtechBartoska, 2022-04-08:
@rezaneam Can I consider this as closed?

closed 2022-04-08 by rezaneam