espressif/arduino-esp32 #6526: ESP32-CAM test crashes in 2.0.2-  OK in 1.0.6

fanfanlatulipe26, 2022-04-03:
### Board

ESP32-CAM 

### Device Description

"Simple" (?) ESP32-CAM module here from [Banggood](https://www.banggood.com/ESP32-CAM-WiFi-+-bluetooth-Camera-Module-Development-Board-ESP32-With-Camera-Module-OV2640-Geekcreit-for-Arduino-products-that-work-with-official-Arduino-boards-p-1394679.html?imageAb=1&cur_warehouse=CN&akmClientCountry=FR)

### Hardware Configuration

No other connection.
Powered by FTDI on an USB port of the PC or with an external 5v power supply. (FTDI only for upload /debug)

### Version

v2.0.2

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

When compiled with release 1.0.6 the test case run as expected.
With exactly the same hardware (same PC, same FTDD, cable ....;) but with release 2.0.2 it crashes very often, with ramdom (??) causes.

The code is exactly  this one from [ a tutorial at randomnerdtutorials.com](https://randomnerdtutorials.com/esp32-cam-take-photo-display-web-server/)
Raw code is [here](https://raw.githubusercontent.com/RuiSantosdotme/ESP32-CAM-Arduino-IDE/master/ESP32-CAM-Take-Photo-and-Display-Web-Server/ESP32-CAM-Take-Photo-and-Display-Web-Server.ino).
Small websever (using ESPAsyncWebServer)  with a web page allowing to take a picture (always under the same name), save it in SPIFFS and display it on demand.
(I am aware of the need of  config.grab_mode = CAMERA_GRAB_LATEST; for this kind of application in 2.02) 

The original demo turns off the brownout detection, but even with the brownout detection enabled the results are bad and there is no explicit crash for brownout detection. Does not seem to be a power supply problem. Tests were also done wih external power supply.

Same test adapted to use the SD card instead of the SPIFFS run perfectly in 2.0.2
For this adaptation I just changed the #include and the .begin
```
//#include <SPIFFS.h>
#include <FS.h>
#include "SD_MMC.h"            // SD Card ESP32 + fs ++++++++++++++++++++++++++++++
#define SPIFFS SD_MMC         // SD Card ESP32 + fs +++++++ and change the .begin(  )

and 

// if (!SPIFFS.begin(true)) {
  if (!SPIFFS.begin("/sdcard",true)) {    //  +++++ changed for SD . Prevent use of pin 4 the LED
```
ESPAsyncWebServer version 1.2.3
Code compiled for AI Thinker ESP32-CAM board or ESP32 Dev Kit with psram enabled. Same results.

### Sketch

```cpp
/*********
  Rui Santos
[... 408 lines cut ...]
mode:DIO, clock div:1
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13508
load:0x40080400,len:3604
entry 0x400805f0
Connecting to WiFi...
Connecting to WiFi...
Connecting to WiFi...
Connecting to WiFi...
Connecting to WiFi...
SPIFFS mounted successfully
IP Address: http://192.168.0.14
```

### Other Steps to Reproduce

Test OK with 1.0.6
Test OK in 2.0.2 if SD card is used instead of SPIFFS
In fact I see other crashes than the ones reported here, but could not reproduce at capture time   :-<
Crashes not the same with debug level none. (see the 2 logs , separted with ++++++++++++++++++)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

fanfanlatulipe26, 2022-04-03:
Simpler  testcase with no wifi webserver.  Only take a picture, save it in SPIFFS
```
// Loop: take a photo, save it in SPIFFS. No WIFI
//
//  ESP32 Dev Kit, psram enabled, partition Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)
// or AI Thinker ESP32-CAM
/*********
  Rui Santos
  Complete project details at https://RandomNerdTutorials.com/esp32-cam-take-photo-display-web-server/

  IMPORTANT!!!
   - Select Board "AI Thinker ESP32-CAM"
   - GPIO 0 must be connected to GND to upload a sketch
   - After connecting GPIO 0 to GND, press the ESP32-CAM on-board RESET button to put your board in flashing mode

  The above copyright notice and this permission notice shall be included in all
  copies or substantial portions of the Software.
*********/

#include "esp_camera.h"
//#include "soc/soc.h"           // Disable brownour problems
//#include "soc/rtc_cntl_reg.h"  // Disable brownour problems
#include <SPIFFS.h>
#include <FS.h>

int count = 0;

// Photo File Name to save in SPIFFS
#define FILE_PHOTO "/photo.jpg"

// OV2640 camera module pins (CAMERA_MODEL_AI_THINKER)
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27
#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

void setup() {
  // Serial port for debugging purposes
  Serial.begin(115200);
  Serial.println("Formating SPIFFS");
  if (!SPIFFS.format()) {
    Serial.println("An Error has occurred while formating SPIFFS");
    ESP.restart();
  }
  else {
    delay(500);
    Serial.println("SPIFFS formated successfully");
  }

  if (!SPIFFS.begin(true)) {
    Serial.println("An Error has occurred while mounting SPIFFS");
    ESP.restart();
  }
  else {
    delay(500);
    Serial.println("SPIFFS mounted successfully");
  }

[... 142 lines cut ...]
ho 0 tail 12 room 4
load:0x40078000,len:13508
load:0x40080400,len:3604
entry 0x400805f0
Formating SPIFFS
SPIFFS formated successfully
SPIFFS mounted successfully
------ -Test nbr 1
Taking a photo...
Picture file name: /photo.jpg
E (13148) cam_hal: EV-EOF-OVF
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 

Core  1 register dump:
PC      : 0x400ea768  PS      : 0x00060035  A0      : 0x80085f38  A1      : 0x3ffbee0c  
A2      : 0x3ffb96f0  A3      : 0x00000000  A4      : 0x3ffbee44  A5      : 0x3ffb23e0  
A6      : 0x00000006  A7      : 0x3ffc240c  A8      : 0x80085ecf  A9      : 0x3ffbeddc  
A10     : 0x3ffb96f0  A11     : 0x3ffbee18  A12     : 0x3ffbee44  A13     : 0x00000000  
A14     : 0x3ffb3678  A15     : 0x3ffb36c4  SAR     : 0x00000018  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x40087b7a  LEND    : 0x40087b85  LCOUNT  : 0x00000000  

Backtrace:0x400ea765:0x3ffbee0c |<-CORRUPTED

ELF file SHA256: 0000000000000000
```

VojtechBartoska, 2022-04-04:
Hello. is the error the same for 2.0.3-RC1?

Jason2866, 2022-04-04:
The test sketch is not working.
```
src/main.cpp: In function 'void loop()':
src/main.cpp:117:5: error: 'capturePhotoSaveSpiffs' was not declared in this scope
     capturePhotoSaveSpiffs();
     ^~~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/esp32cam/src/main.cpp.o] Error 1
```

fanfanlatulipe26, 2022-04-04:
Don't know how to install   2.0.3-RC1
May I have both version 2.0.2 and  2.0.3-RC1 at the same time and an easy way to switch from one to the other ?

me-no-dev, 2022-04-04:
decoding those exceptions can also help much. (And it is a requirement) https://github.com/me-no-dev/EspExceptionDecoder

fanfanlatulipe26, 2022-04-04:
Well: there is no stack in the output, just  "Backtrace:0x400ea765:0x3ffbee0c |<-CORRUPTED"

fanfanlatulipe26, 2022-04-04:
@Jason2866 : no problem in Arduino IDE.
Maybe just need to add the function prototype before

Jason2866, 2022-04-04:
In general there is no problem with the webcam. We have a webcam build with [Tasmota](https://arendst.github.io/Tasmota-firmware/) and it worked never so stable as with core 2.0.x

fanfanlatulipe26, 2022-04-04:
Test done with 2.0.3-RC1:
- small testcase with no wifi / webserver runs perfectly.
- test case with the webserver runs also pretty well, but many messages in the log, such as
cam_hal: EV-EOF-OVF
or 
cam_hal: EV-VSYNC-OVF
Doesn't occur in 1.0.6
The program needs to take 2 or more picture in order to get one really stored in the SPIFFS. Don't understand. 

Doesn't occure if the picture is save on the SD card instead of the SPIFFS

Here is a log (no more information if compiled with debug level "debug")
```
Connecting to WiFi...
Connecting to WiFi...
[  5457][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  5457][D][WiFiGeneric.cpp:914] _eventCallback(): STA IP: 192.168.0.14, MASK: 255.255.255.0, GW: 192.168.0.254
Connecting to WiFi...
SPIFFS mounted successfully
IP Address: http://192.168.0.14
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
Taking a photo...
Picture file name: /photo.jpg
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-EOF-OVF
The picture has been saved in /photo.jpg - Size: 0 bytes
Taking a photo...
Picture file name: /photo.jpg
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-EOF-OVF
The picture has been saved in /photo.jpg - Size: 0 bytes
Taking a photo...
Picture file name: /photo.jpg
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
The picture has been saved in /photo.jpg - Size: 105344 bytes
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
Taking a photo...
Picture file name: /photo.jpg
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-VSYNC-OVF
The picture has been saved in /photo.jpg - Size: 0 bytes
Taking a photo...
Picture file name: /photo.jpg
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
The picture has been saved in /photo.jpg - Size: 0 bytes
Taking a photo...
Picture file name: /photo.jpg
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
The picture has been saved in /photo.jpg - Size: 0 bytes
Taking a photo...
Picture file name: /photo.jpg
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
The picture has been saved in /photo.jpg - Size: 117248 bytes
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF

```

VojtechBartoska, 2022-04-07:
I just see you mentioned  ESPAsyncWebServer problems, it can be probably related to this issue? #6502?

fanfanlatulipe26, 2022-04-07:
The mix SPIFFS +  ESPAsyncWebServer  have a strange behavior.

closed 2025-01-16 by Parsaabasi