espressif/arduino-esp32 #6528: Arduino Core 2.0.2 does not support TLS 1.3 --> WiFiClientSecure

martinius96, 2022-04-03:
### Board

ESP32 Devkit V1 DOIT

### Device Description

There is connected HC-SR04 ultrasonic sensor that have no impact on problem.

### Hardware Configuration

D22  -->  Trigger
D23  -->  Echo
3V3  -->  Vcc
GND --> GND

### Version

v2.0.2

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10 Education x64

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I have developed code in January for ESP32 that is using HTTPS connection via WiFiClientSecure library.
It is doing measurements using HC-SR04 or JSN-SR04T ultrasonic sensor of water level height. Datas arent sent to webserver that is freehosting hosted at 000webhost --> [https://hladinomer.000webhostapp.com/](https://hladinomer.000webhostapp.com/)
I am not sure what version of Arduino Core I was using at development, maybe it was 2.0.2, but not sure...

Sketch is using FreeRTOS, there are other versions of sketch without FreeRTOS, but same problem. In FreeRTOS sketch there is used Queue. When datas are measured, they are pushed to Queue, and HTTPS connection task will read them and will execute once to store data in webserver database.

In similar time as I developed Arduino Core, I have created identical sketch (from point of how it works) in ESP-IDF framework 4.4 on mbed_tls example and it is working without problem for months... Right now I am using Arduino Core 2.0.2 - latest stable release. And code that was originally working in January is not working now. During SSL handshake there is ssl_client return -1, so certificate is not verified. Root CA certificate is same ([DigiCert Global Root CA](https://global-root-ca.chain-demos.digicert.com/)), also website certificate is valid for more than +120 days to go. So, nothing changed on that side for Secure client... Source code is correct.

**Core 2.0.2 output with Core Debug level Verbose - Connection not working:**
[image]
I have tried WiFiClientSecure example and it is working good. When i compared both websites, there is difference at TLS version that is used on webservers. 000webhost is on subdomains using TLS 1.3 and 1.2 and howsmyssl.com (from WiFiClientSecure example) is using TLS 1.2. 

So i think there might be problem with TLS 1.3 connection. Also in WiFiClientSecure of actual 2.0.2 release branch there is comment: "Running on TLS 1.2 using mbedTLS". **Also I am not sure if 000webhost had TLS 1.3 before or it was added in recent days or months, I don't know how to find that out...** Is it possible to force 1.2 TLS connection from side of ESP32?

**I tried to downgrade Arduino Core to 1.0.6. Then I tried same sketch and connection to 000webhost is working normally and datas are successfully sent to server where are stored in database.** So I am a bit confused... I have tried to study documentation and also some Gitter threads, but nothing relevant I have found... **So what should be wrong?**

There is Wokwi simulator with lightweight sketch with one task that is trying to sent example datas to webservers without ultrasonic measurement - no luck on connection: https://wokwi.com/projects/327948646817464914

Core 1.0.6 output with Core Debug level Verbose - Connection WORKING! :
[image]

### Sketch

```cpp
/*|-----------------------------------------------------------------------------------|*/
/*|Projekt: Hladinomer - HTTPS - FreeRTOS - HC-SR04 / JSN-SR04T / HY-SRF05            |*/
/*|ESP32 (DevKit, Generic)                                                            |*/
/*|Autor: Martin Chlebovec (martinius96)                                              |*/
/*|E-mail: martinius96@gmail.com                                                      |*/
[... 162 lines cut ...]
      }
      String line = client.readStringUntil('\n');
    } else {
      Serial.println(F("Connection to webserver was NOT successful"));
    }
    client.stop();
  }
}
```

### Debug Message

```plain
Copy from UART monitor is not working, I don't know why, I can just copy [ character...
I don't have ExceptionDecoder.
```

### Other Steps to Reproduce

I tried Wokwi simulator, output is same.
Link provided in Description. After downgrading Arduino Core to 1.0.6, connection is working without problem.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

martinius96, 2022-04-03:
**Edit 3 hours later:**
I tried to import .json of development branches to my Arduino IDE.
Then in Board Manager I have used 2.0.3-RC1. And connection is working!
There is some SSL related error, but request was done okay and datas were written into MySQL database.

```diff
So tell me, how Arduino Core 2.0.2 can be based on ESP-IDF 4.4, if in ESP-IDF 4.4 it is WORKING but Arduino Core not...
```

**In changelog of 2.0.3-RC1 there is:** 
"Support the updated MbedTLS in ESP-IDF v4.4 by @me-no-dev in https://github.com/espressif/arduino-esp32/pull/6243".
**Then why that wasn't uploaded in 2.0.2 STABLE version that is from v4.4 ESP-IDF?** You guys make me wrinkle and few hairs lost...
[image]

Core 2.0.3-RC1 output with Core Debug level Verbose - Connection WORKING! :
[image]

VojtechBartoska, 2022-04-04:
Hello @martinius96, Arduino ESP32 Core v2.0.3-RC1 is based on ESP-IDF v4.4 and there have been done some fixes in meantime what can explain this behaviour.

We are now waiting for ESP-IDF v.4.4.1 which includes more bug fixes.

I'm sorry that it brings you some confusion.

In summary, under v2.0.3-RC1 everything is working correctly?

martinius96, 2022-04-04:
Thanks for reply, yes, under v2.0.3-RC1 WiFiClientSecure client working good, there is just that _handle_error() output -76 UNKNOWN ERROR CODE (004C), but request was successful and data written into MySQL database.

**That v2.0.2 release was based on ESP-IDF v4.4-beta1, so that caused problem I think, because it wasn't from release ESP-IDF version.** I haven't tried other things such as Bluetooth, UDP connections at WiFi or even buses, so I am not sure if there can be other things that are not working.

**I have found out, sketch was originally developed and working under v2.0.1 release, that was based on stable ESP-IDF v4.4, not beta.**

VojtechBartoska, 2022-04-07:
@martinius96 Your specific issue can be solved by backports. Do you need more help on this? Seems sticking to 2.0.3 is a solution for this :)

martinius96, 2022-04-07:
Yes, 2.0.3 working great, we can close this issue, solved.

closed 2022-04-07 by martinius96

CCarrillo07, 2024-08-20:
Hi all, I'm using esp32 by Espressif Systems version 2.0.3, I'm sending a GET request to www.howsmyssl.com.

I'm getting this response:

Response:
{"given_cipher_suites":["TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384","TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384","TLS_DHE_RSA_WITH_AES_256_GCM_SHA384","TLS_ECDHE_ECDSA_WITH_AES_256_CCM","TLS_DHE_RSA_WITH_AES_256_CCM","TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA384","TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384","TLS_DHE_RSA_WITH_AES_256_CBC_SHA256","TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA","TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA","TLS_DHE_RSA_WITH_AES_256_CBC_SHA","TLS_ECDHE_ECDSA_WITH_AES_256_CCM_8","TLS_DHE_RSA_WITH_AES_256_CCM_8","TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256","TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256","TLS_DHE_RSA_WITH_AES_128_GCM_SHA256","TLS_ECDHE_ECDSA_WITH_AES_128_CCM","TLS_DHE_RSA_WITH_AES_128_CCM","TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA256","TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256","TLS_DHE_RSA_WITH_AES_128_CBC_SHA256","TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA","TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA","TLS_DHE_RSA_WITH_AES_128_CBC_SHA","TLS_ECDHE_ECDSA_WITH_AES_128_CCM_8","TLS_DHE_RSA_WITH_AES_128_CCM_8","TLS_RSA_WITH_AES_256_GCM_SHA384","TLS_RSA_WITH_AES_256_CCM","TLS_RSA_WITH_AES_256_CBC_SHA256","TLS_RSA_WITH_AES_256_CBC_SHA","TLS_ECDH_RSA_WITH_AES_256_GCM_SHA384","TLS_ECDH_RSA_WITH_AES_256_CBC_SHA384","TLS_ECDH_RSA_WITH_AES_256_CBC_SHA","TLS_ECDH_ECDSA_WITH_AES_256_GCM_SHA384","TLS_ECDH_ECDSA_WITH_AES_256_CBC_SHA384","TLS_ECDH_ECDSA_WITH_AES_256_CBC_SHA","TLS_RSA_WITH_AES_256_CCM_8","TLS_RSA_WITH_AES_128_GCM_SHA256","TLS_RSA_WITH_AES_128_CCM","TLS_RSA_WITH_AES_128_CBC_SHA256","TLS_RSA_WITH_AES_128_CBC_SHA","TLS_ECDH_RSA_WITH_AES_128_GCM_SHA256","TLS_ECDH_RSA_WITH_AES_128_CBC_SHA256","TLS_ECDH_RSA_WITH_AES_128_CBC_SHA","TLS_ECDH_ECDSA_WITH_AES_128_GCM_SHA256","TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA256","TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA","TLS_RSA_WITH_AES_128_CCM_8","TLS_EMPTY_RENEGOTIATION_INFO_SCSV"],"ephemeral_keys_supported":true,"session_ticket_supported":true,"tls_compression_supported":false,"unknown_cipher_suite_supported":false,"beast_vuln":false,"able_to_detect_n_minus_one_splitting":false,"insecure_cipher_suites":{},"**tls_version":"TLS 1.2**","rating":"Probably Okay"}
Connection closed
 
I would like that esp32 supports TLS 1.3, Do you know how can I achieve this?