espressif/arduino-esp32 #6529: ESP32-C3-MINI Serial monitor stays blank when uploading with Platform.io, but not with Arduino IDE

ggalisky, 2022-04-04:
### Board

ESP32-C3-MINI Custom board

### Device Description

A ESP32-C3-MINI module with hand soldered wires and decoupling caps (see diagram below)

### Hardware Configuration

Hand wired module with below schematic

[image]

### Version

other

### IDE Name

Platform.io

### Operating System

Windows and Mac

### Flash frequency

80,000,000

### PSRAM enabled

no

### Upload speed

115200

### Description

Serial monitor is completely blank when uploading main.cpp when using Platform.io. The issue does not persist when using the Arduino IDE. The main.cpp runs fine after flashing, but I get no data on the serial monitor from platform.io

I believe it may have something to do with the "USB CDC" option. When I disable this in the Arduino IDE I get no serial output, when I enable it I get serial output. I've tried adding this to my platform.ini file but have not had any luck getting the serial monitor to work with PIO

### Sketch

```cpp
#include "Arduino.h"
void setup() {
    Serial.begin(115200);

}
void loop() {
    Serial.println("I'm alive");
    delay(100);
}
```

### Debug Message

```plain
All I see on the serial monitor: 

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/cu.usbmodem21401  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
```

### Other Steps to Reproduce

I've tried both MacOS and Windows and haven't been able to solve this. Been plugging away at it for the last few days. 

I've checked every thread I could find, but none have solved my issue. Thank you for your help

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-04-04:
If it works in Arduino IDE, but not in platform.io, the issue should be moved to https://github.com/platformio/platform-espressif32/issues

Jason2866, 2022-04-04:
It is 100% Platformio!
The reason is that the Serial Console after flashing is on standard UART (serial) and NOT on USBSerial (GPIO 18 /19)
anymore. I will have a look if i find a solution

Jason2866, 2022-04-04:
Update :-) Board added
```
[env:esp32c3-mini]
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
board = esp32-c3-mini
framework = arduino
monitor_speed = 115200
```

ggalisky, 2022-04-04:
This solved my issue, I also posted in https://github.com/platformio/platform-espressif32/issues as per @Ibernstone suggestion, Jason responded with the same solution and I can confirm it is working on my end: https://github.com/platformio/platform-espressif32/issues/755

closed 2022-04-04 by ggalisky

derFrickler, 2024-01-10:
Hi There, i am trying to get the USB Serial working on my ESP32-C3 Mini in platformio, but i am not able to get a Serial device after flashing and rebooting. Flashing works fine, but after reboot there is no serial device coming up. Using Arduino IDE 2 with USB CDC on Boot Enabled it works just fine. 

Seems that i am missing something?

here is my config: 

>   [env:esp32c3]
>   platform = espressif32
>   board = esp32-c3-devkitm-1
>   framework = arduino
>   monitor_speed = 115200
>   build_flags =
> 	-DARDUINO_USB_MODE=1 
> 	-DARDUINO_USB_CDC_ON_BOOT=1

Any hints on what i can try? As said, it works with Arduino but i prefer pio for all my projects.