espressif/arduino-esp32 #6530: Connecting esp32-c3 to google cloud iot core

kargarisaac, 2022-04-04:
### Board

ESP32-C3

### Device Description

ESP32-C3 devkit

### Hardware Configuration

No additional configuration to the devkit

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I try to connect my esp32-c3 to google cloud iot core via [this library](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/tree/master). I can do it using normal esp32 devkit but when I use my new esp32-c3 devkit, I get an error.
I can use normal examples from MQTT library. It seems a problem with gcc but I'm not an expert in C/C++. Do you have any idea how can I solve this?

Thank you

### Sketch

```cpp
I use this code: 
https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/tree/master/examples/Esp32-lwmqtt
```

### Debug Message

```plain
Arduino: 1.8.19 (Windows 10), Board: "ESP32C3 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 160MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, Verbose"

C:\Users\Isaac\Documents\Arduino\libraries\google-cloud-iot-arduino-master\src\crypto\nn.cpp:12: warning: "assert" redefined
 #define assert(condition) do{} while(0)
 
In file included from c:\users\isaac\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\sys-include\sys\reent.h:503,
                 from C:\Users\Isaac\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/newlib/platform_include/sys/reent.h:17,
                 from c:\users\isaac\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\sys-include\string.h:11,
                 from c:\users\isaac\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\cstring:42,
                 from C:\Users\Isaac\Documents\Arduino\libraries\google-cloud-iot-arduino-master\src\crypto\nn.h:23,
                 from C:\Users\Isaac\Documents\Arduino\libraries\google-cloud-iot-arduino-master\src\crypto\nn.cpp:8:
C:\Users\Isaac\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/newlib/platform_include/assert.h:50: note: this is the location of the previous definition
 #define assert(__e) (__builtin_expect(!!(__e), 1) ? (void)0 : __assert_func (__FILENAME__, __LINE__, \
 
c:/users/isaac/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\google-cloud-iot-arduino-master\CloudIoTCoreMqtt.cpp.o: in function `CloudIoTCoreMqtt::startMQTTAdvanced()':
C:\Users\Isaac\Documents\Arduino\libraries\google-cloud-iot-arduino-master\src/CloudIoTCoreMqtt.cpp:168: undefined reference to `messageReceivedAdvanced(MQTTClient*, char*, char*, int)'
c:/users/isaac/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\Isaac\Documents\Arduino\libraries\google-cloud-iot-arduino-master\src/CloudIoTCoreMqtt.cpp:169: undefined reference to `messageReceivedAdvanced(MQTTClient*, char*, char*, int)'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32C3 Dev Module.

```
```

### Other Steps to Reproduce

Just set the config for wifi and certificates and private keys.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-04:
Hello @kargarisaac, it seems to me more like a problem with that particular Library. Did you consider opening an issue in their repo?

kargarisaac, 2022-04-04:
The problem is solved. Yes it was the library. I close this.

closed 2022-04-04 by kargarisaac