espressif/arduino-esp32 #6532: Error compiling for board ESP32C3 Dev Module.

Skywalkerf34, 2022-04-04:
### Board

ESP32C3DEV

### Device Description

many devices

### Hardware Configuration

i2c, sensorS, OLED....

### Version

latest master

### IDE Name

arduino 1.8.19

### Operating System

WIN10

### Flash frequency

40MHZ

### PSRAM enabled

no

### Upload speed

115200

### Description

I work on this sketch from Nov. It was compiling good up to nowsince

### Sketch

```cpp
no change
```

### Debug Message

```plain
loading hardware from d:\Users\FidL\Documents\Arduino\hardware: loading package espressif: looking for boards.txt in D:\Users\FidL\Documents\Arduino\Hardware\espressif\esp8266\boards.txt: CreateFile D:\Users\FidL\Documents\Arduino\Hardware\espressif\esp8266\boards.txt: Data error (cyclic redundancy check).
```

### Other Steps to Reproduce

WAS WORKING FINE PREVIOUSLY

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-04-04:
logs say ESP8266. You are doing something wrong :)

Skywalkerf34, 2022-04-05:
Whaou, that's a great help ;-)
By the end, the pb come from my HDD that crash some cluster :-(
I temporary fix with a chkdsk /F but will order a new HDD