espressif/arduino-esp32 #6533: Response from a POST request is nowhere to be found

RensMvP, 2022-04-04:
### Board

WT32-ETH01

### Device Description

none

### Hardware Configuration

nothing.

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

I am sending a POST request to pair with a https server on a insecure line to get the timestamp and auth key.
On this Post request the Https server response with the timestamp and auth key, but the httpclient library does not read anything from a POST request

### Sketch

```cpp
httpSecure.setURL(("/pair"));
httpCode = httpSecure.POST(json);
String payload = httpSecure.getString();
Serial.println(payload);
```

### Debug Message

```plain
Sending pair request to https
[D][HTTPClient.cpp:579] sendRequest(): request type: 'POST' redirCount: 0

[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[D][HTTPClient.cpp:1125] connect():  connected to IP:port
[D][HTTPClient.cpp:1257] handleHeaderResponse(): code: 200
[D][HTTPClient.cpp:1260] handleHeaderResponse(): size: 169
[D][HTTPClient.cpp:603] sendRequest(): sendRequest code=200
[D][HTTPClient.cpp:1394] writeToStreamDataBlock(): connection closed or file end (written: 169).
[D][HTTPClient.cpp:385] disconnect(): tcp keep open for reuse
```

### Other Steps to Reproduce

Tried to read the headers and set the arduino IDE in Debug. The debug mode shows that the size of the header is greater then 1 (so i assume that the data is in the header), but can't reach it.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

martinius96, 2022-04-07:
Post() function of HTTPClient class is overloading, so you call it in program with String payload, then it is overloaded and in result it will call http.sendRequest("POST", payload, size); with POST parameter. It is similar with PUT(payload), UPDATE(payload) functions that will use PUT or UPDATE parameter of request in http.sendRequest().

`int HTTPClient::POST(String payload)`
`{`
`    return POST((uint8_t *) payload.c_str(), payload.length());`
`}`

`int HTTPClient::POST(uint8_t * payload, size_t size)`
`{`
`    return sendRequest("POST", payload, size);`
`}`

RensMvP, 2022-04-07:
thank you for the response, i already went to use the http.sendRequest function. The result ended up that it worked that way and the program got quicker as before

VojtechBartoska, 2022-04-11:
Can we close this then? @RensMvP

RensMvP, 2022-04-12:
@VojtechBartoska  yes this issue can be closed. My program works more efficient by using the SendRequest function, so i am using that function. 

A small note thing that is still a littlebit weird but not a problem is that my code got twice as fast using the sendrequest function, instead of the post and get functions.