espressif/arduino-esp32 #6534: Cludge needed to enable both IDF & Arduino log-statements

ankostis, 2022-04-04:
### Board

ESP32 dev (but irrelevant)

### Device Description

[Freematics](https://freematics.com/products/freematics-one-plus-model-h/) but HW is irrelevant

### Hardware Configuration

HW is irrelevant

### Version

latest master

### IDE Name

PlatformIO

### Operating System

Debian SID

### Flash frequency

irrelevant

### PSRAM enabled

no

### Upload speed

irrelevant

### Description

I want to enable logging-statements with `USE_ESP_IDF_LOG` build_flag for **both** 
* ESP_IDF (like `ESP_LOGI(<tag>, format, ...)`), and 
* Arduino ones (like `log_i(format, ....)`.

I've tried both 2.0.2 & 2.0.3-RC1, and i manged to have both log-statements enabled
only under the latest version 2.0.3-RC1, \
reporting an issue discovered rlated to #6351 and my trouble setting log-levels afterwards.

## arduino-esp32-v2.0.2

Under v2.0.0, as soon as i enabled this in my `platformio.ini`:
```
 build_flags=-DUSE_ESP_IDF_LOG 
 ```
compilation broke due to errors in all `log_x()` function calls, eg:

```
~/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-log.h:161:32: error: 'log_to_esp' was not declared in this scope
 #define log_e(format, ...) do {log_to_esp(TAG, ESP_LOG_ERROR, format, ##__VA_ARGS__);}while(0)
                                ^~~~~~~~~~
~/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:1230:9: note: in expansion of macro 'log_e'
         log_e("DNS Failed for %s", aHostname);
         ^~~~~
~/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-log.h:161:32: note: suggested alternative: 'get_sp'
 #define log_e(format, ...) do {log_to_esp(TAG, ESP_LOG_ERROR, format, ##__VA_ARGS__);}while(0)
                                ^~~~~~~~~~
~/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:1230:9: note: in expansion of macro 'log_e'
         log_e("DNS Failed for %s", aHostname);
         ^~~~~
```

Didn't spent time digging further, tried the latest release.

## arduino-esp32-v2.0.3-RC1

Under the latest release, i dealt with 2 problems in order to make 
both log macro-groups work:

### `TAG` undefined

With just `USE_ESP_IDF_LOG` *build_flag* the compilation started breaking
due to missing declarations of `TAG`
(i figured due to TAG-defining code commented-out by #6351), eg:

```
~/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-log.h:163:67: error: 'TAG' was not declared in this scope
#define log_e(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR, TAG, format, ##__VA_ARGS__);}while(0)
                                                                ^~~
~/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/log/include/esp_log.h:421:81: note: in definition of macro 'ESP_LOG_LEVEL'
        if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                ^~~
~/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-log.h:163:32: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
#define log_e(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR, TAG, format, ##__VA_ARGS__);}while(0)
                                ^~~~~~~~~~~~~~~~~~~
~/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiAP.cpp:168:10: note: in expansion of macro 'log_e'
        log_e("set AP config failed");
```

I had to add this definition into my *build_flags* `-DTAG='"arduino-tag"'`,
and then i got just ERROR-level logs, as expected (the default log-level).

HINT: I suggest more polite approach, to leave the default tag present before #6351 got merged,
and introduce something like a `DEFAULT_ARDUINO_LOG_TAG` to override it.

### `CORE_DEBUG_LEVEL` does not enable arduino logs

Then i wanted to change the log-level.
Unfortunately, [IDF's logging instructions](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/log.html) :
are not quite fit for *platformio* because simply defining *build_flags* for
`CONFIG_LOG_DEFAULT_LEVEL` & `CONFIG_LOG_MAXIMUM_LEVEL` don't work due to `sdkconfig.h` 
overriding them.

I thought to experiment with `CORE_DEBUG_LEVEL`:
```
build_flags=... -DCORE_DEBUG_LEVEL=ESP_LOG_DEBUG
```
which it affected IDF-LOG statements only, and got *no arduino logs at all!*
(see sample code & sample-outputs, below)

I didn't manage to find why it happened, but i worked around it by using
*plain integers*, e.g.:
```
build_flags=... -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
```

I have no hint there, but i believe that `CORE_DEBUG_LEVEL` should continue working
with `ARDUHAL_LOG_LEVEL_XXX` along with `ESP_LOG_XXX` constants
(reason: the later works on runtime, so it's natural the developer to try these constants),
anyhow, please provide guidance in the docs how to configure log-level for both macro-calls
on *compile time*.

Note that *run-time* levels worked like charm for both macros, afterwards.

### Sketch

```cpp
void do_logs()
{
    esp_log_level_set("*", ESP_LOG_DEBUG);
    ESP_LOGD("MyTAG", "ESP_LOGD");
    ESP_LOGI("MyTAG", "ESP_LOGI");
    ESP_LOGW("MyTAG", "ESP_LOGW");
    ESP_LOGE("MyTAG", "ESP_LOGE");

    log_d("log_d");
    log_i("log_i");
    log_w("log_w");
    log_e("log_e");
}

OUTPUT without arduino logs:
D (239) MyTAG: ESP_LOGD
I (241) MyTAG: ESP_LOGI
W (244) MyTAG: ESP_LOGW
E (248) MyTAG: ESP_LOGE

OUTPUT with arduino logs:

D (239) MyTAG: ESP_LOGD
I (241) MyTAG: ESP_LOGI
W (244) MyTAG: ESP_LOGW
E (248) MyTAG: ESP_LOGE
D (251) arduinag: log_d
I (253) arduinag: log_i
W (256) arduinag: log_w
E (259) arduinag: log_e
```

### Debug Message

```plain
Nothing
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ankostis, 2022-04-04:
To clarify the 2nd issue i wrote above with latest release, what really buged me was that Arduino-logs were * silently ignored* when i had set `-DCORE_DEBUG_LEVEL=ESP_LOG_DEBUG`, so any warning regarding the value type would be welcomed.

VojtechBartoska, 2022-04-07:
Hello @ankostis, I'm not really sure if we can help on this. Platformio officialy doesn't support version 2.0.0 and newest so not sure if we are able to make this compatible.

@me-no-dev Can you please take a look on this?

ankostis, 2022-04-07:
What about the undefined `TAG` (#6351), why was it done?
There is still code out there expecting this to be defined.  Is this covered through `sdkconfig.defaults`?

ankostis, 2022-04-08:
> Platformio officialy doesn't support version 2.0.0 and newest so not sure if we are able to make this compatible.

My apologies for asking here for support, but where can i find such information?

VojtechBartoska, 2022-04-11:
> > Platformio officialy doesn't support version 2.0.0 and newest so not sure if we are able to make this compatible.
> 
> My apologies for asking here for support, but where can i find such information?

Take a look on this issue which is pinned at the top of Arduino core Issue tracker https://github.com/espressif/arduino-esp32/issues/6044. You can find there related issue opened in PIO.

VojtechBartoska, 2022-04-11:
Not officialy, there are just workarounds