espressif/arduino-esp32 #6535: tusb.h not found

TheButlah, 2022-04-04:
### Board

esp32-c3-mini-1

### Device Description

I am using my own custom PCB design, the schematic of which is [here](https://github.com/TheButlah/slimevr_pcb/blob/main/hardware/ferrous_slime/schematic/ferrous_slime.pdf). It is using the ESP32-C3-MINI-1 module from espressif.

### Hardware Configuration

D+ and D- pins on esp32c3 mini are connected to a USB connector. There is no external USB to UART chip. The other UART pins on the esp32c3 are not connected to anything but IO headers.

### Version

other

### IDE Name

PlatformIO on VSCode

### Operating System

Windows 10

### Flash frequency

Not sure, I believe its DIO

### PSRAM enabled

no

### Upload speed

Default PlatformIO

### Description

I marked "no" for PSRAM, but I have no idea what it means 😕 

Unfortunately I'm not familiar enough with how PlatformIO works to say *which* version of Arduino I'm using. All I know is that I am using this fork or espressif's platformio support, which adds support for the esp32c3: https://github.com/tasmota/platform-espressif32/tree/948c46dbbdb5c24c9ea543358e116bc0b0cbab34

I am trying to use the USB serial feature on the ESP32c3, where the serial is done over the D+ and D- pins on the board instead of the UART pins. I believe this is referred to as the USB CDC feature.

I found no documentation for this, so I searched through the headers, looking for a way to replace the UART based `Serial` global with a USBCDC `Serial` global. I found what I believe are the relevant defines, and set them. In the end, I got an error saying that `tusb.h` was missing.

### Sketch

```cpp
++
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("hello world");
  delay(100);
}

Here is the platformio.ini file (note the defines):

```
[env:arduino-esp32c3]
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3.zip
board = esp32-c3-devkitm-1
framework = arduino
upload_port = "COM6"
build_flags =
	-DARDUINO_USB_CDC_ON_BOOT=1
	-DCONFIG_TINYUSB_ENABLED=1
	-DCONFIG_TINYUSB_CDC_ENABLED=1
```
```

### Debug Message

```plain
Compiling .pio\build\arduino-esp32c3\FrameworkArduino\esp32-hal-bt.c.o
In file included from D:/Ryan/.platformio/packages/framework-arduinoespressif32/cores/esp32/USB.cpp:20:
D:/Ryan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-tinyusb.h:24:10: fatal error: tusb.h: No such file or directory

**************************************************************
* Looking for tusb.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:tusb.h"
* Web  > https://registry.platformio.org/search?q=header:tusb.h
*
**************************************************************

 #include "tusb.h"
          ^~~~~~~~
compilation terminated.
In file included from D:/Ryan/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:18:
D:/Ryan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-tinyusb.h:24:10: fatal error: tusb.h: No such file or directory

**************************************************************
* Looking for tusb.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:tusb.h"
* Web  > https://registry.platformio.org/search?q=header:tusb.h
*
**************************************************************

 #include "tusb.h"
          ^~~~~~~~
compilation terminated.
Compiling .pio\build\arduino-esp32c3\FrameworkArduino\esp32-hal-cpu.c.o
Compiling .pio\build\arduino-esp32c3\FrameworkArduino\esp32-hal-dac.c.o
Compiling .pio\build\arduino-esp32c3\FrameworkArduino\esp32-hal-gpio.c.o
Compiling .pio\build\arduino-esp32c3\FrameworkArduino\esp32-hal-i2c-slave.c.o
Compiling .pio\build\arduino-esp32c3\FrameworkArduino\esp32-hal-i2c.c.o
Compiling .pio\build\arduino-esp32c3\FrameworkArduino\esp32-hal-ledc.c.o
*** [.pio\build\arduino-esp32c3\FrameworkArduino\USB.cpp.o] Error 1
*** [.pio\build\arduino-esp32c3\FrameworkArduino\USBCDC.cpp.o] Error 1
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-04-05:
C3 has an integrated USB CDC+JTAG. It's realized purely in hardware and is not a full-featured USB peripheral, like the one in S2 or S3. So no TinyUSB on C3 :)

closed 2022-04-05 by me-no-dev

TheButlah, 2022-04-05:
In that case, how do I use its integrated USB CDC? I cannot find any documentation on it :(

kostuch, 2022-04-05:
@TheButlah 
Normally you should see it as a COM port in DeviceManager (when using windows)
It require libUSB-WinUSB driver manipulation to use it. You can use this tool https://visualgdb.com/UsbDriverTool/
Then put simple statement in your code: printf("something via usb\n");
...and you should see it in terminal window.

TheButlah, 2022-04-05:
But how would I replace the `Serial` instance?

kostuch, 2022-04-05:
> But how would I replace the `Serial` instance?

Just don't use Serial.
Serial.print(); will be printf();

kostuch, 2022-04-05:
> I need to use `Serial`, many third party libraries call that instance. I cannot maintain separate forks for every third party library 😅

You may find some inspiration here: https://www.esp32.com/viewtopic.php?f=19&t=24280

TheButlah, 2022-04-05:
@me-no-dev Could you reopen? If it is intended for there to be no tinyusb.h, I think a better title for my issue would be "Support `Serial` instance on the ESP32-C3

Its important for software compatibility for this to work out of the box, and going through the USBCDC and HardwareSerial files, it certainly looks like the intention is there for a define to have the onboard USBCDC used by `Serial`

Jason2866, 2022-04-05:
Add this and it should work
```
build_flags =
                -DARDUINO_USB_CDC_ON_BOOT=1
                -DARDUINO_USB_MODE=1
```

TheButlah, 2022-04-05:
@Jason2866 Trying your suggestion out, but it still doesn't work. I call `Serial.println("Hello")` in `loop()` with a delay, it prints nothing but this initial debug info:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4200c97c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: a9753a4fc647c6545c1b919ef08db429130a48592727edca270f1e5a3da0d0a9
Expected: 3bf6ef2cf3b9eefcd4b3c70cc5d1ce5138292d101a5cb1d5db6fbebf081b0a19
Attempting to boot anyway...
entry 0x403ce000
```

TheButlah, 2022-04-05:
In the above code, I forgot to call `Serial.begin()`. 🤦‍♂️ 

Sorry for the dumb mistake! I can confirm that @Jason2866's build works!

pierreSCpapon, 2022-04-11:
Just wanted to say I got it working using the @Jason2866 build, thank you.

`Serial0` is the hardware serial hooked to the serial to USB chip
`Serial` is the HWCDC

I was able to monitor the HWCDC using serial terminal as well as upload.

`platform.ini` looks like:

```
[env:esp32dev]
platform = https://github.com/Jason2866/platform-espressif32.git
platform_packages = 
	toolchain-riscv-esp
	framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.3-RC1
	platformio/tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v3.3/esptool-3.3.zip
framework = arduino
board = esp32-c3-mini
board_build.mcu = esp32c3
board_build.partitions = huge_app.csv
board_build.variant = esp32c3
board_build.f_cpu = 160000000L
board_build.f_flash = 80000000L
board_build.flash_mode = dio
board_build.arduino.ldscript = esp32c3_out.ld
build_unflags = 
	-DARDUINO_ESP32_DEV
	-DARDUINO_VARIANT="esp32"
build_flags = 
	-DARDUINO_ESP32C3_DEV
	-DARDUINO_VARIANT="esp32c3"
	-DARDUINO_USB_CDC_ON_BOOT=1
	-DARDUINO_USB_MODE=1
build_type = debug
monitor_speed = 115200
```

Jason2866, 2022-04-11:
@pierreSCpapon  Your setup is redundant. Not more needed than this.
```
[env:esp32-c3]
platform = https://github.com/Jason2866/platform-espressif32.git
framework = arduino
board = esp32-c3-mini
board_build.partitions = huge_app.csv
build_type = debug
monitor_speed = 115200
```