espressif/arduino-esp32 #6539: esp32_gpioMux

xinkiknix, 2022-04-05:
### Board

ESP32 Dev Mod

### Device Description

BMP180 pressure sensor

### Hardware Configuration

int Pin_I2C_SDA = 32;
int Pin_I2C_SCL = 33;
uint32_t FREQ = 0;

### Version

v2.0.2

### IDE Name

Arduino IDE 2.0.0-rc5

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

After upgrading the espressif boards manager to version 2.0.3-RC1 I get compiler errors. Errors that were /are not pressent in version 2.0.2

### Sketch

```cpp
not really relevant, error occurs during compilation  of include file:
Message :->
In file included from C:\Users\Johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\cores\esp32/Arduino.h:36,
                 from c:\Users\Johan\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:30,
                 from c:\Users\Johan\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:17:
c:\Users\Johan\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c: In member function 'void TFT_eSPI::gpioMode(uint8_t, uint8_t)':
c:\Users\Johan\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:163:32: error: 'esp32_gpioMux' was not declared in this scope
   ESP_REG(DR_REG_IO_MUX_BASE + esp32_gpioMux[gpio].reg) // Register lookup
                                ^~~~~~~~~~~~~
C:\Users\Johan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\cores\esp32/esp32-hal.h:73:47: note: in definition of macro 'ESP_REG'
 #define ESP_REG(addr) *((volatile uint32_t *)(addr))
                                               ^~~~
```

### Debug Message

```plain
not available
```

### Other Steps to Reproduce

Works ok in previous version 2.0.2

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-04-05:
The lib `TFT_eSPI` is not compatible to the changes done in core.
To solve ask the maintainer of the lib to update to be compatible with latest core version

mc-hamster, 2022-04-05:
We're also getting this error when including paulstoffregen/OneWire@^2.3.6 with 2.0.3-RC1

mc-hamster, 2022-04-05:
This PR ( https://github.com/meshtastic/Meshtastic-device/pull/1286 ) is being blocked by this defect.

Jason2866, 2022-04-06:
Yes, since onewire is using not supported anymore internal gpio pin manipulating too.
I knew because i have adopted our [Onewire fork](https://github.com/arendst/Tasmota/pull/14861) to work with latest core

xinkiknix, 2022-04-06:
removing #include <TFT_eSPI.h>;  solved the compile error. (it was not needed for this but still included from a previous test)
OneWire is still used and gives no issue.

VojtechBartoska, 2022-04-07:
@xinkiknix can we close this then?

xinkiknix, 2022-04-07:
Yes, this can be closed.

closed 2022-04-07 by xinkiknix