espressif/arduino-esp32 #6541: ESP32 is crashing when WiFi is connecting / BLE is initializing

umerm64, 2022-04-05:
### Board

ESP32 DEVKIT C

### Device Description

I am using simple ESP32 module and there is no external device attached to it.

### Hardware Configuration

I don't have any external device connected to it.
I am using the WLED project from here: https://github.com/Aircoookie/WLED
Only the GPIO pins used by this project are used.

### Version

v2.0.2

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I am using the WLED project here: https://github.com/Aircoookie/WLED
I am trying to add BLE support to this project.
For that, I am using partitions based on the `huge_app.csv` file
Currently, I am facing an exception at run time when using BLE as well as WiFi in this.
Basically when I am initializing BLE first then the exception occurs when connecting to the WiFi network.
Similarly when I am connecting to WiFi network then an exception occurs at the BLE initializing.
I am attaching the trace for both in this ticket.

I have also uploaded the code from the `usermod.cpp` file where I am initializing BLE server, its basically the code taken from BLE example given in the ESP32.

### Sketch

```cpp
#include "wled.h"
/*
 * This v1 usermod file allows you to add own functionality to WLED more easily
 * See: https://github.com/Aircoookie/WLED/wiki/Add-own-functionality
 * EEPROM bytes 2750+ are reserved for your custom use case. (if you extend #define EEPSIZE in const.h)
 * If you just need 8 bytes, use 2551-2559 (you do not need to increase EEPSIZE)
 *
 * Consider the v2 usermod API if you need a more advanced feature set!
 */

//Use userVar0 and userVar1 (API calls &U0=,&U1=, uint16_t)

#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

[... 165 lines cut ...]
  #8  0x400f55a2:0x3ffd32b0 in ble_init() at wled00/usermod.cpp:110
  #9  0x400f56a4:0x3ffd32f0 in userConnected() at wled00/usermod.cpp:123
  #10 0x400f6b03:0x3ffd3310 in WLED::handleConnection() at wled00/bus_manager.h:149
  #11 0x400f6b4a:0x3ffd33d0 in WLED::loop() at wled00/bus_manager.h:149
  #12 0x400f6f62:0x3ffd3410 in loop() at D:/MyStuff/WattsLights/WLED/wled00/wled00.ino:20
  #13 0x40116e2c:0x3ffd3430 in loopTask(void*) at C:\Users\umer.arshad\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:23
  #14 0x4009217a:0x3ffd3450 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1084
load:0x40078000,len:11220
load:0x40080400,len:5360
entry 0x4008067c
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-04-06:
Not directly solving your issue, for Tasmota we use [h2zero/NimBLE](https://github.com/h2zero/NimBLE-Arduino) since the beginning of using BLE because it has a lower footprint for RAM and flash usage. It does work fine with latest core.
Maybe you give it a try.

umerm64, 2022-04-10:
@Jason2866 I tried this library and the results are a bit better.
So it works correctly with the AccessPoint configuration of ESP32 but the issue occurs when the following configuration is used:
BLE init -> WiFi network connect -> AsyncWebServer creation.
Basically when WiFi is connected it remains ok but as soon as the webServer is initialized then the following exception occurs:
```
abort() was called at PC 0x40199fff on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x400902f8:0x3ffdba20 0x40090655:0x3ffdba40 0x40199fff:0x3ffdba60 0x40179764:0x3ffdba80 0x40178aa2:0x3ffdbaa0 0x4019cb8e:0x3ffdbac0 0x4009209e:0x3ffdbaf0
#0 0x400902f8:0x3ffdba20 in invoke_abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:648
#1 0x40090655:0x3ffdba40 in abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:648
#2 0x40199fff:0x3ffdba60 in pm_set_sleep_type at ??:?
#3 0x40179764:0x3ffdba80 in wifi_set_ps_process at ??:?
#4 0x40178aa2:0x3ffdbaa0 in ieee80211_ioctl_process at ??:?
#5 0x4019cb8e:0x3ffdbac0 in ppTask at ??:?
#6 0x4009209e:0x3ffdbaf0 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1084
load:0x40078000,len:11220
load:0x40080400,len:5360
entry 0x4008067c
```

Jason2866, 2022-04-10:
You have to search the bug. This combination does work without issues for Tasmota.
If you are using AsyncWebserver or LittleFS you are hitted by this bug #6502 #6507

VojtechBartoska, 2022-05-05:
Can you please retest this with 2.0.3 Version? Thanks!

VojtechBartoska, 2022-05-26:
Closing as expired, if needed please reopen the issue.

closed 2022-05-26 by VojtechBartoska

jacobk, 2024-02-01:
@umerm64 Did you ever resolve this issue? (running into the same problem)

Jason2866, 2024-02-01:
AsyncWebserver is outdated and not maintained. When used a lot of weird issues can happen.

jacobk, 2024-02-02:
> AsyncWebserver is outdated and not maintained. When used a lot of weird issues can happen.

I see @Jason2866 . I was encountering this using the latest version of WLED. Are you saying that the error is caused by AsyncWebserver (I guess WLED is still using an old version https://github.com/Aircoookie/ESPAsyncWebServer)

umerm64, 2024-02-02:
@jacobk I used h2zero/NimBLE for adding support of BLE in my code since with BLE lib provided by sdk was crashing.
After that, the issue did not occur.
But I didn't exactly find the solution.
But I would advise you to not use BLE and WiFi simultaneously.