espressif/arduino-esp32 #6542: PWM frequency issue on the ESP32 WROOM Module

mistrykm4u, 2022-04-06:
### Board

ESP32 Dev Module (MH-ET LIVE)

### Device Description

I have attached my own designed Light Dimmer module. It works on a single PWM Pin
[image]
[image]
.

### Hardware Configuration

Any GPIO (PWM supported as per datasheet) 

### Version

v2.0.2

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 7

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Hello Team,

I am using the ESP8266 12E module for my own designed light dimmer. My light dimmer is working on the ESP8266 12E module very well and really awesome.
My light dimmer works on a single GPIO (PWM) pin. its normal analogWrite() PWM function does not need a set  frequency. But I want to transfer to the ESP32 WROOM module because I need more GPIO pins. It is not possible on ESP8266 12E module. Recently I bought the ESP32 WROOM (MH-ET LIVE) module and read about this module's whole documents. ESP32 does not work with the analogWrite() function its work ledcWrite() function. So I am testing a Blynk app with a slider. 
It needs simple code. i am using 1 Khz frequency, Channel 0 and resolution 8 bit. But my dimmer is fluctuating if im on/off or change pwm values by Blynk app slider. I don't understand it. If I test the same on my ESP8266 12E module it will run very smoothly and flicker free. I am changing different channels but this ESP32 WROOM module does not give me stable PWM frequency. I don't know why.

### Sketch

```cpp
This example runs directly on ESP32 chip.

  Note: This requires ESP32 support package:
    https://github.com/espressif/arduino-esp32

  Please be sure to select the right ESP32 module
  in the Tools -> Board menu!

  Change WiFi ssid, pass, and Blynk auth token to run :)
  Feel free to apply it to any other example. It's simple!
 *************************************************************/

/* Comment this out to disable prints and save space */
#define BLYNK_PRINT Serial

/* Fill-in your Template ID (only if using Blynk.Cloud) */
//#define BLYNK_TEMPLATE_ID   "YourTemplateID"

#include <WiFi.h>
#include <WiFiClient.h>
#include <BlynkSimpleEsp32.h>

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "kAUyqL8SKQXW9QXAxpuXXXXXXXXXXXX";

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "Sector VII";
char pass[] = "VviXXXXXXXXXXXX";
int ledPin = 16;
int pinValue;
// setting PWM properties
const int freq = 1000;
const int ledChannel = 0;
const int resolution = 10;

BLYNK_WRITE(V0)
{
  pinValue = param.asInt(); // assigning incoming value from pin V1 to a variable
  ledcWrite(ledChannel, pinValue);   // process received value
}

void setup()
{
  ledcSetup(ledChannel, freq ,resolution);
  
  // attach the channel to the GPIO to be controlled
  ledcAttachPin(ledPin, ledChannel);
  
  // Debug console
  Serial.begin(115200);

  Blynk.begin(auth, ssid, pass);
}

void loop()
{
  Blynk.run();
}
```

### Debug Message

```plain
Not need any message.
```

### Other Steps to Reproduce

My light dimmer is working on the ESP8266 12E module very well and really awesome.
My light dimmer works on a single GPIO (PWM) pin. its normal analogWrite() PWM function does not need a set  frequency.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-04-06:
Need more info.  I would start by putting a scope on that PWM output pin.  You should see the duty cycle change as you adjust the brightness.  Compare the waveforms between the ESP8266 and ESP32.  BTW, nice clean board layout.

mistrykm4u, 2022-04-07:
Hi David @mrengineer7777
Thank you so much. Yes this is my most beautiful invention and a very compact size designed light dimmer. It's a 555 timer CMOS based dimmer and just working on a single PWM pin. we do not need to set a specific frequency on the PWM signal. This dimmer works on reverse function so we give high value on the PWM pin then its going fully OFF and low value then its full ON. If we use 8 bit resolution then we need to give 235 approx value for fully OFF and give 0 then it's going full ON. Same like using 10 bit then we need to give 925 approx value for fully OFF. This is working on any PWM based microcontroller without any issues.But I bought this ESP32 WROOM module and tested on it then I am facing a flickering problem. I have attached my ESP8266 based smart switch board video. You can see how it works smoothly

https://user-images.githubusercontent.com/26956930/162120322-3553ae05-7c49-4d4a-88e1-232020c6d5b7.mp4

.

mistrykm4u, 2022-04-07:
@mrengineer7777 

https://user-images.githubusercontent.com/26956930/162142990-0757784b-5b69-466d-a139-484794b34846.mp4

Unfortunately I have no osciloscop so I am not able to check both (ESP8266 & ESP32 WROOM) board PWM frequency.
But I am sure there's something going wrong in this ESP32 WROOM module. I have attached a video. 
See how it is flickering. If light brightness is full then we do not see this flickering but if i am down the brightness like give value 210 -230 then we see properly how it is flickering. If I have an oscilloscope then I definitely compare this PWM frequency with ESP8266. then we all understand where is going wrong with this PWM on ESP32 WROOM.

mistrykm4u, 2022-04-08:
Any one help me regarding this issue?

@mrengineer7777 @kumekay @sctv @earlephilhower

mistrykm4u, 2022-04-13:
I have bought one new ESP32 WROOM module. Today it came and I am testing it. but same PWM frequency issue on it. Please give me a solution to this problem. My whole project is pending and also it's going to be delayed.

TD-er, 2022-04-13:
Looks like you're using either pin 16 or 17.
Please try another pin, as those may also be used on PSRAM and I've seen the ESP32 still trying to send rather high frequency signals to those pins.
Just an idea for your project to not getting delayed, you can also add one of the many available I2C PWM chips.
Then it's only taking 2 pins for the I2C bus, which you're probably already using and those chips often have multiple PWM pins which can be controlled separately.

mistrykm4u, 2022-04-14:
Hi @TD-er 
Thank you so much for your reply and for giving me a good solution. I also checked all suitable GPIO Pins like GPIO - 2, 4, 13, 16, 17, 18, 19, 21, 22, 23, 25, 26, 27, 32, 33 (5, 14, 15 outputs PWM signal at bootup). But all pins behave the same. You say I am using an I2C based PWM module. This is a good idea but in my project I am not able to use it because I am also using one I2C based capacitive touch sensor module. If I add more modules then this task is more critical. So finally it's not suitable for me. I don't understand why it's not running like ESP8266. I do not need a specific frequency, I just want a basic 1kHz frequency like given by ESP8266. Why not make a basic analogWrite() function in this API.

mistrykm4u, 2022-04-14:
I think in this portal no one is interested in resolving this problem. Just give a solution like add an extra PWM module and that's your problem is resolved. Just now I am shown @ VojtechBartoska labeled (Resolution: Awaiting response) What is this? No one properly talks on this PWM issue.

TD-er, 2022-04-14:
> I think in this portal no one is interested in resolving this problem. Just give a solution like add an extra PWM module and that's your problem is resolved

Maybe you misinterpreted my intention here.
Given this is an issue with the SDK. Even if they fix it immediately (which is not very likely as things take time to find the cause, test etc), then it takes time for the fix to get merged, update the SDK etc.
So just as an idea for you to keep in mind (think of it as thinking outside the box) for your project, you may consider using a special PWM chip.
Just to explain why I think that may be a fix in the long term; Even if this issue with the ESP32 is fixed, then you might run into similar issues later when you try to have > N PWM signals at the same time. Not sure if N is 3 or 8, but it may become an issue when you run out of channels for the hardware PWM thingies in the ESP32.
Also some libraries use those timers, so it might be those will at some time interfere with your project.

I was just trying to be practical and by **no means** it was intended to be like "meh, this is never going to be fixed, throw in some more hardware"

VojtechBartoska, 2022-04-14:
@mistrykm4u I labelled this issue with "awaiting response" after @TD-er which sounds to me reasonable so that label meant that it's need to be answer by you.

VojtechBartoska, 2022-04-14:
We can investigate this further if still needed but you need to understand that we have our work plans as well.

And please be patient, this issue tracker is meant to be a friendly place, we are not responsible for your deadlines.

VojtechBartoska, 2022-04-14:
Btw, Do you have Espressif HW or only MH-ET (thrid party hw) based on ESP32?

mistrykm4u, 2022-04-15:
> Maybe you misinterpreted my intention here. Given this is an issue with the SDK. Even if they fix it immediately (which is not very likely as things take time to find the cause, test etc), then it [...]

@TD-er 
No dear, this is not my intention or I am not misinterpreting your intention. Sorry to say this. I was really irritated that time. You have given me a very good idea. You say and explain everything is true and very good.

mistrykm4u, 2022-04-15:
> > You say I am using an I2C based PWM module. This is a good idea but in my project I am not able to use it because I am also using one I2C based capacitive touch sensor module. If I add more modules then this task is more critical.
> 
> Not sure if that will be an issue, unless your touch sensor is acting rather 'strange'. What I mean is that usually those chips have some kind of register, which you must mark as "I have read it" for it to become active again or being able to notice the change in value. Some have also an interrupt pin to signal a new value which you can read.
> 
> The I2C PWM chip only needs I2C traffic when you try to set a new value. It is not like you're communicating with it for a long time, nor there is much traffic (not like I2C displays or I2C serial ports with rather "chatty" devices like a GPS)

@TD-er 
Yes this is true. I know about it.

mistrykm4u, 2022-04-15:
> We can investigate this further if still needed but you need to understand that we have our work plans as well.
> 
> And please be patient, this issue tracker is meant to be a friendly place, we are not responsible for your deadlines.

@VojtechBartoska 
I am really sorry to say this. I apologize for it. Yes, it is true because the investigation about this issue and testing on it all is not a simple task. it takes time.

mistrykm4u, 2022-04-15:
> Btw, Do you have Espressif HW or only MH-ET Live (thrid party hw) based on ESP32?

@VojtechBartoska 
No, I do not have an Espressif HW. I have only MH-ET Live and Node MCU ESP-32S (third party hw)

VojtechBartoska, 2022-04-20:
> > Btw, Do you have Espressif HW or only MH-ET Live (thrid party hw) based on ESP32?
> 
> @VojtechBartoska No, I do not have an Espressif HW. I have only MH-ET Live and Node MCU ESP-32S (third party hw)

this complicate a testing for us a bit as we can only test this on Espressif HW and it can relate to specific HW issue.

VojtechBartoska, 2022-08-23:
Hello @mistrykm4u, is this still valid?

closed 2022-10-10 by VojtechBartoska