espressif/arduino-esp32 #6545: ESP32c3 - No serial output from built in jtag hardware

Sys64736, 2022-04-06:
### Board

ESP32c3 

### Device Description

Nothing attached

### Hardware Configuration

GPIO 18 & 19 are attached USB D- and D+ along with GRD and VCC

### Version

latest master

### IDE Name

PlatformIO

### Operating System

Win 10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

When connected to a C3 chip without a uart chip, directly to USB+ and USB-, I cannot get any Serial output.

Espressif docs say it is possible.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/usb-serial-jtag-console.html

Chip runs code and works as usual though.

These values are set in [framework-arduinoespressif32\tools\sdk\esp32c3\include\config\sdkconfig.h] and I believe it should give output to the console using HardwareSerial

#define CONFIG_ESP_CONSOLE_UART_DEFAULT 1
#define CONFIG_ESP_CONSOLE_SECONDARY_USB_SERIAL_JTAG 1
#define CONFIG_ESP_CONSOLE_UART 1
#define CONFIG_ESP_CONSOLE_UART_NUM 0
#define CONFIG_ESP_CONSOLE_UART_BAUDRATE 115200

Any help or advice would be appreciated as I'm not 100% sure this is an issue or something I'm missing.

### Sketch

```cpp
Serial.begin(115200);

Loop()
Serial.println("test");
```

### Debug Message

```plain
No output in console even from esp_log etc

Does initial boot log and that's it.
```

### Other Steps to Reproduce

Connect using USB to on chip debugger, upload code.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-04-06:
Problems with Platform.io need to be directed to https://github.com/platformio/platform-espressif32.  In Arduino IDE, select "Enable CDC on boot".
Note that you cannot change settings in sdkconfig.h, those are the settings that have been compiled into the libraries included in arduino-esp32.

Sys64736, 2022-04-06:
Ok thanks, I didn't realize this was a pio issue.  I'll try in the Arduino IDE with that CDC option.

closed 2022-04-06 by Sys64736

Sys64736, 2022-04-08:
In case someone else is searching this issue.  Without setting ARDUINO_USB_CDC_ON_BOOT,(issues with pio atm), I used USBSerial instead of Serial to get debug info to show up.   You have to disconnect the terminal once and re-reconnect then USBSerial.print will start working.  (i used CoolTerm), but the built-in one works too.  This was connecting directly to pin 18 & 19 with a USB cable.