espressif/arduino-esp32 #6546: How to Encode a HMAC SHA256 has with base64 on a ESP32

RensMvP, 2022-04-07:
### Board

WT32-ETH01

### Device Description

Nothing added

### Hardware Configuration

Nothing added

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

I want to encode a HMAC SHA256 hash with Base64. But i get all sorts of mishaps with the datatypes between both functions of the mbtls library.

See code below

### Sketch

```cpp
// hasing and encryption library's
#include "mbedtls/md.h"     // HMAC hashing used for the pairing process
#include "mbedtls/base64.h" // Base64 encoding/decoding for the paring proces

// encryption and hashing
mbedtls_md_context_t ctx;
mbedtls_md_type_t md_type = MBEDTLS_MD_SHA256;

char *pin = "6577";
char *timestamp = "1156";

void init_Hash(char *key)
{
  // calculate size of the key
  const size_t keyLength = strlen(key);

  // init hash function
  mbedtls_md_init(&ctx);
  mbedtls_md_setup(&ctx, mbedtls_md_info_from_type(md_type), 1);        // setup with the prespecified hash (md_type)
  mbedtls_md_hmac_starts(&ctx, (const unsigned char *) key, keyLength); // start the hash with the Secret key
}

void Update_Hash(char *data)
{
  // calculate size of the key
  const size_t dataLength = strlen(data);

  // update hash
  mbedtls_md_hmac_update(&ctx, (const unsigned char *) data, dataLength);
}

void Finalize_Hash()
{
  /*
     base64:
     https://pcbartists.com/firmware/base64-encoding-and-decoding-on-esp32/#:~:text=The%20ESP%2DIDF%20contains%20functions,mbedTLS%20base64%20encoding%20and%20decoding.
     https://siliconlabs.github.io/Gecko_SDK_Doc/mbedtls/html/base64_8h.html

     hashing:
     https://tls.mbed.org/api/md_8h.html#adf77ab9b0f117367883988ee5f0af1f4
     https://techtutorialsx.com/2018/01/25/esp32-arduino-applying-the-hmac-sha-256-mechanism/

     calculators
     https://www.freeformatter.com/hmac-generator.html#ad-output
     https://www.base64encode.org/

     HMAC from 11566577 result:
     81d4d97d850c29795ed5e8d6435b3347e982c2fdfc3b5d1b2fc976a3490cc1e4

     Encode with Base64 result:
     ODFkNGQ5N2Q4NTBjMjk3OTVlZDVlOGQ2NDM1YjMzNDdlOTgyYzJmZGZjM2I1ZDFiMmZjOTc2YTM0OTBjYzFlNA==

  */

  byte output_HMAC[32];		// byte datatype is needed to read the output_hmac correct 
  unsigned char output[64];
  size_t outlen;
  
  // return the final HMAC number
  mbedtls_md_hmac_finish(&ctx, output_HMAC);
  mbedtls_md_free(&ctx);
  
  mbedtls_base64_encode(output, 64, &outlen, (unsigned char *)output_HMAC, strlen(output_HMAC));
  
  
}

void setup()
{
  // intialize the serialport
  Serial.begin(115200);

  // initialize hash hmac
  init_Hash(SECRET_KEY);

}

void loop()
{
  Serial.println("hashing data");
  Update_Hash(timestamp);
  Update_Hash(pin);
  Finalize_Hash();
	
  //slow down
  delay(10000);
}
```

### Debug Message

```plain
invalid conversion from 'byte* {aka unsigned char*}' to 'const char*' [-fpermissive]
```

### Other Steps to Reproduce

No extra steps.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

igrr, 2022-04-08:
There are a few issues in the sketch:

* when taking pointers to strings and when referring to constant strings, you need to use `const char*`
* output_HMAC is an array of bytes, not a null-terminated string. So you can't use `strlen` with it.

with these two issues fixed and with another two lines to print `output`, the sketch works: https://wokwi.com/projects/328395940175020627.

_One more note, please try to not use this issue tracker to ask for programming advice. The issues with your sketch are related to general usage of data types and pointers in C language. There are several resources offering help to C developers, for instance a quick google search reveals a bunch of Discord channels. Please keep this issue tracker for bugs and feature requests related to ESP32 Arduino core._

closed 2022-04-08 by igrr

torntrousers, 2022-04-09:
@igrr I didn't know about wokwi.com, pretty cool, thanks for pointer!