espressif/arduino-esp32 #6547: PWM and hardware interruption

Samuel-Oliveira-BR, 2022-04-07:
### Board

ESP-WROOM-32D

### Device Description

ESP-WROOM-32D with two coils + RFID of 125khz circuit.

### Hardware Configuration

The RFID card reader circuit. Pin 5 generates a 125Khz PWM to excite the circuit and pin 19 reads the pulse width through an external interrupt.

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Win10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

We need generate the PWM and read the RFID card number using just on ESP (with one or two coil).
Test 1: Using only one coil for generate and reader: Card not read;
ESPressif engineer told us for use different coil (for generate and for read).
Test2: Using separated coil (one for generate the PWM and other for reader): Card not read;
Test3: Using diferent ESP. One for generate and other for read: Good. Card read.

Important: Before use the ESP, was used ATMEGA328 and just with one was genarated and read perfectly.

### Sketch

```cpp
#include "src/Rfid125_esp32.h" //Biliblioteca criada para a SGBras para leitura de cartões Rfid 125khz.

uint64_t cartao = 0; //Número do cartão lido pelo RFID. Somente cartões cadastrados.
Rfid125khz rfid(19,100); //Cria um objeto Rfid125khz. "rfid(Pino de interrupção,Timeout)";

void setup() {
  Serial.begin(115200);
}

void loop() {
  cartao = rfid.read();
  delay(1000);
  if(cartao) Serial.println(cartao);
}

//------------------------------------LIBRARY-------------------------------------:
/*
*/
#include <Arduino.h>
#include "Rfid125_esp32.h"

volatile uint16_t count   = 0;
volatile uint64_t timeout = 0;

uint8_t inputRfidPin;
volatile char cardbits[NBITS];

hw_timer_t * timer = NULL;

static void IRAM_ATTR signalchange()
{
  bool bitnow = 0;
  uint64_t delta = 0;
  uint64_t timenow;
  static uint64_t timeprev  = 0;
  static bool midbit = true;
  static bool synced = false;

  timenow = timerReadMicros(timer);
  bitnow = digitalRead(inputRfidPin);

  delta = timenow - timeprev;

  if (!synced && delta > BIT2TMIN && delta < BIT2TMAX) synced = true;
  
  if (synced && count < NBITS)
  {
    if (delta > BIT2TMIN && delta < BIT2TMAX)
    {
      cardbits[count] = bitnow;
      midbit = true;
      count++;
    }
    else
    {
      if (midbit) midbit = false;
      else
      {
        cardbits[count] = bitnow;
        midbit = true;
        count++;
      }
    }
  }
  timeprev = timenow;
}

Rfid125khz::Rfid125khz(int RfidPin, int timeOut)
{
  inputRfidPin = RfidPin;
  _timeOut = (uint64_t) timeOut * 1000;
  pinMode(inputRfidPin, INPUT);
}

uint32_t Rfid125khz::read()
{
  count   = 0;
  timeout = 0;

  Rfid125khz::on();

  while (1)
  {
    uint8_t data     = 0;
    uint8_t digit    = 0;
    uint8_t digitcnt = 0;
    uint8_t bitscnt  = 0;
    uint8_t datacnt  = 0;
    uint32_t cardid  = 0;
    bool nineseq = false;

    if (count >= NBITS)
    {
      for (int i = 0; i < NBITS; i++)
      {
        if (!nineseq)
        {
          if (cardbits[i] == 1) bitscnt++;
          else bitscnt = 0;
          
          if (bitscnt == 9) nineseq = true;
          continue;
        }
        data = (data << 1) | cardbits[i];
        datacnt++;
        if (datacnt > 4)
        {
          uint8_t parbit;
          uint8_t tmpdata;
          parbit = data & 1;
          data = data >> 1;
          tmpdata = data;
          tmpdata ^= tmpdata >> 2;
          tmpdata ^= tmpdata >> 1;

          if ( (tmpdata & 1) == parbit)
          {
            if (digitcnt >= 2) cardid = (cardid << 4) | data;
          }
          else break;

          data    = 0;
          datacnt = 0;
          digitcnt++;
        }

        if (digitcnt >= 10)
        {
          Rfid125khz::off();
          return cardid;
        }
      }
      count = 0;
    }
    if (timerReadMicros(timer) > _timeOut)
    {
      Rfid125khz::off();
      return 0;
    }
  }

}

void Rfid125khz::on()
{
  ledcAttachPin(5, 10);
  ledcSetup(10, 125000, 8);
  ledcWrite(10,126);
  timer = timerBegin(0, 80, true);
  attachInterrupt(digitalPinToInterrupt(inputRfidPin), signalchange, CHANGE);
}

void Rfid125khz::off()
{
  timerStop(timer);
  timer = NULL;
  detachInterrupt(inputRfidPin);
}
```

### Debug Message

```plain
cOMPILED SUCESSFULLOmpiled sucessfull
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-04-08:
ATMEGA328 does not run FreeRTOS and has a more "tight" timing on interrupts. I will suggest that you look into the RMT peripheral and use it's Input capabilities to precisely measure the low and high widths of the pulses. It will not require hw timer and will run all in the hardware.

Samuel-Oliveira-BR, 2022-04-08:
Why when I use two ESP (one for generate and otehr for reader) the ESP used for reader detect the card number?

atanisoft, 2022-04-08:
> I will suggest that you look into the RMT peripheral and use it's Input capabilities

Or MCPWM capture mode similar to https://github.com/espressif/esp-idf/tree/release/v4.4/examples/peripherals/mcpwm/mcpwm_capture_hc_sr04.

VojtechBartoska, 2022-04-22:
Do you still need help?

Samuel-Oliveira-BR, 2022-05-04:
Hi.
Sorry my delay.
Is not working yet.
We don't know why but if we use two ESP with the same program work correctly.
Let me explain the program:
The program generate the PWM for the 125khz circuit from a port.
Other port is used to reader the 125khz card.
If I use the same ESP for excite the circuit and reader the card, NO WORK.
But if I use the same program in two ESP, put one for excite the RFID circuit and the second ESP just for reader, the ESP work correctly.

We really don't know why.
We are using two different coil (one for generate and other for reader).

VojtechBartoska, 2022-05-04:
thanks for your feedback, we will take a look on this :)

Samuel-Oliveira-BR, 2022-05-04:
Thank you too.
I hope you can help us.
Our alternative for fix this issue is finishing.

Parsaabasi, 2025-01-08:
Hello @Samuel-Oliveira-BR 
Do you still need help with this?

Samuel-Oliveira-BR, 2025-01-08:
Hi @Parsaabasi.
I stopped this project. But I will try again.
I let you know some news asap.

Parsaabasi, 2025-01-09:
Dear @Samuel-Oliveira-BR 

This report contains a release that we no longer support. Please try the newer versions. 

I will close this for now, in case you faced the same problem with the newer versions, please feel free to reopen it. 

Thanks.