espressif/arduino-esp32 #6549: Guru Meditation Error when WiFi + analogRead

cod3gen, 2022-04-07:
### Board

ESP32-S3-WROOM-1

### Device Description

ESP32-S3-WROOM-1 on custom board. GPIO 5 Currently not connected to anything on board, also tested with GPIO 6 which is pulled low, same result on both
[image]
[image]
[image]
.

### Hardware Configuration

GPIO35/36/37/1(CS): SPI. GPIO48 also CS, but not connected atm.
GPIO15/16: I2C

### Version

latest master

### IDE Name

Arduino IDE + Platform IO

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1): error happens on some cycles when analogRead is being executed. Seems to only happen when wifi is connected(STA), does not seem to happen when in AP mode. 

Same result on both Arduino 1.8.19 with 2.0.3-RC1, and PIO with https://github.com/tasmota/platform-espressif32.git & framework-arduinoespressif32@https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/803/framework-arduinoespressif32-v4.4_dev-40d126bfd1.tar.gz

Tested on GPIO 5 and 6. Commenting out analogRead(6); and the code have been running for one whole day without issues.

### Sketch

```cpp
/****************************************************************************************************************************
  Async_ESP32_FSWebServer - Example WebServer with SPIFFS backend for ESP32
  For ESP32 boards

  ESPAsync_WiFiManager is a library for the ESP8266/Arduino platform, using (ESP)AsyncWebServer to enable easy
  configuration and reconfiguration of WiFi credentials using a Captive Portal.

  Modified from 
  1. Tzapu               (https://github.com/tzapu/WiFiManager)
  2. Ken Taylor          (https://github.com/kentaylor)
  3. Alan Steremberg     (https://github.com/alanswx/ESPAsyncWiFiManager)
  4. Khoi Hoang          (https://github.com/khoih-prog/ESP_WiFiManager)

  Built by Khoi Hoang https://github.com/khoih-prog/ESPAsync_WiFiManager
  Licensed under MIT license
 *****************************************************************************************************************************/
/*****************************************************************************************************************************
   Compare this efficient Async_ESP32_FSWebServer_DRD example with the so complicated twin ESP32_FSWebServer 
[... 1138 lines cut ...]
19:55:37.962 -> Backtrace:0x4003586e:0x3fce2eb0 |<-CORRUPTED
19:55:37.962 -> 
19:55:37.962 -> 
19:55:37.962 -> Core  0 register dump:
19:55:37.962 -> PC      : 0x400356db  PS      : 0x00060834  A0      : 0x80035725  A1      : 0x3fce95c0  
19:55:37.962 -> A2      : 0x02000000  A3      : 0x035d026b  A4      : 0x6000e004  A5      : 0x01000862  
19:55:38.008 -> A6      : 0x00000800  A7      : 0x00000000  A8      : 0x01000000  A9      : 0x6000e000  
19:55:38.008 -> A10     : 0x00000000  A11     : 0x6000e05c  A12     : 0x00000001  A13     : 0x00060823  
19:55:38.008 -> A14     : 0x3fce840c  A15     : 0x3fc99c24  SAR     : 0x0000001c  EXCCAUSE: 0x00000006  
19:55:38.008 -> EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  
19:55:38.008 -> 
19:55:38.008 -> 
19:55:38.008 -> Backtrace:0x400356d8:0x3fce95c0 |<-CORRUPTED
19:55:38.008 -> 
19:55:38.008 -> 
19:55:38.008 -> 
19:55:38.008 -> 
19:55:38.008 -> ELF file SHA256: 0000000000000000
19:55:38.008 -> 
19:55:38.008 -> Rebooting...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-04-08:
Please also provide actual minimal example that can reproduce the issue and decode those exceptions using the ExceptionDecoder

Jason2866, 2022-04-08:
Probably this issue is hitted from bug #6507 since it uses LittleFS.

cod3gen, 2022-04-08:
@me-no-dev Updated with minimal example, and updated serial output. Not able to decode the exeption, just says "Decoding stack results" Maybe due to Backtrace:0x4003586e:0x3fce2eb0 **|<-CORRUPTED**

cod3gen, 2022-04-08:
@Jason2866 Updated with minimal example, without usage of FileFS, same error.

cod3gen, 2022-04-08:
Ive created a similiar application in ESP-IDF, and it seems i have the same results here. Will be opening an issue over there.

Jason2866, 2022-04-08:
You can not use ADC2 with wifi.

cod3gen, 2022-04-08:
> You can not use ADC2 with wifi.

Auch, darn. New round with JLC soon then 😝

closed 2022-04-08 by cod3gen

igrr, 2022-04-08:
> You can not use ADC2 with wifi.

@Jason2866 I thought this is an ESP32-specific limitation. Are you sure it applies to S3?

According to the docs, on ESP32-S2, C3, S3 both ADC units can be used with Wi-Fi. There is a limitation that in certain cases Wi-Fi will have priority over user's ADC reading, and then `adc2_get_raw` function will return an error code. But it's still possible to retry reading the ADC value in that case.

Jason2866, 2022-04-08:
@igrr Thx, i tried (a while ago). The results doing this where not really useable and i had wdt now and than. So i thought the docs maybe are not correct.

igrr, 2022-04-08:
Okay, not sure about "usable" result (maybe indeed something unavoidable?) but the interrupt watchdog triggering is definitely a bug. Let's see what IDF folks say.

cod3gen, 2022-04-09:
GPIO 5/6 is not ADC2 btw @Jason2866. Will keep the thread over at IDF open

richcj10, 2022-10-24:
Do you have an update on this? I am getting wdt on CPU 1 with reading pin one on the C3 module. It has been bugging me for a while, lol