espressif/arduino-esp32 #6550: LEDC  - ERROR when setting FREQUENCY to less than 1 Hz ---> IntegerDivideByZero

HubertHuber, 2022-04-07:
### Board

DOIT ESP32 DEVKIT V1

### Device Description

only using LED_BUILTIN

### Hardware Configuration

no

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

should be able to set LEDC-frequency to values less than 1.00 hz
this was OK in versions 1.0.6 / 2.0.0 / 2.0.1 - but NOT in version 2.0.2 

### Sketch

```cpp
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);  
  ledcAttachPin(LED_BUILTIN, 0);
  ledcSetup(0, 0.5, 16); 
  ledcWrite(0, 1234);
}
void loop() {}
```

### Debug Message

```plain
E (32) ledc: freq_hz=0 duty_resolution=16
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x4000d099  PS      : 0x00060f30  A0      : 0x800d224e  A1      : 0x3ffb2760  
A2      : 0x0f424000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00060f20  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x0f424000  A12     : 0x00060f20  A13     : 0x3ffb2660  
A14     : 0x007bdb68  A15     : 0x003fffff  SAR     : 0x00000001  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40085f81  LEND    : 0x40085f91  LCOUNT  : 0xfffffffe  

Backtrace:0x4000d096:0x3ffb27600x400d224b:0x3ffb2780 0x400d10d9:0x3ffb27c0 0x400d0e35:0x3ffb2800 0x400d149a:0x3ffb2820
```

### Other Steps to Reproduce

works OK in previous versions

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

HubertHuber, 2022-04-07:
Arduino 1.8.19

me-no-dev, 2022-04-07:
@HubertHuber we need to check if IDF supports that. If so, we will implement it.

P-R-O-C-H-Y, 2022-04-12:
@HubertHuber IDF does not support that. I will make change to API so frequency won't be double but uint32_t.

HubertHuber, 2022-04-13:
That's not pretty, but i'll find another solution ;-)

closed 2022-04-13 by HubertHuber