espressif/arduino-esp32 #6551: esp32-C3 can not create access points

kramzarales, 2022-04-08:
### Board

esp32-c3

### Device Description

Tested on esp32-c3 dev module v2 and also bear esp32-c3 wroom02.

### Hardware Configuration

issue present regardless of pins used.

### Version

v2.0.2

### IDE Name

arduino IDE

### Operating System

windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

The issue is that the esp32-c3 will not create a visible access point. the code is from the Wifi library examples and it compiles fine and reports that it creates the access point. The board also will not connect to the internet if other examples are used.

My guess would be something to do with the WiFi library since everything else works fine.

### Sketch

```cpp
#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiAP.h>

#define LED_BUILTIN 10  

// Set these to your desired credentials.
const char *ssid = "yourAP";
const char *password = "yourPassword";

WiFiServer server(80);

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  Serial.begin(115200);
  Serial.println();
  Serial.println("Configuring access point...");

  // You can remove the password parameter if you want the AP to be open.
  WiFi.softAP(ssid, password);
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  server.begin();

  Serial.println("Server started");
}

void loop() {
  WiFiClient client = server.available();   // listen for incoming clients

  if (client) {                             // if you get a client,
    Serial.println("New Client.");           // print a message out the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out the serial monitor
        if (c == '\n') {                    // if the byte is a newline character

          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            // the content of the HTTP response follows the header:
            client.print("Click <a href=\"/H\">here</a> to turn ON the LED.<br>");
            client.print("Click <a href=\"/L\">here</a> to turn OFF the LED.<br>");

            // The HTTP response ends with another blank line:
            client.println();
            // break out of the while loop:
            break;
          } else {    // if you got a newline, then clear currentLine:
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }

        // Check to see if the client request was "GET /H" or "GET /L":
        if (currentLine.endsWith("GET /H")) {
          digitalWrite(LED_BUILTIN, HIGH);               // GET /H turns the LED on
        }
        if (currentLine.endsWith("GET /L")) {
          digitalWrite(LED_BUILTIN, LOW);                // GET /L turns the LED off
        }
      }
    }
    // close the connection:
    client.stop();
    Serial.println("Client Disconnected.");
  }
}
```

### Debug Message

```plain
no errors reported.
```

### Other Steps to Reproduce

I have tested the same code on 2 esp32 wroom 02e boards and an esp8266 and both of those worked fine. I tried with older releases of the board manager with the same problem.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

davenull, 2022-04-08:
I can confirm the same issues with this code using the esp32-s3-devkitC-1 module. Wifi does not connect to the AP, always returning a status 255. Using wifi code in esp-idf functions as expected.

VojtechBartoska, 2022-04-08:
@davenull on v2.0.3-RC1?

davenull, 2022-04-08:
Yea that is correct.

Jason2866, 2022-04-08:
Mhh,
```
New Client.
GET /H HTTP/1.1
Host: 192.168.4.1
Connection: keep-alive
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Linux; Android 7.0; SM-T815) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/99.0.4844.88 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
Referer: http://192.168.4.1/L
Accept-Encoding: gzip, deflate
Accept-Language: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7

Client Disconnected.
New Client.
GET /favicon.ico HTTP/1.1
Host: 192.168.4.1
Connection: keep-alive
User-Agent: Mozilla/5.0 (Linux; Android 7.0; SM-T815) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/99.0.4844.88 Safari/537.36
Accept: image/avif,image/webp,image/apng,image/svg+xml,image/*,*/*;q=0.8
Referer: http://192.168.4.1/H
Accept-Encoding: gzip, deflate
Accept-Language: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7

Client Disconnected.
New Client.
GET /L HTTP/1.1
Host: 192.168.4.1
Connection: keep-alive
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Linux; Android 7.0; SM-T815) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/99.0.4844.88 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
Referer: http://192.168.4.1/H
Accept-Encoding: gzip, deflate
Accept-Language: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7

Client Disconnected.
New Client.
GET /favicon.ico HTTP/1.1
Host: 192.168.4.1
Connection: keep-alive
User-Agent: Mozilla/5.0 (Linux; Android 7.0; SM-T815) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/99.0.4844.88 Safari/537.36
Accept: image/avif,image/webp,image/apng,image/svg+xml,image/*,*/*;q=0.8
Referer: http://192.168.4.1/L
Accept-Encoding: gzip, deflate
Accept-Language: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7

Client Disconnected.
```
Led can be switched on/off. What is not correct is the Name of the AP

Jason2866, 2022-04-11:
You can try WiFi.hostname(hostname);  before WiFi.mode
Iirc correctly this is needed since 2.0.x to get the hostname set.

pierreSCpapon, 2022-04-18:
@Jason8266 pointer solved this issue for me. Thank you.
Adding:
```
  String newHostname = "ESP32Node";
  WiFi.hostname(newHostname.c_str());
```
Just before `server.begin();`

Sys64736, 2022-05-02:
I have two boards, using latest master, only the devkitc-02 scans, creates an AP and will connect to router, the Lolin C3 mini will only scan wifi but no AP or connect to router.  Tried with and without serial.
Lolin WiFi scan works good but will not create AP and will not connect to router.   Using AP sample code above and a simple connect code.

```
C3 mini output
--- Miniterm on COM15  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
.....[  5461][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: ATP-240, BSSID: d8:0d:17:48:77:34, Reason: 2
[  5462][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  5469][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  5476][D][WiFiGeneric.cpp:889] _eventCallback(): WiFi Reconnect Running
[  5483][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
.......[  8541][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: ATP-240, BSSID: d8:0d:17:48:77:34, Reason: 2
[  8542][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  8549][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE

this keeps repeating every second

The devkit has no issues connecting.
```

For the AP, both boards give the same debug output and it looks okay but only the devkit board actually creates the AP.  I did try to set Hostname also but didn't make a difference.

```
entry 0x403ce000
[   187][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)

Configuring access point...
[  3201][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  3326][V][WiFiGeneric.cpp:325] _arduino_event_cb(): AP Started
[  3326][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 10 - AP_START
AP IP address: 192.168.4.1
Server started
```

The only thing I can think of is the devkit board has a serial chip and the Lolin is connected directly to pin 18,19 but not sure why that would make a difference with WiFi.

ljbotero, 2023-07-04:
The only way I was able to fix the AP not showing up was by setting the TX Power to a very low value, here's my full sample code:

```c
#include <WiFi.h>
#include <esp_wifi.h>  

void setup(){ 
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.setDebugOutput(true);
  delay(2000);
  Serial.println("Startup....");

  WiFi.hostname("esp32");
  WiFi.begin("BOTEROS-NET","qti6-9mi6-p5eu");
  WiFi.setTxPower(WIFI_POWER_8_5dBm);
  int txPower = WiFi.getTxPower();
  Serial.print("TX power: ");
  Serial.println(txPower);
  
  while (WiFi.status() != WL_CONNECTED && millis()<20000) {
    delay(500);
    Serial.print(".");
  }

  if(WiFi.status() == WL_CONNECTED){
    Serial.println("");
    Serial.println("WiFi connected.");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else {
    Serial.println("WiFi NOT CONNECTED, starting ap");
    delay(2000);    
    WiFi.mode(WIFI_AP);
    WiFi.softAP("espsoftap","12345678"); // no errors that I can see, but ap never shows up
  }
}

void loop(){
}
```

nkxingxh, 2024-12-21:
I had the same problem with esp32 c3 supermini.

Fortunately, after using `WiFi.setTxPower(WIFI_POWER_8_5dBm);` I could find the AP

NoFear23m, 2025-01-10:
The same issue her on my latest modules. I have esp32-ca-mini devboard from Waveshare
Is there any other fix without reduce the TxPower?

NoFear23m, 2025-01-10:
Many thanks for that link @lbernstone, that will help me a lot.