espressif/arduino-esp32 #6552: Missing definition of XTOS_DISABLE_ALL_INTERRUPTS for ESP32 C3

hallard, 2022-04-08:
### Board

ESP32 C3

### Device Description

Lolin ESP32 C3 Mini

### Hardware Configuration

No

### Version

latest master

### IDE Name

PlatformIO

### Operating System

Mac OS

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

When compiling missing some definition of `XTOS_DISABLE_ALL_INTERRUPTS` or `XTOS_RESTORE_INTLEVEL` 

it's defined for ESP32, ESP32-S2 and ESP32-S3 in `xtruntime.h` but can't find them for C3

### Sketch

```cpp
void setup() {
  XTOS_DISABLE_ALL_INTERRUPTS;
}
```

### Debug Message

```plain
Compiling /Users/charles/piobuild/esp32-c3-mini/src/test.ino.cpp.o
/Users/charles/Library/CloudStorage/OneDrive-Personnel/devt/esp32/_Tests/test_pio/src/test.ino: In function 'void setup()':
/Users/charles/Library/CloudStorage/OneDrive-Personnel/devt/esp32/_Tests/test_pio/src/test.ino:5:3: error: 'XTOS_DISABLE_ALL_INTERRUPTS' was not declared in this scope
   XTOS_DISABLE_ALL_INTERRUPTS;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/charles/Library/CloudStorage/OneDrive-Personnel/devt/esp32/_Tests/test_pio/src/test.ino:5:3: note: suggested alternative: 'taskDISABLE_INTERRUPTS'
   XTOS_DISABLE_ALL_INTERRUPTS;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
   taskDISABLE_INTERRUPTS
*** [/Users/charles/piobuild/esp32-c3-mini/src/test.ino.cpp.o] Error 1
============================================================ [FAILED] Took 1.00 seconds
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

igrr, 2022-04-08:
That error makes some sense, `xtos*` are Xtensa-specific functions, and ESP32-C3 CPU uses RISC-V architecture. I don't think these functions will be provided for C3. Could you please describe the problem you are trying to solve on C3 by using these functions? Perhaps we can suggest another way to solve it.

hallard, 2022-04-08:
Yes makes sense that was my 1st thought but macro is defined as follow

```cpp
#define XTOS_DISABLE_ALL_INTERRUPTS	XTOS_SET_INTLEVEL(15)
```

and saw somewhere on C3 code some calls to `XTOS_SET_INTLEVEL` so thought it was existing.

Anyway I'm trying to run excellent radiohead library and error fires on ATOMIC function, a kind of change of IRQ priority I guess, but I don't see any ESP32 specific so may be it's just unused in this case.

https://github.com/PaulStoffregen/RadioHead/blob/66744582c2a7cf93094d7f26a58d79837aeafc07/RadioHead.h

igrr, 2022-04-08:
I don't think that defining atomic-related macros to nothing is a good idea. It will give the impression that the code compiles correctly, but it's not actually going to work correctly.

I would suggest making a pull request to RadioHead library, suggesting different functions for ESP32:
```c++
#define ATOMIC_BLOCK_START int state = portSET_INTERRUPT_MASK_FROM_ISR();
#define ATOMIC_BLOCK_END portCLEAR_INTERRUPT_MASK_FROM_ISR(state);
```
This will work on both Xtensa- and RISC-V-based ESP32 chip series.

hallard, 2022-04-08:
Awesome thanks,
unfortunately this official lib (and so the latest one) is not on any github like, so PR (also looking on code) is not possible.
Official is here and only way to see is download zip file 
https://www.airspayce.com/mikem/arduino/RadioHead/index.html
So I'll try going thru mailing list to ask

igrr, 2022-04-08:
@hallard thanks, please report back if there's any issue. I have noticed Adafruit has a fork of this library as well, perhaps they would be open to a PR if it doesn't get accepted upstream.

I would try to avoid defining `XTOS_DISABLE_ALL_INTERRUPTS` for RISC-V to avoid more confusion. We never know, perhaps someone else does `#ifdef XTOS_DISABLE_ALL_INTERRUPTS  /* Xtensa specific code follows */`...

hallard, 2022-04-08:
@PaulStoffregen has also a fork and even I already made a fork 6 years ago [here](https://github.com/hallard/RadioHead), to get it working without interrupt on ESP target and run on PI (haha long time ago) but the deal is that everyone doing a fork need to maintain by hands on each release comparing each file which is a pain and why so much forks are outdated.
I solved this issue subclassing what I need, like that I can overwrite lib with latest version at no pain, but in this case, it's located on `radiohead.h` master include file and should be done at source level, but I'm pretty sure Mike will take care of this one.

PaulStoffregen, 2022-04-09:
Arduino's official API has only noInterrupts() and interrupts(), but no (officially support API) way to save & restore the state.

Indeed I've ported many libraries over the years, usually ones hard coded for AVR timers.  It's not fun, but much less tedious than trying to get people to agree on APIs.

VojtechBartoska, 2022-05-30:
Hello @hallard, is this still valid issue?

VojtechBartoska, 2022-08-10:
I'm closing this issue due to no answer. If you need, you can reopen it.

closed 2022-08-10 by VojtechBartoska

husnoo, 2024-10-14:
> Awesome thanks, unfortunately this official lib (and so the latest one) is not on any github like, so PR (also looking on code) is not possible. Official is here and only way to see is download zip [...]

Hi,

Did you get anywhere with this? I'd like to run RH on an esp32-c3.