espressif/arduino-esp32 #6554: Rx timeout seems not activated?

RobKeij, 2022-04-08:
### Board

Custom board with ESP32 PICO-D4

### Device Description

[image]

### Hardware Configuration

GPIO4 (RX)  GPIO2 (TX)  (board in test is only receiving)

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

  board ESP32 PICO-D4
  For me received serial data is only available when Fifo received 120 bytes.
  Fifo is giving a callback any time when FIFO reaches 120 bytes if timout is set to 0
  If I don't read the data from serial buffer I see 120 and later 240 Bytes
  in buffer (In the previuous version it's was max 480

  I want to read data directly when its received I read that possible  usable because 
  of the timeout, after a delay the FiFO data is transfered to the receive data.

  It looks that that the Timeout function is not working for me?

  What do I miss?

### Sketch

```cpp
//HardwareSerial Serial1(1); // if use this I got a error:  multiple definition of `Serial1'

unsigned char buff[10];

void setup() {

  Serial.begin(115200); //Debug port

  Serial1.begin(115200, SERIAL_8N1, 4, 2); // I use pin 4 (RX) and pin 2 (TX)
  //Serial1.setTimeout(100);              // set the Timeout to 100ms (should be default 10ms)

  pinMode(0, OUTPUT); // TX enable RS485
  digitalWrite(0, 0); // This pin must be low on mij test board (TX enable RS485)

  Serial.printf("Init ready\r\n");
}

void loop() {
  while (Serial1.available())  { // There can be more byte in the RS485 ringbuffer!
    
    // I sent every second 9 bytes  so with a delay of 1 second the 9 byte should be
    // available after a short time after receiving the 9 bytes.
    // So on the serial terminal I should see 9, 18, 27, 36 ......
    // Bur I see 120, 240  (so after sending 13 time 9 bytes)
    
    Serial.println(Serial1.available()); // Check how many bytes in buffer
  }
}
```

### Debug Message

```plain
No error messages, But Data is not transfered to the RX buffer
```

### Other Steps to Reproduce

The sketch expained it all

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

gonzabrusco, 2022-04-08:
Try with `Serial1.setRxTimeout(2);`

When setting a timeout of 0 you are disabling the timeout functionality. 

The callback should be set like this:

Serial1.onReceive(callbackFunction);

RobKeij, 2022-04-09:
Thanks for the replay gonzabrusco,
I still not work, if I look in  HardwareSerial.cpp  it must be ok but For me it don't work'.

if I look in HardwareSerial.cpp I see:  rxTimeout is default 100  
( I als try with added Serial1.setRxTimeout(2); in the sketch)

copied from HardwareSerial.cpp :
//////////////////////////////////////////////////////////////////////////
`HardwareSerial::HardwareSerial(int uart_nr) : 
_uart_nr(uart_nr), 
_uart(NULL),
_rxBufferSize(256),
_txBufferSize(0), 
_onReceiveCB(NULL), 
_onReceiveErrorCB(NULL),
_onReceiveTimeout(true),
_rxTimeout(100),
_eventTask(NULL)
#if !CONFIG_DISABLE_HAL_LOCKS
    ,_lock(NULL)
#endif
{
#if !CONFIG_DISABLE_HAL_LOCKS
    if(_lock == NULL){
        _lock = xSemaphoreCreateMutex();
        if(_lock == NULL){
            log_e("xSemaphoreCreateMutex failed");
            return;
        }
    }
#endif
}`
//////////////////////////////////////////////////////////////////////////

And in this function I see that after 120 bytes the Callback is triggered 
even if if set the callback from the sketch is seems not to work, It's still only works on evere 120 Bytes
received (with a max of 240) because default the rxbuffersSize is  256  ( _rxBufferSize(256))

copied from HardwareSerial.cpp :

//////////////////////////////////////////////////////////////////////////
`// timout is calculates in time to receive UART symbols at the UART baudrate.
// the estimation is about 11 bits per symbol (SERIAL_8N1)
void HardwareSerial::setRxTimeout(uint8_t symbols_timeout)
{
    HSERIAL_MUTEX_LOCK();
    
    // Zero disables timeout, thus, onReceive callback will only be called when RX FIFO reaches 120 bytes
    // Any non-zero value will activate onReceive callback based on UART baudrate with about 11 bits per symbol 
    _rxTimeout = symbols_timeout;   
    if (!symbols_timeout) _onReceiveTimeout = false;  // only when RX timeout is disabled, we also must disable this flag 

    if(_uart != NULL) uart_set_rx_timeout(_uart_nr, _rxTimeout); // Set new timeout
    
    HSERIAL_MUTEX_UNLOCK();
}`
//////////////////////////////////////////////////////////////////////////

RobKeij, 2022-04-09:
In my setup I use the "ESP32 PICO-D4"   ( RX (IO4)  and TX(IO2) )
I try the same test software  with a ESP32-WROOM and that works 
I still selected the "ESP32 PICO-D4" board in Arduino IDE.
But used different RX (IO16)  and TX(IO17)  

I try to make a setup with a other board  "ESP32 PICO-D4"  where I can use the same pins
as the ESP32-WROOM boards.

gonzabrusco, 2022-04-09:
Default r timeout should be 10. There's something wrong with your setup.

https://github.com/espressif/arduino-esp32/blob/6cfe4613e4b4846e1ab08c7f78b7ea241f52c7da/cores/esp32/HardwareSerial.cpp#L131

RobKeij, 2022-04-09:
Your right, I changed that for a test.
But Problem solved,
I use RS485 when serial is idle the output op the RS485 chip was low.
And then the RX time out is not triggered.  
@gonzabrusco "Thanks for thinking along"

[image]

[image]

[image]

RobKeij, 2022-04-09:
Problem solved, if idle signal is 0 volt RX time-out is not working. So keep serial data positive when serial is idle.

closed 2022-04-09 by RobKeij