espressif/arduino-esp32 #6556: Arduino serial monitor hang/freezes the microcontroller

janczeresnia, 2022-04-09:
### Board

ESP32C3 Dev Module

### Device Description

NodeMCU ESP-C3-01M-Kit

### Hardware Configuration

New board straight from the store connected only to USB

### Version

latest master

### IDE Name

Arduino IDE 1.8.19

### Operating System

Linux Gentoo

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Serial Monitor in Arduino freezes and locks the microcontroller. Every other program like minicom and cutecom works fine. Every time after connecting via UART to ESP32C0 the microcontroller is restarting. But it works. Only the serial monitor in Arduino completely blocks the MCU, it only happens with the ESP32C0 microcontroller, all others like XIAO Pico or ESP8266 work properly with the serial Arduino monitor.

### Sketch

```cpp
This happens with every sketch. The sketch doesn't matter here.
```

### Debug Message

```plain
There are no messages.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-11:
Hello @janczeresnia, thanks for reporting this.

@SuGlider Can you please help with triaging of this issue?

SuGlider, 2022-04-18:
@janczeresnia 

I guess, this is the board you are using:
https://docs.ai-thinker.com/_media/esp32/docs/esp-c3-01m-kit-v1.0_specification.pdf

[image]

SuGlider, 2022-04-18:
Unfortunately, I don't have such board here to test it.

I can see that it uses CH340 for Serial communication, which may be the port used to connect to the Arduino IDE Serial Monitor.
I also can see you are using Linux Gentoo. Is the Linux driver working correctly for the CH340?

In order to identify that ESP32-C3 really hangs/freezes, I would suggest using a sketch that creates a visual effect along with Serial writing, such as this:

``` cpp
//
//Sample code to control the RGB 5050 LED on the Ai-Thinker ESP32-C3-01M-Kit
//

// On the ESP32-C3, the RED Led of the RGB5050 is GPIO 3
#define PIN        3
#define DELAYVAL   500 // Time (in milliseconds) to pause between blinking

void setup() {

  Serial.begin(115200);  // activates UART Serial port on CH340 (USB)
  Serial.printf("\nRGB LED and UART Testing...\nBlinking delay is %d milliseconds.\n", DELAYVAL);
  Serial.printf("LED IO #%d\n", PIN);
  pinMode(PIN, OUTPUT);
}

void loop() {

    //Set the LED on and writes to Serial Monitor
    digitalWrite(PIN, HIGH);
    Serial.println("LED on."):
    delay(DELAYVAL);

    //Set the LED off and writes to Serial Monitor
    digitalWrite(PIN, LOW);
    Serial.println("LED off."):
    delay(DELAYVAL);

}
```

If LED continues to blink, even when Serial Monitor freezes, it means that the ESP32-C3 didn't hang up. 
It may be some issue with Linux. Try it with Windows if possible.

tonychengtl, 2022-04-28:
I am experiencing similar issue as well. When I enable "tools --> Serial Monitor" in Arduino IDE 2.0.0-rc3, the ESP32 Dev Module board freezes. It behaves like the Reset button is press and hold. When I close the Serial Monitor, the ESP32 board reboot and run normally. 
Last week everything was fine.
I have tried to search for information in the web just now, and this is the thread I found having similar symptom.

SuGlider, 2022-04-28:
@tonychengtl  - is you issue related to the ESP32-C3 or ESP32 board?
Does this issue happens with UART or USB CDC (C3/S2/S3 only)?

tonychengtl, 2022-04-28:
@SuGlider the board I am using is a 30-pin ESP32 Dev PCBA with ESP32-WROOM-32 module and CP2102 USB-to-UART bridge.
I guess something related to RTS and DTR. Perhaps I should try using my spare ESP32 board to identify hardware issue to IDE issue or USB driver (I am using Win10)

SuGlider, 2022-05-03:
@VojtechBartoska - I think it is not software related and we can close it soon.

VojtechBartoska, 2022-05-03:
I'm closing as HW issue.

@janczeresnia If it's needed please reopen the issue.

Thank you all for contribution.

closed 2022-05-03 by VojtechBartoska

guidol70, 2022-09-23:
> ### Board
> ESP32C3 Dev Module
> 
> ### Device Description
> NodeMCU ESP-C3-01M-Kit
> 
> ### Hardware Configuration
> New board straight from the store connected only to USB
> 
> 
> ### Description
> Serial Monitor in Arduino freezes and locks the microcontroller. Every other program like minicom and cutecom works fine. Every time after connecting via UART to ESP32C0 the microcontroller is restarting. But it works. Only the serial monitor in Arduino completely blocks the MCU, it only happens with the ESP32C0 microcontroller, all others like XIAO Pico or ESP8266 work properly with the serial Arduino monitor.

I also used the ESP-C3-01M-Kit today with the same problem, but it seems to be a problem how the terminal is accessing the USB-serial port of this board.

When accessing the USB-serial port via putty/kitty or Tera Term for Windows (also when set to none for flow control) the board seem to be in reset state until the connection is closed.

When using HTerm for Windows (https://www.der-hammer.info/pages/terminal.html)
the board does not go in freeze-mode (like with mincom and cutecom on Linux).
But HTerm has no ANSI-Emulation :(

I did get the Arduino IDE serial Monitor running , when you change (before starting the Arduino IDE)
the board.txt for esp32 Board-Support and change the lines
( see https://www.reddit.com/r/esp32/comments/pf61v1/esp32c3_serial_output_in_arduino_ide/ )
from
esp32c3.serial.disableDTR=false
esp32c3.serial.disableRTS=false

to
esp32c3.serial.disableDTR=true
esp32c3.serial.disableRTS=true

With a TTL-serial USB-converter on RX/TX/GND  (not CDC) the output works without these freeze ;)

Mira141, 2023-05-03:
Unfortunately the same problem:
ESP-C3-01M-Kit stopped when serial monitor in Arduino is turned ON.
Arduino IDE 2.1.0 , Windows 8.1,  will be checked at SuSE Linux Leap15.4 soon too.

Other Kits works well: ESP32-WROOM-32U is OK, ESP32-S3-DevKitC-1 N16R8 is OK .

Unfortunately, setting ...
esp32c3.serial.disableDTR=true
esp32c3.serial.disableRTS=true

... in boards.txt ... did not help ... Probably hardware issue on ESP-C3-01M-Kit, as when switchng Serial Monitor ON, some readable data (about 2-3 lines goes thru, start and end words are cutted off in middle of word) sent from board came thru, before all stops. If swithed serial monitor off, board works well from reset status. When activating Serial monitor, level at pin "EN" (enable) drops to LOW. Solution probably need somehow keep pin EN in high state, but resistor 1k5 to +3,3V was not successfull, probably CH340 keeps low level hard and I want not to damage it by connecting direct to +3,3V . 
Has anybody any solution except "Do not use ESP-C3-01M-Kit" ???

guidol70, 2023-05-04:
My ESP-C3-01M-Kit doesnt get damaged/keeps working while using directly the 3.3v to set EN high.
When programing/flashing via Arduino-IDE I have to disconenct the cable, but for using the Kit after flashing I have to connect the cable EN<->3.3v

The text before it turn off is the Boot-ROM Message (which I also have on a ESP32 Lite V1.0.0 Rev1)

guidol70, 2023-05-04:
@Mira141 
If you have the CHIP enabled via EN and a HIGH signal and dont get any out-/input from the Serial. from your sktech, then you have maybe to reroute SerialUART or SerialCDC to Serial depending on which one do you want to use.
SerialUART is connected to the CH340G TTL-serial chip and GPIO20(RX0) & GPIO21(TX0)
SerialCDC is connected to GPIO18(USB D-) & GPIO19(USB D+) as a 2nd USB-serial-CDC-Port

Rerouting is done in a Arduino-Sketch via (activate the needed lines):

```
// ====================================================================
// HW_CDC_ON_BOOT is DISABLED = USB CH340G Bridge
// https://www.esp32.com/viewtopic.php?t=24280
// reconnect 3.3V & EN(ABLE) after flashing = setting EN HIGH = ENABLE CHIP/CPU
// ====================================================================
#define SerialUART Serial
#define  SerialCDC USBSerial

// ====================================================================
// HW_CDC_ON_BOOT is ENABLED = USB GPIO Pin 18 D- (white) and 19 D+ (green)
// ====================================================================
// #define SerialUART Serial0
// #define  SerialCDC Serial

```

Ariel-International, 2023-05-08:
Same issue in VS Code, it was DTR/RTS set in Serial monitor

florinbaciuu, 2023-05-08:
I also own this development board and I confirm that the same result occurs when I open the serial monitor in arduino ide. only in arduino ide. if I use another application, the board works without problems. modified in boards.txt the parameters related to DTR and RTS and in vain.  I want to believe that the board was not supposed to behave like this. I think it's a hardware bug.